raspiblitz/raspiblitz #954: Change raspiblitz to use IPv6

janis-mueller, 2020-01-04:
My ISP uses dual stack (see https://whatismyipaddress.com/dual-stack)
So the external IPv4 address I am using can not be used for port forwarding (because I share it with a lot of other people and it does not point directly to my network like the IPv6 does), so I need to open the ports for the IPv6 address (which is no problem), but the raspiblitz uses the IPv4 address by default but I need it to work with the IPv6 one, and I see no way to solve this issue yet.

frennkie, 2020-01-04:
Which applications do you want to make accessible? Bitcoin should work directly. Also additional services like RTL should work.

For LND I'm not quite sure.

janis-mueller, 2020-01-04:
inside my network, everything works fine, I mean I want to be able to reach my Bitcoin and Lightning Node from the outside and accept incoming connections from port 8333 and 9735 which is only possible via the IPv6 address but raspiblitz uses my external IPv4 and I need to change that

frennkie, 2020-01-04:
Run `netstat -6 - tulpen`. You should see that bitcoind is listening on v6 - so this should work. You might need to allow inbound traffic for that port to the raspiblitz on your router/gateway/firewall.

For LND there are (at least) 3 different services: 

* 9735 for the p2p traffic.
* 10009 gRPC
* 8080 Rest API

On my Raspiblitz they currently only listen on v4. I'm not sure whether v6 is fully supported on all... You should be able to modify the lnd.conf file to try it...

janis-mueller, 2020-01-04:
> Run netstat -6 - tulpen. You should see that bitcoind is listening on v6 - so this should work. You might need to allow inbound traffic for that port to the raspiblitz on your router/gateway/firewall

You don`t seem to understand my problems. I followed the steps above, but my ISP only allows me to forward the IPv6 address of the raspiblitz, not the IPv4 one which the device uses and I am looking for a way to change that

PatrickScheich, 2020-01-05:
From my experience raspiblitz uses the IPv6 addresses as well as IPv4 addresses but it is showing only the IPv4 address.

**First for the bitcoind part...**
Please check the IPv6 addr with 
`ip -o -6 a s eth0 | grep global` 

after you configured the port forwarding of port 8333 to that IPv6 addr on your router you can check the reachability of your bitcoin node at [https://bitnodes.earn.com/](https://bitnodes.earn.com/).

Tip: Be sure to omit leading and trailing spaces when you copy/paste your IPv6 addr into the address box at the above website.

janis-mueller, 2020-01-05:
Thanks, just checked, my node is answering on  port 8333 via IPv6 even though the pi still displays the IPv4 address in red, would be nice to get a fix for this

frennkie, 2020-01-06:
If the red line on the Raspiblitz dashboard is your concern then one fix is to use Tor.

This is anyway the recommended operation mode if you care about privacy (at all).

janis-mueller, 2020-01-06:
> If the red line on the Raspiblitz dashboard is your concern then one fix is to use Tor.
> 
> This is anyway the recommended operation mode if you care about privacy (at all).

No the red line is not the concern at all. Not using TOR is a feature of Raspiblitz and it clearly doesn't even display IPv6 addresses, which is a major flaw in my eyes!

PatrickScheich, 2020-01-06:
> Thanks, just checked, my node is answering on port 8333 via IPv6 even though the pi still displays the IPv4 address in red, would be nice to get a fix for this

I just took a brief look into the scripts....maybe this might help
(this should be the code that determines "the public IP address" of the raspiblitz)

File: /home/admin/_background.sh
see section 
`  ####################################################`
`  # RECHECK PUBLIC IP`
`  # when public IP changes, restart LND with new IP`
`  ####################################################`
there is a line 
`freshPublicIP=$(curl -s http://v4.ipv6-test.com/api/myip.php 2>/dev/null)`
replace it by
`freshPublicIP=$(curl -s http://v6.ipv6-test.com/api/myip.php 2>/dev/null)`

janis-mueller, 2020-01-06:
> I just took a brief look into the scripts....maybe this might help
> (this should be the code that determines "the public IP address" of the raspiblitz)
>
> File: /home/admin/_background.sh
> see section
> ` [...]

Tried that, but did not work, but really appreciate your inputs!

frennkie, 2020-01-06:
> Not using TOR is a feature of Raspiblitz

Well actually.. I would say the real **feature** is to be able to run the Raspiblitz with `bitcoind`, `LND` and a lot of other services **with** TOR.

But I completely agree that - if possible - it should work in all scenarios the operator selects to run:
1. IPv4 public IP only
2. IPv6 public IP only
3. TOR only
4. IPv4 private IP only (will need NAT or SSH Reverse Tunnels to get working)
5. IPv4 and IPv6 (both public)
6. ...?!

> Thanks, just checked, my node is answering on port 8333 via IPv6 even though the pi still displays the IPv4 address in red, would be nice to get a fix for this

So bitcoind is working - but a fix for the dashboard showing IPv6 in red would be nice, right? This would most likely go into `~/00infoBlitz.sh`.

Next: how about `LND`..? What you can try is to manually put your IPv6 address into the service file and restart lnd:

/etc/systemd/system/lnd.service

```
[...]
ExecStart=/usr/local/bin/lnd --externalip=[2001:...:6875]:9735
[...]
```

Then run: 

```
sudo systemctl daemon-reload
sudo systemctl restart lnd
```

frennkie, 2020-01-06:
By the way... is your IPv6 IP static? I get a different IPv6 prefix assigned everyday (similar as I get a new IPv4 every 24 hours). So I would need to do a similar "hack" and update the service script + restart every day. Unless `lnd` implments a way to set the `externalip` via RPC.

janis-mueller, 2020-01-06:
> By the way... is your IPv6 IP static? I get a different IPv6 prefix assigned everyday (similar as I get a new IPv4 every 24 hours). So I would need to do a similar "hack" and update the service script + restart every day. Unless `lnd` implments a way to set the `externalip` via RPC.

Thanks for your answer, yes my IPv6 address is static, luckily. I will try what you suggested and will update you here tomorrow on how it went.

janis-mueller, 2020-01-10:
Tried what you suggested now, my raspiblitz shows the IPv6 address as the lnd address at the bottom now, so this seems to work as I have 2 opened channels with other lightning nodes.
My bitcoin core address shows my dyn dns address, I am not sure how to check if that is working, I think I need some help with that.
Sorry for the late response.

frennkie, 2020-01-10:
Hm... so you now you want to make sure that also `bitcoind` mainly/only uses your public IPv6 address..? You could try this setting: `-onlynet=ipv6` (on the `ExecStart` line in `/etc/systemd/system/bitcoind.service`).

Or what is your question?

janis-mueller, 2020-01-11:
Wasn`t directly looking for that option but this was helpful anyway, I think this resolved the issue, I would close the issue from my end

PatrickScheich, 2020-01-11:
I just looked into the shell scripts. It seems that the public IP is determined in 2 scripts

- _background.sh
- _bootstrap.sh

the following code is used
`freshPublicIP=$(curl -s http://v4.ipv6-test.com/api/myip.php 2>/dev/null)`
in my case (Carrier-grade NAT) this results in the IPv4 address of my ISP, which is undesirable.

with the easy fix

`freshPublicIP=$(curl -s http://v6.ipv6-test.com/api/myip.php 2>/dev/null)`

I would get the routeable IPv6 of my raspiblitz. The IPv6 address would be in the format

`2a02:6d40:xxxx:yyyy:zzzz:aaaa:bbbb:cccc`

unfortunately that leads to a problem when starting lnd, as the systemd config `/etc/systemd/system/lnd.service` says...

`EnvironmentFile=/mnt/hdd/raspiblitz.conf`
`ExecStart=/usr/local/bin/lnd --externalip=${publicIP}:${lndPort}`

which will result in the cmnd

`ExecStart=/usr/local/bin/lnd --externalip=2a02:6d40:xxxx:yyyy:zzzz:aaaa:bbbb:cccc:8333`

and lnd will shutdown. I fixed that (temporary) by add the line

`freshPublicIP='['${freshPublicIP}']'`

after IPv6 is verified through the regex in the above *.sh. Now lndservice is happily starting (and lncli getinfo shows the IPv6 addr of my node), but now `00infoBlitz.sh` fails in the line 

`public_check=$(nc -z -w6 ${public_ip} ${public_port} 2>/dev/null; echo $?)`

as nc does not like the brackets

rootzoll, 2020-01-14:
Maybe a future version should have a `forceIPV6` option?

janis-mueller, 2020-01-14:
It definitly should, I think I will close this for now, lnd and bitcoind are now working with IPv6 with the help of the posts of @PatrickScheich  and @frennkie

rootzoll, 2020-01-14:
I was adding it to future release collector .. keep it open, so that it pops up later on.

Maxstgt, 2020-02-21:
> When trying to connect my node to zap android I still get the following output:
> http://www.mediafire.com/file/rmcdcvjpgdgwi6r/lnd_ipv4-edited.png/file
> The public IPv6 address is attached to my subdomain name (and successfuly resolving)
> but the script still tells me to open ports for my local IPv4 address, I think this is the reason lnd cannot connect to my node (zap used to be able to connect locally before, just not on a public level now via my dyn dns), all necessary ports are open
> any ideas?

I'm using 4G on my phone and I only got a ipv4 for my smartphone. You may have to setup a portmapper that maps a ipv4 to a ipv6 address. But I'm not an expert on this topic, I may be wrong!

rootzoll, 2020-09-17:
Everybody interested in testing IPv6 support on RaspiBlitz see: https://github.com/rootzoll/raspiblitz/issues/869

cstenglein, 2023-06-07:
Closing this for now since there is a resolution in the other thread

closed 2023-06-07 by cstenglein