raspiblitz/raspiblitz #955: Question: How to mount the blockchain HDD on an other linux system

PatrickScheich, 2020-01-04:
due to some corruption on the RaspiBlitz HDD I had done a fschk, after that bitcoind complaint about missing indicees on the blockchain files. So I started it once with the "-reindex" option. Unfortunately even the RPi4 takes a really long time to reindex the whole blockchain. 

I tried to mount the HDD on another linux system (with a bit more beefy hardware) to do that work and then switch back to the RPi4 but it seems that there is some kind of folder/file encryption. 

Is the some password or keyfile I need to get from the original RaspiBlitz?

rootzoll, 2020-01-06:
There is no encryption on the HDD .. another linux system should be able to mount the EXT4 file systems.

closed 2020-01-06 by rootzoll

PatrickScheich, 2020-01-07:
yes, sometimes "sudo" is not enough you have to be root