raspiblitz/raspiblitz #958: After recover no open funded  chances in my wallet

btc21cluib, 2020-01-05:
Since 16. Nov I try to force close 2 old empty channel and my wallet broke this action.
I made a channel backup via RaspiBlitz and I recover for this file and my seeds. The Wallet comes back with about 50% of my funds but no pending to open channel. After finish syncing, there was no more funds and channels in my wallet - on 1ml.com there are still funds and  8 channels shown . I try it out with RaspiBlitz 0.14 and LND 0.8.2 and bitcoin bitcoin qt 0.19, but still no channel, only in the rtf show this 2 old pending to close channels but not my funded one. 
Anyone has a idea to recover my wallet?
Thank you.

openoms, 2020-01-06:
Please read through the FAQ where all options are described in detail:
https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-can-i-recover-my-coins-from-a-failing-raspiblitz

21-btcclub, 2020-01-11:
Success - With the Zap Desktop Wallet I was able to recover and back up my broken wallet with only the seed words of all my funds.

closed 2020-01-14 by rootzoll