raspiblitz/raspiblitz #960: RTL indirect depedency on Python2

frennkie, 2020-01-06:
Testing v1.4 I get a warning/error regarding Python2 (which we removed):

```
gyp verb check python checking for Python executable "python2" in the PATH
```` 

The dependency chain is (at least):
`RTL` needs `node-sass` which needs `node-gyp@3.8.0` which is completely outdated but the node-sass developer refuses to upgrade at the moment: https://github.com/sass/node-sass/issues/2716.

I also got this: https://github.com/Ride-The-Lightning/RTL/issues/173

```
*** Run: npm install ***

> node-sass@4.12.0 install /home/admin/RTL/node_modules/node-sass
> node scripts/install.js

Downloading binary from https://github.com/sass/node-sass/releases/download/v4.12.0/linux-arm-64_binding.node
Cannot download "https://github.com/sass/node-sass/releases/download/v4.12.0/linux-arm-64_binding.node":

HTTP error 404 Not Found
```

I consider the *node eco-system* completely broken.. :-(

frennkie, 2020-01-06:
@saubyk is there an alternative to `node-sass` v4 ? Is a (breaking) upgrade to 5 (an therewith to a newer node-gyp version) expected soon?

ShahanaFarooqui, 2020-01-09:
@frennkie There is no alternative to node-sass or node-gyp. They are being used for Angular Material library which is being heavily used in RTL for UI/UX. Waiting for upgrade might also not be helpful.

But, I am a bit confused about the issue here. RTL's 173 issue shows the error for node-sass but actually other dependencies were missing from their OS and they have to install them separatly. Hope this helps otherwise please feel free to connect on Telegram/Twitter for troubleshooting session.

ShahanaFarooqui, 2020-01-09:
@frennkie, Just recalled below 2 solutions with almost related issues. 

1) Try npm install with '--unsafe-perm' flag. It looks like the issue is not related to node-sass/node-gyp istallation but related to user permissions. Some of our other users were also NOT able to install dependencies with 'sudo' command also. '--unsafe-perm' ensures that npm will always run within the same context as running script and solves the problem.

https://github.com/Ride-The-Lightning/RTL/issues/148#issuecomment-513583859
https://github.com/Ride-The-Lightning/RTL/issues/195#issuecomment-544104393

2) Install RTL v0.5.3 or greater. There was an issue with v0.5.2.
https://github.com/Ride-The-Lightning/RTL/issues/187#issuecomment-541366948
https://github.com/Ride-The-Lightning/RTL/releases/tag/v0.5.3

openoms, 2020-01-09:
Did not manage the new RTL version to work yet on RaspiBlitz 1.3 or 1.4RC1.
Some of the errors:
https://pastebin.com/raw/15XU2z7s
https://pastebin.com/raw/zNyxFZz8

tried to build with `--unsafe-perm` with the same outcome.

frennkie, 2020-01-09:
@ShahanaFarooqui the short term fix is pretty simple. We have to reinstall Python2 before we install RTL.

This is not nice, because Python2 has been deprecated - but currently not too bad.

The midterm solution is that RTL upgrades to a new version of node-sass (v5+). I have opened an issue for this on the RTL repo.

ShahanaFarooqui, 2020-01-09:
@frennkie, @openoms, @rootzoll, What is your NodeJS version? 

We are currently using node-sass ^4.13.0 which is compatible with NodeJS 13:
https://github.com/sass/node-sass#supported-nodejs-versions-vary-by-release-please-consult-the-releases-page-below-is-a-quick-guide-for-minimium-support

I found a good explanation on python requirement for node-sass below. Please check and let me know if this version incompatibility can be resolved by downgrading node-sass from our end. We can try to replace it with some other library but most of those are also dependent on node-sass :).

We will be happy to upgrade node-sass to v5 as soon as they will release but their last update is 2 months old. 

https://stackoverflow.com/questions/57560857/node-sass-sass-loader-alternatives
=====================================================
Which says:
node-sass does not require Python 2. Building native modules in Node.js does require Python 2 https://github.com/nodejs/node-gyp. node-sass ships prebuilt bindings for supported platforms https://github.com/sass/node-sass#supported-nodejs-versions-vary-by-release-please-consult-the-releases-page-below-is-a-quick-guide-for-minimium-support so you should only need Python to rebuild if your network blocks downloading those files from the GitHub releases or you're running old node-sass + new Node.js versions.

If that doesn't work for you, Dart-Sass can sometimes be used as an option in those tools https://github.com/sass/dart-sass
===================================================

ShahanaFarooqui, 2020-01-09:
@frennkie Can you please test RTL from its master branch? We moved the node-sass dependency to 'development dependency' list only and it should not be an issue anymore.

openoms, 2020-01-09:
The latest consensus in the RTL-App TG is to switch to Node 12 and test building from the latest RTL master.
Will test soon.
Will also need to check BTC-RPC-explorer if it works with the updated NodeJS.

frennkie, 2020-01-10:
> Can you please test RTL from its master branch? We moved the node-sass dependency to 'development dependency' list only and it should not be an issue anymore.

@ShahanaFarooqui Not 100% sure I did the right thing.. but here is the (failed) output from a `npm install` of commit `917feebfa4fb583360c140e817c266649307ef72` https://pastebin.com/8AUACxLT

ShahanaFarooqui, 2020-01-11:
@frennkie Looks like the value of NODE_ENV is not set to 'production' on the node. I should have mentioned earlier that, only to install production dependencies, it should be set to production.
Or, you can run 'npm install --only=production'. To avoid future confusions, we have updated the documentation to install production dependencies with 'npm install --only=production'.
But, please note that '--only=production' will take precedence over NODE_ENV value. So to install development dependencies for any reason, please do not forget to remove '--only=production' even when you have NODE_ENV set to development. 

Also, we went through all other dependencies and updated/deleted them to keep the build as small as possible. So please pull the latest commit and try with 'npm install --only=production'.

rootzoll, 2020-01-26:
OK tested the install of RTL again and it seems stable working for v1.4 release. Closing issue.

closed 2020-01-26 by rootzoll