raspiblitz/raspiblitz #963: Idea: precomile electrs

frennkie, 2020-01-07:
The electrs script first installs rust and and compiles electrs which take about 30 minutes or so.

An alternative approach would be to install rust at sd card build. Then get the elects code and run cargo-deb. This create a debian .deb package that can be installed in less that 30 seconds.

rootzoll, 2020-01-08:
As long as Rust is not installed from an official debian repo I would not like to have it part of a sd build script - so that it stays optional to install. Idea: Reduce dependencies on external code (outside base os ecosystem) as much as possible for the basic sd card build.

frennkie, 2020-01-08:
Well.. we don't necessarily need to include the installation of rust and compilation of the .deb package in the sd card build. This could be done separately (maybe even on a Travis CI host?!). All we need is the resulting file. Which is all we need to copy onto the sd card (e.g. into the assets folder).

frennkie, 2020-01-08:
Like this: https://github.com/frennkie/raspiblitz/blob/use-precomiled-electrs/home.admin/assets/electrs_0.8.0_armhf.deb

openoms, 2020-01-09:
Installing electrs takes time anyway due to the indexing.
I agree with rootzoll to not include more things with the basic sdcard build.
I do like that we are compiling code from source when possible and only use prepapacked packages from apt or  signed by the devs. I understand that is a UX issue, but also a good differentiation from other projects.

For my own use would prefer not using a precompiled electrs from a trusted source and also would not want to take up more responsibility.

frennkie, 2020-01-09:
I see your points..thanks for the feedback. 😊

closed 2020-01-09 by frennkie