raspiblitz/raspiblitz #966: BTC and LTC on same Network How to change Port Number

3DPrintingNoob, 2020-01-09:
I'm running Raspiblitz on two different raspberry pi's (pi 3B+, and pi 4 4GB) with one running LTC and another BTC. 

My Bitcoin Raspiblitz shows my Public IP in yellow with the port 8333 and it has 9 connections and 3 peers and RTL page works fine

My Litecoin Raspiblitz shows my Public IP as red with port 8333 and it has 13 connections and 0 peers and the RTL page DOESN'T work as well.

My guess is that both raspberry pi units with the clients are conflicting because they are using the same port number. How does one change the port in the Raspiblitz from 8333 to 9333 ? I'm a complete noob with linux and have been going through a bunch of youtube videos to learn linux commands..... I think this will allow my LTC RTL to work properly as currently when I type in my RPC password it doesn't work so I can't logon and I know I have my RPC password correct.

I created the nodes with the help of this video.
https://youtu.be/cUd2TtnGRCk?t=1103

3DPrintingNoob, 2020-01-09:
Where I'm at now since I literally know almost nothing about linux. 

I found litecoin under assets file

I googled how to open .conf files and the command cat came up.

Then I found out that nano command can edit files so I tried opening the file and it says directory is not write-able. So now working on this step.

rootzoll, 2020-01-09:
see: Script to change port of bitcoind #487 - its a future idea to have a script to make port changes for BTC/LTC.

rootzoll, 2020-01-09:
to edit config files use `sudo` before the editor command to get permission

3DPrintingNoob, 2020-01-09:
> to edit config files use `sudo` before the editor command to get permission. I edited the files with nano

Thanks for the tip. I was able to view the files below located in the assets directory and make changes
litecoin.conf 
# Connection settings
zmqpubrawblock=tcp://127.0.0.1:29332
zmqpubrawtx=tcp://127.0.0.1:29333

litecoind.service 
No changes in this directory as I didn't see anything relevant to change

lnd.litecoin.conf
[Litecoind]
litecoind.zmqpubrawblock=tcp://*:29332
litecoind.zmqpubrawtx=tcp://*:29333

Basically I changed the 28332 and 28333 to 29332 and 2933. The question is no where did I see in any of these three .conf files a place to change the port number from 8333 to 9333

frennkie, 2020-01-09:
Use `port=`

```
# Port on which to listen for connections (default: 8333, testnet: 18333, regtest: 18444)
#port=
```

3DPrintingNoob, 2020-01-09:
> Use `port=`
> 
> ```
> # Port on which to listen for connections (default: 8333, testnet: 18333, regtest: 18444)
> #port=
> ```

So I edited my litecoin.conf file and added

#Connection settings
rpcuser=Removed Entry
rpcpassword=Removed Entry
port=9333
zmqpubrawblock=tcp://127.0.0.1:29332
zmqpubrawtx=tcp://127.0.0.1:29333

I saved file and reopened it to verify the port number was there. I then proceeded to sudo shutdown -r now and then turned it back on and checked the Info screen it still shows my public IP with port as 8333. 

I decided maybe something was still referencing the bitcoin.conf file so I did the same thing to that file

# Connection settings
rpcuser=Removed Entry
rpcpassword=Removed Entry
port=9333
rpcport=8332
rpcallowip=127.0.0.1
rpcbind=127.0.0.1:8332
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333

I saved file and reopened it to verify the port number was there. I then proceeded to sudo shutdown -r now and then turned it back on and checked the Info screen it still shows my public IP with port as 8333. 

Any other ideas?

rootzoll, 2020-01-14:
The Infoscreen at the moment is not able to show a different port as 8333 (mainnet) or 18333 (testnet). So your litecoind is most prob running already on your set port, but infoscreen gives you false info. 

Once we have a script to set a custom port with the following feature https://github.com/rootzoll/raspiblitz/issues/487 also the infoscreen will be able to show that custom port. This is a future feature, not implemented yet.

openoms, 2021-11-08:
litecoin is not supported any more

closed 2021-11-08 by openoms