raspiblitz/raspiblitz #971: BlitzTUI: touch is upside down with default LCD rotate

openoms, 2020-01-11:
In v1.4 in the build_sdcard.sh we have changed the default LCD rotation.

This need to be recorded in the raspiblitz.conf as `lcdrotate=1` otherwise the BlitzTUI touch will remain upside down until the LCD rotate option is selected in the SERVICES.

related: https://github.com/rootzoll/raspiblitz/issues/892

openoms, 2020-01-11:
Would it make more sense to assign `lcdrotate=0` to the new default value and change the default rotation in the BlitzTUI?

In this case we could avoid having a box ticked  in the SERVICES menu on a fresh install. 

Will need to modify the `blitz.lcdrotate.sh` to work accordingly.

frennkie, 2020-01-11:
Well... What is the 'official' correct rotation of screen. So if I can read the text on the screen. Should the USB/LAN port face to the right or to the left..? 😇

openoms, 2020-01-11:
The untouched rotation is when the USB ports are on the left.

We have changed the default because of the metal case being orientated the other way. Just looking at the CryptoCloaks cases, those actually have the USB-s on the left too so probably should change the default back. 

Doing it now.

frennkie, 2020-01-11:
Ok. Thanks.

openoms, 2020-01-11:
Being confused now.

In the 1.3 (master) the default is: https://github.com/rootzoll/raspiblitz/blob/master/build_sdcard.sh#L681
```
sudo sed --in-place -i "57s/.*/dtoverlay=tft35a:rotate=270/" /boot/config.txt 
```
And is still the same in 1.4 https://github.com/rootzoll/raspiblitz/blob/v1.4/build_sdcard.sh#L685

It is either:
`dtoverlay=tft35a:rotate=270` - LightningShell (Cryptocloaks)
`dtoverlay=tft35a:rotate=90` - USB on the right (good for the metal case)

in  the  /boot/config.txt.

Not sure now why my LCD was oriented with the USB to right on 1.4RC2, cant` a find  acommit either where we would have changed it.

Will keep an eye on this with the next fresh build.

rootzoll, 2020-01-14:
Keep in mind that raspiblitz tries to respect old rotate settings on version upgrade. If you started with an old version (where there was no LCDrotate option) the USB-Ports were to the left - that was the deafult which is now (lcdrotate=0). If you upgrade from an old version it tries to keep that setting. But if you start fresh with a new version, it should have lcdrotate=1 (with USB ports on the right) from the beginning ... thats what the new cases go for.

rootzoll, 2020-01-25:
OK in tests on a fresh new Blitz when turning Touchscreen ON it was in wound touch-rotate. Working on fix.

rootzoll, 2020-01-26:
OK - I tested my fix and it looks good. Closing issue.

kilrau, 2020-02-13:
Upgraded my Pi4 from 1.3 to 1.4RC3 -> screen upside down (usb ports are left)
* Enabled lcd rotate via service menu, reboot, still upside down
* Set `lcdrotate=0` (was 1) in `raspiblitz.conf`, reboot, still upside down
* Set `lcdrotate=1` (was 0) in `raspiblitz.conf`, reboot, still upside down

openoms, 2020-02-13:
@kilrau can you please specify which screen you have? A photo of the back of the screen would be most useful.

kilrau, 2020-02-13:
[image]

PS: Touchscreen Service Option is off

openoms, 2020-02-13:
@kilrau thanks, this seems to be the same screen I use, but can`t reproduce the problem.

Try 
`./config.scripts/blitz.lcdrotate on` or `off`

You can edit manually the file: `/boot/config.txt`
Change the `dtoverlay=tft35a:rotate=` betwen `90` or `270`

This is what the `./config.scripts/blitz.lcdrotate.sh` changes.

kilrau, 2020-02-16:
`bash ./config.scripts/blitz.lcdrotate.sh off` did the trick

SvenSommer, 2020-12-11:
I have the issue also in v1.6.2. Upgraded from 1.3
The mentioned script is no longer available in version 1.6.2

rootzoll, 2020-12-14:
The script is now:

`bash ./config.scripts/blitz.lcd.sh rotate off`

openoms, 2021-07-08:
great, just noting here that managed to find some variation with the screens.
Got two which has the touch upside down despite `lcdrotate=1` sitting correctly in the metal case.

Applying the stting from this commit https://github.com/rootzoll/raspiblitz/commit/f33af14951a3b367b1bd48aee1885968cfbdcb32 solves it after a  restart.

Anyone comes across this issue, just paste these couple of lines to the terminal:

```
    echo "Activate Touchscreen Rotate"
    cat << EOF | sudo tee /etc/X11/xorg.conf.d/40-libinput.conf >/dev/null
Section "InputClass"
        Identifier "libinput touchscreen catchall"
        MatchIsTouchscreen "on"
        Option "CalibrationMatrix" "0 1 0 -1 0 1 0 0 1"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
EndSection
EOF

```

type `menu` and choose REBOOT.

rootzoll, 2021-08-19:
Marked issue to take a look again for screen rotation with v1.8