raspiblitz/raspiblitz #975: update to 1.4RC2 wiped drive

ostermayer, 2020-01-13:
Updated from 1.3 to 1.4RC2 wiped drive and redownloaded blockchain
Steps
1. Activated update through menu on 1.3
2. downloaded copy of lnd-backup
3. blitz shutdown and i inserted new sd card with 1.4rc2 image
4. Prompted for new passwords then blockchain redownload started and drive was wiped

rootzoll, 2020-01-14:
I took the download of the RC2 offline - for now. Will check whats going on.

rootzoll, 2020-01-15:
@ostermayer the only place where a ssd/hdd format is happening is during the initial setup. So it seems that your recovering/update process was not triggered and you went thru the setup like a fresh raspiblitz - or did you see any "recover" or "update" info? 

To prevent an unwanted formatting of the HDD/SSD I added now for the final v1.4 release a Dialog Box asking to format the HDD/SSD - so that you still can cancel it at that point. Still not sure why recovering/update process was not triggered.

rootzoll, 2020-01-28:
This is working and will be part of v1.4 release. Closing issue.

closed 2020-01-28 by rootzoll