raspiblitz/raspiblitz #979: Multiple nodes connection

thephizzz, 2020-01-16:
I have my Raspiblitz up and running fine. Peers, channels, funding. Thank you for that.
I decided to run a second lnd node on my desktop. Installed and git it running. Peers, no channels yet, funding.
Both show up fine in RTL and i'm trying to open a channel between the two.
But - when I try to lookup the other node, I always receive an error. Doesn't matter if node1 looks up node2 or vice versa.
Any thoughts?

node1: Raspiblitz v1.3 with lnd v 0.90beta; bitcoin v0.18.1
node2: lnd v0,90-beta; bitcoin v0.18.1
(2 separate bitcoin)

Peer Lookup Failed
Code: 500
Message: error:unable to find node 
Message: unable to find node 
Code: 2
Url: ./api/lnd/network/node/03498c2308b2bc24457af7f82605a112186bdb40835e9283548b5b9b574b1bf84c

thephizzz, 2020-01-16:
Public addresses:
node1: 03498c2308b2bc24457af7f82605a112186bdb40835e9283548b5b9b574b1bf84c (Tor)
node2: 030bfe79edbab48b93bab7a550bb87d0c1a32bb8c8db1cc22fe4a13b54a3b0442e (clearnet)

talvasconcelos, 2020-07-22:
Ok, so I´'m facing the same problem! Runing 2 nodes Raspibolt and Raspiblitz, both on TOR, at home.

In RTL on the blitz i think it only sees me.  The Network tab show Network capacity 0 Number of nodes 1
On the raspibolt, RTL does see the network but when trying to add a peer I get "tor general error" 

Not sure what I missed or screwed...

rootzoll, 2020-07-23:
@talvasconcelos it looks like something with your TOR install went wrong. Can you go into "Services" or "Settings" and switch off TOR reboot and then witch on TOR again and reboot when installed. Then check the TOR monitor in the main menu.

talvasconcelos, 2020-07-24:
OK, did that still can´t connect to network...
```COMMAND LINE:
lncli --chain=bitcoin --network=mainnet connect 02c16cca44562b590dd279c942200bdccfd4f990c3a69fad620c10ef2f8228eaff@[2a01:4f8:c0c:7b31::1]:9735

RESULT (might have to wait for timeout):
[lncli] rpc error: code = Unknown desc = dial tcp [2a01:4f8:c0c:7b31::1]:9735: connect: network is unreachable```

rootzoll, 2020-07-24:
can you check that that ne node that the node has Internet connection - run on terminal: `ping 8.8.8.8 -c 1 -W 2`

talvasconcelos, 2020-07-24:
```PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=116 time=78.1 ms

--- 8.8.8.8 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 78.112/78.112/78.112/0.000 ms```

Node syncs bitcoin and has internet... it's LND that is not connecting, can´t figure out why. I´ve nano'd the bitcoin.conf and lnd.conf and it doesn´t have any tor entries. I'm comparing it with raspibolt's confs...

It's some weird rpc issue, I guess

rootzoll, 2020-07-24:
The Raspiblitz puts the LND behind TOR with parameters given on the systemd service. If you deinstall TOR does the LND connect thru clear net?

rootzoll, 2020-07-24:
While yu are still on TOR what does the "Monitor TOR Service" saying (MAIN MENU > TOR)

talvasconcelos, 2020-07-24:
> The Raspiblitz puts the LND behind TOR with parameters given on the systemd service. If you deinstall TOR does the LND connect thru clear net?

No, it still sptis out an error, ```[lncli] rpc error: code = Unknown desc = dial tcp [2a01:4f8:c0c:7b31::1]:9735: connect: network is unreachable```

talvasconcelos, 2020-07-24:
> While yu are still on TOR what does the "Monitor TOR Service" saying (MAIN MENU > TOR)

```
11:43:47 [NOTICE] Have tried resolving or connecting to address '[scrubbed]' at 3 different places. Giving up.  /home/bitcoin/.nyx/config (see https://nyx.torproject.org/nyxrc.sample for
 │   its options).
 │ 11:43:39 [NOTICE] New control connection opened.
 │ 11:43:27 [NOTICE] New control connection opened from 127.0.0.1. [1 duplicate hidden]
 │ 11:41:15 [NOTICE] Your network connection speed appears to have changed. Resetting timeout to 60s after 18 timeouts and 432 buildtimes.
 │ 11:41:12 [WARN] Intro point 4 at [scrubbed]: circuit is doing handshakes [1 duplicate hidden]
 │ 11:41:12 [WARN] Intro point 3 at [scrubbed]: no circuit
 │ 11:41:12 [WARN] Intro point 2 at [scrubbed]: no circuit [1 duplicate hidden]
 │ 11:41:12 [WARN] Intro point 1 at [scrubbed]: no circuit
 │ 11:41:12 [WARN] Intro point 0 at [scrubbed]: no circuit
 │ 11:41:12 [WARN] Service configured in [EPHEMERAL]:
 │ 11:41:12 [WARN] Intro point 4 at [scrubbed]: no circuit
 │ 11:41:12 [WARN] Intro point 3 at [scrubbed]: circuit is doing handshakes [1 duplicate hidden]
 │ 11:41:12 [WARN] Intro point 1 at [scrubbed]: circuit is doing handshakes [1 duplicate hidden]
 │ 11:41:12 [WARN] Intro point 0 at [scrubbed]: circuit is doing handshakes [1 duplicate hidden]
 │ 11:41:12 [WARN] Service configured in "/mnt/hdd/tor/lndrest8080fallback/":
 │ 11:41:12 [WARN] Intro point 2 at [scrubbed]: circuit is doing handshakes
 │ 11:41:12 [WARN] Service configured in "/mnt/hdd/tor/lndrpc10009fallback/":
 │ 11:41:12 [WARN] Hidden service wcmn2fk6q3h3hbsi exceeded launch limit with 10 intro points in the last 12 seconds. Intro circuit launches are limited to 10 per 300 seconds.
 │ 11:41:12 [NOTICE] Guard whatconfig ($9EB3FD84065E5622A57EFEF14E41A01B5B99A022) is failing more circuits than usual. Most likely this means the Tor network is overloaded. Success counts are
 │   129/185. Use counts are 127/127. 129 circuits completed, 0 were unusable, 0 collapsed, and 3 timed out. For reference, your timeout cutoff is 60 seconds.
 │ 11:41:04 [NOTICE] Bootstrapped 100% (done): Done
 │ 11:41:03 [NOTICE] Bootstrapped 95% (circuit_create): Establishing a Tor circuit
 │ 11:41:03 [NOTICE] Bootstrapped 90% (ap_handshake_done): Handshake finished with a relay to build circuits
 │ 11:41:03 [NOTICE] Bootstrapped 75% (enough_dirinfo): Loaded enough directory info to build circuits
 │ 11:41:03 [NOTICE] Bootstrapped 15% (handshake_done): Handshake with a relay done
 │ 11:41:03 [NOTICE] Bootstrapped 14% (handshake): Handshaking with a relay
 │ 11:41:03 [NOTICE] Bootstrapped 10% (conn_done): Connected to a relay
 │ 11:41:03 [NOTICE] Opened Control listener on /run/tor/control
 │ 11:41:03 [NOTICE] Opening Control listener on /run/tor/control
 │ 11:41:03 [NOTICE] Bootstrapped 5% (conn): Connecting to a relay
 │ 11:41:03 [NOTICE] Signaled readiness to systemd
 │ 11:41:03 [NOTICE] Starting with guard context "default"
 │ 11:41:01 [NOTICE] Bootstrapped 0% (starting): Starting
```

rootzoll, 2020-07-24:
What kind of internet connection are you using, anything special that could interfere with TOR?
@Emzy do you have an idea what is happening here?

talvasconcelos, 2020-07-24:
> What kind of internet connection are you using, anything special that could interfere with TOR?
> @Emzy do you have an idea what is happening here?

I'm using a DSL connection! Bellow is my router firewall settings. Just to be sure, if using TOR, no ports need to be open on the router, right? 

```
BlockAll
 
  |   | Use this Security Level to block all traffic from and to the Internet. Game and Application sharing is not allowed by the firewall. |  
 
  |   | Standard
 
  |   | Use this Security Level to allow all outgoing connections and block all incoming traffic. Game and Application sharing is allowed by the firewall. |  
 
  |   | Disabled
 
  |   | Disable the firewall. All traffic is allowed to pass through your gateway. Game and Application sharing is allowed by the firewall.
```

I can try connecting through 4G by sharing internet to the PI from my phone, see if it's maybe some router fuck up. How do I configure the Blitz to connect from wlan?

rootzoll, 2020-07-24:
To setup WIFI connection: https://stadicus.github.io/RaspiBolt/raspibolt_20_pi.html#prepare-wifi

Emzy, 2020-07-24:
It seems that the ipv6 address [2a01:4f8:c0c:7b31::1] is simply not reachable via Tor!
I tested it and I only can reach it via clear-net. 
I'm not sure what the ipv6 exit node status on tor is. Best could be to use ipv4 addresses if possible.

talvasconcelos, 2020-07-24:
> It seems that the ipv6 address [2a01:4f8:c0c:7b31::1] is simply not reachable via Tor!
> I tested it and I only can reach it via clear-net.
> I'm not sure what the ipv6 exit node status on tor is. Best could be to use ipv4 addresses if possible.

Can you give a node to connect, that you tested? Even so, lncli getnetworkinfo gets me this:

```
{
    "graph_diameter": 0,
    "avg_out_degree": 0,
    "max_out_degree": 0,
    "num_nodes": 1,
    "num_channels": 0,
    "total_network_capacity": "0",
    "avg_channel_size": 0,
    "min_channel_size": "0",
    "max_channel_size": "0",
    "median_channel_size_sat": "0",
    "num_zombie_chans": "0"
}
```

Emzy, 2020-07-24:
> Can you give a node to connect, that you tested? Even so, lncli getnetworkinfo gets me this:

You can try mine, it's on a hosted machine.

```
0215bea618806194ad88b80de8bc11c6391edb2fa1beeea6174305856596faeee8@62.171.129.32:9735
```

talvasconcelos, 2020-07-24:
Ok, on response i get {} that's the expected response i guess?

Still getnetworkinfo: 
`{
    "graph_diameter": 0,
    "avg_out_degree": 0,
    "max_out_degree": 0,
    "num_nodes": 1,
    "num_channels": 0,
    "total_network_capacity": "0",
    "avg_channel_size": 0,
    "min_channel_size": "0",
    "max_channel_size": "0",
    "median_channel_size_sat": "0",
    "num_zombie_chans": "0"
}`
and getinfo:
`{
    "version": "0.10.0-beta commit=v0.10.0-beta",
    "commit_hash": "1a3194d302f33bb52823297d9d7f75cd37516053",
    "identity_pubkey": "029c607d02e7c32adfb53a37f0a07d1e98de670bded096d56460d56c0ba66b6e3b",
    "alias": "sparkBlitz",
    "color": "#68f442",
    "num_pending_channels": 0,
    "num_active_channels": 0,
    "num_inactive_channels": 0,
    "num_peers": 2,
    "block_height": 640560,
    "block_hash": "000000000000000000021768da14325a8730a9de078e8af2508bfa22d1d38605",
    "best_header_timestamp": "1595589631",
    "synced_to_chain": true,
    "synced_to_graph": false,
    "testnet": false,
    "chains": [
        {
            "chain": "bitcoin",
            "network": "mainnet"
        }
    ],
    "uris": [
        "029c607d02e7c32adfb53a37f0a07d1e98de670bded096d56460d56c0ba66b6e3b@mww6cvetfb24ej6ckl42sqa5imqpyvzkpxwsumioax4uxbd6i4am7zqd.onion:9735"
    ],
    "features": {
        "0": {
            "name": "data-loss-protect",
            "is_required": true,
            "is_known": true
        },
        "5": {
            "name": "upfront-shutdown-script",
            "is_required": false,
            "is_known": true
        },
        "7": {
            "name": "gossip-queries",
            "is_required": false,
            "is_known": true
        },
        "9": {
            "name": "tlv-onion",
            "is_required": false,
            "is_known": true
        },
        "13": {
            "name": "static-remote-key",
            "is_required": false,
            "is_known": true
        },
        "15": {
            "name": "payment-addr",
            "is_required": false,
            "is_known": true
        },
        "17": {
            "name": "multi-path-payments",
            "is_required": false,
            "is_known": true
        }
    }
}`

Emzy, 2020-07-24:
Looks good.
> Still getnetworkinfo
lnd needs to sync first. After that it should show something.

talvasconcelos, 2020-07-24:
> Looks good.
> 
> > Still getnetworkinfo
> > lnd needs to sync first. After that it should show something.

it's syncing now... so, it wouldn't sync without connecting to a peer first? Why wasn't it connecting on boot as bitcoin does? 
weird stuff, well for now it seems to be working, thanks, to @rootzoll and @Emzy for your help!

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-18 by thephizzz