raspiblitz/raspiblitz #981: Lightning Progress stuck from backup

er1x, 2020-01-17:
Restoring LND from lnd-rescue backup. The lightning progress stucks at 93.XX% for days and even goes lower.

I have tried from another backup without success, same problem, different %.

Maybe the backups are corrupted? (I have a lot of power outages in my region). Right now, I can start a new node without problem but would like to recover my funds. 

I have the seed and all the passwords. The node had 2 channels (I know the ID's, looks like they are closed now).

rootzoll, 2020-01-17:
Have you tried all options listed in the FAQ to recover funds?
https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-can-i-recover-my-coins-from-a-failing-raspiblitz

er1x, 2020-01-17:
Didn't noticed #278 and the medium post. I'm going to check them. I have noticed this on the logs, in case it helps:

`2020-01-17 17:24:17.015 [ERR] LNWL: Unable to complete chain rescan: -1: Block not found on disk`

Full [debug] report(https://termbin.com/ldj2)

er1x, 2020-02-11:
I've partially solve this. Looks like blockchain was corrupted and lnd *data* folder was throwing a panic error after start. My steps to solve this:

* Synced a new node in another computer, without pruning.

* Started lnd against this node, and created a wallet with the same seed, and with a recovery window of 250000.

* From a previous lnd-backup, I picked the channel.backup file, and executed this command:

  ```bash
  ./lncli restorechanbackup /Users/er1x/lnd-backups/channel.backup
  ```

After that and some time, funds from one channel appeared (looks like channel was closed).

Only channel left is in *waiting_close_channels* state, and looks like it's stuck there:

```⚡./lncli pendingchannels
{
    "total_limbo_balance": "0",
    "pending_open_channels": [
    ],
    "pending_closing_channels": [
    ],
    "pending_force_closing_channels": [
    ],
    "waiting_close_channels": [
        {
            "channel": {
                "remote_node_pub": "020ca546d600037181b7cbcd094818100d780d32fd9f210e14390e0d10b7ec71fb",
                "channel_point": "e733abdb15647d3fe2592e4975da7919cccd542a58d29ea67e8366df95c0514c:1",
                "capacity": "143847",
                "local_balance": "0",
                "remote_balance": "0",
                "local_chan_reserve_sat": "0",
                "remote_chan_reserve_sat": "0"
            },
            "limbo_balance": "0"
        }
    ]
}
```

closed 2020-02-12 by er1x