raspiblitz/raspiblitz #982: Waiting for bitcoind to get ready ---> Verifying blocks.... !! Please login for more details !!

WesleyLuth, 2020-01-17:
Hi all,

I am a total n00b with linux/raspberry/raspiblitz so bare with me. :)

The message i see on the display is: 
   'Waiting for bitcoind to get ready ---> Verifying blocks.... !! Please login for more details!!'

I have RaspiBlitz 1.1 running. I cannot login with ssh. I get 
   'ssh: connect to host 192.168.178.30 port 22: Connection refused.'

If i ping the IP i do get a good respons (TTL=64).

Some history: the Blitz was offline for like a month because it gave an file reading error. I disconnected the external HD and connected that to my laptop,running live cd Ubuntu. Ubuntu noticed that the drive had some issues and fixed them. Then i reconnected the HD and booted the Blitz. After that the blockchain was updating from like from 94,xyz% to 100%. Then i got the error as listed above.

Could someone give me some pointers? Haven't got a clue what to do now. I already rebooted the Blitz (dirty way: disconnected power). Same problem comes back.

openoms, 2020-01-18:
The ssh keys are on the disk, so if the data on disk is injured (as it provably happened before) you won`t be able to log in either.

The most likely reason for disk/file corruption happening is low power for your HDD (most likely with an RPi3)

You should consider some external power supply for the disk like a powered case, powered USB hub or Y-cable to prevent this happening.

Unfortunately once you sorted out the power might need to resync the blockchain as it appears corrupted if it is endlessly verifying.

WesleyLuth, 2020-01-19:
Thanks for your response. How would i determine if the ssh keys are damaged? Connect the drive to the laptop again and look for them? Do you know where i can find them? And could i fix them?

And will giving the external hd it's own power supply solve the problem (if i am able to fix the SSH keys)?

And is there a possibility to retrieve the bitcoin in the channel(s) the node had other then by getting the blitz up and running again? If so, i can just do that and start from scratch and build a new node (now with own power supply for the hd).

PatrickScheich, 2020-01-19:
> 
> 
> Unfortunately once you sorted out the power might need to resync the blockchain as it appears corrupted if it is endlessly verifying.

I had a quite similar case (corrupted ext4fs on hdd) and can highly recommend to get a beefy Linux PC to do the reindex / resync work. As it easily can take **weeks** on a RPi to handle a 
`bitcoind -reindex`
it is a bit tedious to hook the hdd to another linux system, as the ownership on the hdd refers to that users on the raspiblitz. So do it as `root`, in my case `sudo` has not been enough.
Once the hdd is mounted you can delete the **key** files in the `ssh` and `ssh/ssh` dirs
see [issue #860 ](https://github.com/rootzoll/raspiblitz/issues/860#issuecomment-565781729) which should fix the ssh connection problem

rootzoll, 2020-01-20:
To "repair" a damaged blockchain I recommend to call `repair` or go into the REPAIR menu in the main menu and use `RESET-CHAIN` option. This will delete blockchain data and offers you after that to get it fresh. If you dont have a RaspiBerryPi4 I would recommend the COPY option - if you have a laptop where you can install Bitcoin Core, validate chain and then use it as a source to copy it over to the RaspiBlitz over the local network.

WesleyLuth, 2020-01-24:
Thanks!

Can i somehow (without the raspiblitz actually running) see if there are any channels open? And retrieve my sats? For example by using the cipher seed?

rootzoll, 2020-01-29:
@WesleyLuth  See what channels are open with: 1ml.com and search for your node id.

> And retrieve my sats? For example by using the cipher seed?

Just the cipher seed is only securing your on-chain funds. To recover channel balances check out the static-channel-backup: https://github.com/rootzoll/raspiblitz#backup-for-on-chain---channel-funds

WesleyLuth, 2020-02-15:
[image]
What is there is no 'channel.backup' file in /home/admin/.lnd/data/chain/ ?

rootzoll, 2020-02-17:
The "channel.backup" file should be in "/mnt/hdd/lnd/data/chain/bitcoin/mainnet"

WesleyLuth, 2020-02-17:
[image]
Here?

rootzoll, 2020-02-17:
From the path showed on top you have "mnt/hdd" open - not "/mnt/hdd/lnd/data/chain/bitcoin/mainnet" ... but if "mnt/hdd" is empty, that is a very strange sign ... because thats the place where you should see a lot more files and directories - not just LND. Are you sure you mounted tha HDD correct?

WesleyLuth, 2020-02-21:
This is on the SD card, and indeed the folder /mnt/hdd/ is empty. Should I be looking on the external HDD? I don't have a fold /mnt/hdd there. I do have /bitcoin, /lnd, /lost+found and /ssh there.

rootzoll, 2020-03-02:
You have to look on the external HDD "/mnt/hdd" is just the mount point on the sd card (system) where your HDD content gets mounted to by RaspiBlitz. If you have a "lnd" directory on your HDD thats great ... check if you find a channel backup file unter "lnd/data/chain/bitcoin/mainnet".

nyxnor, 2021-04-18:
pls close this.

closed 2021-04-19 by rootzoll