raspiblitz/raspiblitz #985: Touchscreen Failures

rootzoll, 2020-01-20:
OK - on a fresh install I got the basic touchscreen working and the rotation is correct. So I see the buttons and when I press on it I see shortly the mousepointer at the correct place - but nothing is happening. Also the log file is not producing new logs when pressing buttons

@frennkie can you see anything still to fix to get this working in the logs? What is this: `blitztui.client - WARNING - client:230 - <_InactiveRpcError of RPC that terminated with` see full logs here:

```
** Message: 13:28:46.987: environement.vala:58: Exporting primary_variable
** Message: 13:28:46.987: environement.vala:59: desktop_environnement XDG_CURRENT_DESKTOP
** Message: 13:28:46.987: environement.vala:150: confir_dirs is null, export : /etc/xdg
** Message: 13:28:46.987: environement.vala:184: Exporting XDG_CONFIG_DIRS
** Message: 13:28:46.987: environement.vala:218: custom_data :
** Message: 13:28:46.987: environement.vala:219: data_dirs :/usr/local/share:/usr/share/raspi-ui-overrides:/usr/share:/usr/share/gdm:/var/lib/menu-xdg
** Message: 13:28:46.987: environement.vala:220: data_dirs not null, export : /usr/local/share:/usr/share/raspi-ui-overrides:/usr/share:/usr/share/gdm:/var/lib/menu-xdg
** Message: 13:28:46.987: environement.vala:225: Exporting XDG_DATA_DIRS
** Message: 13:28:46.988: utils.vala:138: Config system location : /etc/xdg/lxsession/LXDE-pi
** Message: 13:28:46.988: utils.vala:144: System system path location : /etc/xdg/lxsession/LXDE-pi/desktop.conf
** Message: 13:28:46.988: utils.vala:148: Final file used : /etc/xdg/lxsession/LXDE-pi/desktop.conf
** Message: 13:28:46.989: utils.vala:39: Config system location : /etc/xdg/lxsession/LXDE-pi
** Message: 13:28:46.989: utils.vala:45: System system path location : /etc/xdg/lxsession/LXDE-pi/desktop.conf

** (lxsession:891): WARNING **: 13:28:46.990: utils.vala:70: No such file or directory
** Message: 13:28:46.990: settings.vala:540: Key file does not have key “iGtk/ColorScheme” in group “GTK”
** Message: 13:28:46.990: settings.vala:540: Key file does not have key “windows_manager/command” in group “Session”
** Message: 13:28:46.990: settings.vala:540: Key file does not have key “panel/command” in group “Session”
** Message: 13:28:46.990: settings.vala:540: Key file does not have key “dock/command” in group “Session”
** Message: 13:28:46.991: settings.vala:540: Key file does not have key “file_manager/command” in group “Session”
** Message: 13:28:46.991: settings.vala:540: Key file does not have key “desktop_manager/command” in group “Session”
** Message: 13:28:46.991: settings.vala:540: Key file does not have key “launcher_manager/autostart” in group “Session”
** Message: 13:28:46.991: settings.vala:540: Key file does not have key “composite_manager/command” in group “Session”
** Message: 13:28:46.991: settings.vala:540: Key file does not have key “im1/command” in group “Session”
** Message: 13:28:46.991: settings.vala:540: Key file does not have key “im2/command” in group “Session”
** Message: 13:28:46.991: settings.vala:540: Key file does not have key “widget1/command” in group “Session”
** Message: 13:28:46.991: settings.vala:540: Key file does not have key “notification/command” in group “Session”
** Message: 13:28:46.991: settings.vala:540: Key file does not have key “keybindings/command” in group “Session”
** Message: 13:28:46.992: settings.vala:540: Key file does not have key “im_manager/command” in group “Session”
** Message: 13:28:46.992: settings.vala:540: Key file does not have key “screensaver/command” in group “Session”
** Message: 13:28:46.992: settings.vala:540: Key file does not have key “power_manager/command” in group “Session”
** Message: 13:28:46.992: settings.vala:540: Key file does not have key “audio_manager/command” in group “Session”
** Message: 13:28:46.992: settings.vala:540: Key file does not have key “workspace_manager/command” in group “Session”
** Message: 13:28:46.992: settings.vala:540: Key file does not have key “screenshot_manager/command” in group “Session”
** Message: 13:28:46.992: settings.vala:540: Key file does not have key “message_manager/command” in group “Session”
** Message: 13:28:46.992: settings.vala:540: Key file does not have key “upgrade_manager/command” in group “Session”
** Message: 13:28:46.993: settings.vala:540: Key file does not have key “updates_manager/command” in group “Session”
** Message: 13:28:46.993: settings.vala:540: Key file does not have key “updates_manager/timeout” in group “Session”
** Message: 13:28:46.993: settings.vala:540: Key file does not have key “crash_manager/command” in group “Session”
** Message: 13:28:46.993: settings.vala:540: Key file does not have key “crash_manager/dev_mode” in group “Session”
** Message: 13:28:46.993: settings.vala:540: Key file does not have key “crash_manager/timeout” in group “Session”
** Message: 13:28:46.993: settings.vala:540: Key file does not have key “upstart_user_session” in group “Session”
** Message: 13:28:46.993: settings.vala:540: Key file does not have key “proxy_manager/http” in group “Session”
** Message: 13:28:46.993: settings.vala:540: Key file does not have key “a11y/command” in group “Session”
** Message: 13:28:46.994: settings.vala:540: Key file does not have key “xrandr/command” in group “Session”
** Message: 13:28:46.994: settings.vala:540: Key file does not have key “network_gui/command” in group “Session”
** Message: 13:28:46.994: settings.vala:540: Key file does not have key “webbrowser/command” in group “Session”
** Message: 13:28:46.994: settings.vala:540: Key file does not have key “email/command” in group “Session”
** Message: 13:28:46.994: settings.vala:540: Key file does not have key “pdf_reader/command” in group “Session”
** Message: 13:28:46.994: settings.vala:540: Key file does not have key “video_player/command” in group “Session”
** Message: 13:28:46.994: settings.vala:540: Key file does not have key “audio_player/command” in group “Session”
** Message: 13:28:46.994: settings.vala:540: Key file does not have key “images_display/command” in group “Session”
** Message: 13:28:46.995: settings.vala:540: Key file does not have key “text_editor/command” in group “Session”
** Message: 13:28:46.995: settings.vala:540: Key file does not have key “archive/command” in group “Session”
** Message: 13:28:46.995: settings.vala:540: Key file does not have key “charmap/command” in group “Session”
** Message: 13:28:46.995: settings.vala:540: Key file does not have key “calculator/command” in group “Session”
** Message: 13:28:46.995: settings.vala:540: Key file does not have key “spreadsheet/command” in group “Session”
** Message: 13:28:46.995: settings.vala:540: Key file does not have key “bittorent/command” in group “Session”
** Message: 13:28:46.995: settings.vala:540: Key file does not have key “document/command” in group “Session”
** Message: 13:28:46.995: settings.vala:540: Key file does not have key “webcam/command” in group “Session”
** Message: 13:28:46.995: settings.vala:540: Key file does not have key “burn/command” in group “Session”
** Message: 13:28:46.996: settings.vala:540: Key file does not have key “notes/command” in group “Session”
** Message: 13:28:46.996: settings.vala:540: Key file does not have key “disk_utility/command” in group “Session”
** Message: 13:28:46.996: settings.vala:540: Key file does not have key “tasks/command” in group “Session”
** Message: 13:28:46.996: settings.vala:540: Key file does not have group “Keymap”
[... 48 lines cut ...]
** Message: 13:28:47.209: options.vala:107: Create build-in Clipboard
** Message: 13:28:47.213: main.vala:441: Check keymap_mode (null)
** Message: 13:28:47.239: app.vala:76: Launching /usr/bin/ssh-agent
SSH_AUTH_SOCK=/tmp/ssh-6aHsauDNuo1w/agent.964; export SSH_AUTH_SOCK;
SSH_AGENT_PID=969; export SSH_AGENT_PID;
echo Agent pid 969;
** Message: 13:28:47.343: app.vala:130: xset exit with this type of exit: 0
** Message: 13:28:47.343: app.vala:139: Exit normal, don't reload
** Message: 13:28:47.343: app.vala:130: xset exit with this type of exit: 0
** Message: 13:28:47.344: app.vala:139: Exit normal, don't reload
** Message: 13:28:47.344: app.vala:130: xset exit with this type of exit: 0
** Message: 13:28:47.344: app.vala:139: Exit normal, don't reload
** Message: 13:28:47.344: app.vala:130: /usr/bin/ssh-agent exit with this type of exit: 0
2020-01-20 13:28:47,590 - root - INFO - __init__:9 - Starting BlitzTUI v0.44.0
Openbox-Message: Unable to find a valid menu file "/usr/share/lxde/openbox/menu.xml"
[... cut ...]
	debug_error_string = "{"created":"@1579526932.156833863","description":"Failed to pick subchannel","file":"src/core/ext/filters/client_channel/client_channel.cc","file_line":3941,"referenced_errors":[{"created":"@1579526932.156819677","description":"failed to connect to all addresses","file":"src/core/ext/filters/client_channel/lb_policy/pick_first/pick_first.cc","file_line":393,"grpc_status":14}]}"
>
2020-01-20 13:28:52,580 - root - INFO - main:620 - starting beat
2020-01-20 13:28:52,587 - blitztui.file_watcher - INFO - file_watcher:29 - starting config watcher
```

rootzoll, 2020-01-21:
Another Touchscreen problem to fix for v1.4: On updating a RaspiBlitz that has touchscreen activated, there was a blackscreen+mouse after first reboot during recovery. So the user cannot see the "FINAL LOGIN with password: raspiblitz" info. I logged in anyway and set the password A - then after the following reboot everthing was good and the touchscreen worked. 

TODO: Check recover logs on RaspiBlitz with activated touchscreen to see why that blackscreen+mouse did occured.

rootzoll, 2020-01-21:
OK I got rid of that blackscreen during final recovery by letting the BlitzTUI wait for a ready flag in the info file before checking the LND status. But because on my test machine the buttons are still not working (see report above) I cannot say if this is a correct fix @frennkie are my changes OK or do we need to do this another way?

rootzoll, 2020-01-21:
OK I added a exception catching around the LND RPC calling ... now there is no blackscreen on the final recovery screen. @frennkie but still my buttins dont work (see above) . how should I contunie testing/fixing?

Juized, 2020-01-22:
@rootzoll did you try running  `sudo -u pi DISPLAY=:0.0 LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libatomic.so.1.2.0 /home/admin/python3-env-lnd/bin/python3 /home/admin/raspiblitz/home.admin/BlitzTUI/blitztui/main.py` 
I had similar issues but after running this command I was able to get the buttons to work, although when I cancel the process it stops working again.

rootzoll, 2020-01-22:
@Juized yeah correct: If I run that command the Buttins are working as long as I keep the command running. @frennkie so it seems the man.py is kind of exiting before because I cannot see any logs adding when I press buttons ... until I start the above command again. How can we assure that the main.py keeps running or is not exiting?

rootzoll, 2020-01-22:
This are the logs when I start up the node with touchscreen and the Buttons are inactive:
```
2020-01-22 16:16:38,681 - root - INFO - main:180 - init lnd.conf
2020-01-22 16:16:38,699 - root - INFO - main:189 - init raspiblitz.conf
2020-01-22 16:16:38,702 - root - INFO - main:198 - init raspiblitz.info
2020-01-22 16:16:38,702 - blitztui.config - INFO - config:114 - loading RaspiBlitzInfo config from file: /home/admin/raspiblitz.info
2020-01-22 16:16:38,703 - blitztui.config - INFO - config:128 - status --> booting
2020-01-22 16:16:38,839 - blitztui.client - WARNING - client:229 - an unknown RpcError occurred
2020-01-22 16:16:38,840 - blitztui.client - WARNING - client:230 - <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "failed to connect to all addresses"
	debug_error_string = "{"created":"@1579709798.839245656","description":"Failed to pick subchannel","file":"src/core/ext/filters/client_channel/client_channel.cc","file_line":3941,"referenced_errors":[{"created":"@1579709798.839240082","description":"failed to connect to all addresses","file":"src/core/ext/filters/client_channel/lb_policy/pick_first/pick_first.cc","file_line":393,"grpc_status":14}]}"
>
2020-01-22 16:16:39,010 - root - INFO - main:632 - starting beat ..
2020-01-22 16:16:39,011 - blitztui.file_watcher - INFO - file_watcher:29 - starting config watcher
2020-01-22 16:16:39,142 - root - INFO - main:154 - /usr/bin/xterm: cannot load font "-Misc-Fixed-medium-R-*-*-13-120-75-75-C-120-ISO10646-1"
```

frennkie, 2020-01-25:
I can reproduce this. But I don't know what is causing it. 

Running this restores the functionality:

```
sudo init 3 ; sleep 2 ; sudo init 5
```

I'll try to add some more debugging/logging to find the cause.

rootzoll, 2020-01-26:
OK the restart monitoring is a good enough workaround for the v1.4 to keep the touchscreen working. So I move this issue to v1.5 to investigate further why there is at minumum 1 restart of the BlitzTUI needed to keep the touchscreen working.

rootzoll, 2020-01-29:
New Blackscreen Report: I was switching to TESTNET that worked. Then I did a update/recover test while on testnet and I ended up with a touchscreen-blackscreen.

@frennkie any idea where this error comes from?
```
Traceback (most recent call last):
  File "/home/admin/python3-env-lnd/bin/blitz-tui", line 6, in <module>
    from blitztui.main import main
  File "/home/admin/python3-env-lnd/lib/python3.7/site-packages/blitztui/main.py", line 17, in <module>
    from PyQt5.QtCore import Qt, QProcess, QThread, pyqtSignal, QCoreApplication, QTimer, QEventLoop
ModuleNotFoundError: No module named 'PyQt5'
** Message: 03:47:30.812: app.vala:130: sh exit with this type of exit: 256
** Message: 03:47:30.813: app.vala:145: Exit normal, don't reload
Openbox-Message: Unable to find a valid menu file "/usr/share/lxde/openbox/menu.xml"
```

rootzoll, 2020-01-29:
I got touch screen back working with: Switch to mainnet. Switch touchscreen off. Install Touchscreen again. I also tested: The problem does not occur when I do a recovery on mainnet.

frennkie, 2020-02-01:
> ModuleNotFoundError: No module named 'PyQt5' 

Hm.. switching from MAINNET to TESTNET does not modify `/home/admin/python3-env-lnd`, or does it? 

The Python bindings for Qt5 (`PyQt5`) are installed system-wide using `sudo apt-get install -y unclutter xterm python3-pyqt5` in `blitz.touchscreen.sh`. `"build_sdcard.sh"` setups `/home/admin/python3-env-lnd` with `--site-package` and therefore uses a python libraries that are available to the system python (`dpkg -l | grep -i python3-`).

ChuckNorrison, 2020-09-26:
unclutter is using nearly 100% cpu 24/7 after some runtime. in that circumstance the lcd is black nd the mouse cursor wants to hide with the unclutter. 

have to uninstall lcd touchscreen to get it smooth again

rootzoll, 2021-04-17:
I also have test machins were the touchscreen is crashing contantly with
`Openbox-Message: Unable to find a valid menu file "/usr/share/lxde/openbox/menu.xml"`
@frennkie 

This is someting to fix/check for the v1.7.1 to determine if the touchscreen needs bigger refactoring to get more stable.

ChuckNorrison, 2021-04-19:
No Touchscreen issues here on 1.7.0 RC2

Touchscreen is disabled in Blitz node settings, LCD Rotate is enabled. This works great for me since 1.6.3.

closed 2021-08-24 by rootzoll