raspiblitz/raspiblitz #993: New Setup with Touchscreen & TOR-Browser

rootzoll, 2020-01-23:
I am adding this idea scribble for discussion - would be great to aim for this with the v1.5 version:

[image]

rootzoll, 2020-01-23:
To align the HTML-Website-Setup with the SSH-Setup the setup process of the RaspiBlitz would need a small refactoring.

The idea is to collect all user setup settings into one file up front - basically producing a raspiblitz.conf file and then the RaspiBlitz just runs thru its recovery/update process and installs everything.

This raspiblitz.conf gets either produced by that HTML-Website you setup thru the TOR-Browser or the thru Dialogs on the command line thru SSH. Some info (like the seed words) needs to get stored during the process and when the setup/recovery is finished it will get displayed to the user and deleted/shredded from the Raspiblitz.

rootzoll, 2020-01-23:
And dont worry that WIFI option is just there as possible idea for the future. The button can be empty in the first version.

openoms, 2020-01-23:
I really like the idea of using the Tor browser as default. 

There could be also an option (maybe instead of the wifi) to show a QR with a TorV3 address and pasted to the mobile Tor Browser for initialization.

The wifi process is tricky especially if using other boards. Some need extra drivers or lacking a wifi module, but they can be still set up securely via HDMI through Tor or SSH. Would need to test to show the status screen on HDMI for those.

Storing the seed words temporarily does increase the attack surface so best would be to avoid if possible by default. When setting up through SSH the seed could come straight from the LND terminal like now and only be stored ( temporarily ) in case of a HTML setup.

rootzoll, 2020-01-23:
> There could be also an option (maybe instead of the wifi) to show a QR with a TorV3 address and pasted to the mobile Tor Browser for initialization.

On the TOR screen behind the written onion-address there could ne a small "show as QR-Code"-Button (QR-Code icon) that will show the address as QR code .. so need to give it an extra button on the side.

> The wifi process is tricky especially if using other boards. Some need extra drivers or lacking a wifi module, but they can be still set up securely via HDMI through Tor or SSH. Would need to test to show the status screen on HDMI for those.

Forget for the WiFi-Option for now - having TOR & SSH is the next goal. For non-LCD setups we can leave it as now - that you find the IP of your raspiblitz manually and login with the standard password `raspiblitz` (like the the RaspberryPi does) - I dont think a HDMI info is needed.

> Storing the seed words temporarily does increase the attack surface so best would be to avoid if possible by default. When setting up through SSH the seed could come straight from the LND terminal like now and only be stored ( temporarily ) in case of a HTML setup.

Maybe to avoid having the the seed words stored on disk they can be stored in memory. That means that the creation of the LND wallet during setup has to be after the final reboot. We have to see what works best here.

frennkie, 2020-04-04:
@rootzoll regarding vanity addresses https://opensource.com/article/19/8/how-create-vanity-tor-onion-address - according to the table I would say 4 characters is fine.. 5 ("1 minute") might be acceptable... would give "blitz".. :-D

rootzoll, 2020-05-27:
With having @frennkie prepared the web interface basics for v1.6 release ... working then for v1.7 on coming closer to the goal of this issue seems possible ... so moving to v1.7 release.

rootzoll, 2020-05-27:
BTW on last discussion we were in favor of not neededing an "extra password" when connecting thru TOR to a RaspiBlitz waiting for setup - the address is "code" enough and makes sure that you really talk to this node. But IF a second user opens that setup page (can be simulated with making a request from a second TOR browser) the page should say "setup in progress - if this is not you then power down RaspiBlitz and make a fresh sd card".

rootzoll, 2020-08-13:
About a future WIFI option on that start screen see: https://github.com/rootzoll/raspiblitz/issues/1432

nyxnor, 2021-04-18:
I dont understand the reason this was moved to 1.8, maybe the webUI. If this is it, I would recommend addind a QR for the onion address, so a user can scan it and open with Tor Browser on the android device.

rootzoll, 2021-04-19:
Yes the v1.8 will be the point to refactor touchscreen and its part in the setup process playing into the WebUI for setup. A QR code for a Torv3 could be an idea .. will try to collect& sort all ideas once we get started towards v1.8