raspiblitz/raspiblitz #996: lnd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT

thephizzz, 2020-01-23:
My Raspi had been up and operating for a for weeks now.
Unfortunately, I shut the Raspiblitz down and was copying the blockchain to another HDD. 
I stopped before it was done and rebooted.
But when i re-booted, lnd does not start and i get this error in the logs:
 lnd.service: Main process exited, code=exited, status=2/INVALIDARGUMENT

here is a copy of my log: https://termbin.com/fkg5

any suggestions are appreciated.
Thanks

thephizzz, 2020-01-23:
When I try to run the lnd command: lnd --tor.active --tor.v2 --listen=127.0.0.1:9735 (which is giving the error in the logs)
I get this error: 
[ERR] LTND: Unable to load TLS credentials: open /home/admin/.lnd/tls.key: no such file or directory

I thought that file was used from /mnt/hdd/lnd directory where is does exist.

thephizzz, 2020-01-24:
Update: I've reburned the img file to a new micro SD.
Same issue. Lnd won't start with same error and log file as above.
I have funds in a channel that i hope to recover. I been down now for a day. Will the channel close automatically after time? and will it return my funds?
Of course, I'm assuming I can restore my lnd wallet from the seed.

openoms, 2020-01-24:
The LND 0.9.0 release is available now.

You can reset your TLS cert with the command:
`./config.scripts/lnd.newtlscert.sh`

openoms, 2020-01-24:
This is suspicious if some corruption in your LND, but don't know the details. 
```
Jan 23 17:48:36 raspivoice systemd[1]: Started LND Lightning Daemon.
Jan 23 17:48:36 raspivoice lnd[32471]: panic: freepages: failed to get all reachable pages (page 12435: multiple references)
Jan 23 17:48:36 raspivoice lnd[32471]: goroutine 54 [running]:
Jan 23 17:48:36 raspivoice lnd[32471]: github.com/coreos/bbolt.(*DB).freepages.func2(0x3076b00)
Jan 23 17:48:36 raspivoice lnd[32471]:         github.com/coreos/bbolt@v1.3.3/db.go:1003 +0xcc
Jan 23 17:48:36 raspivoice lnd[32471]: created by github.com/coreos/bbolt.(*DB).freepages
Jan 23 17:48:36 raspivoice lnd[32471]:         github.com/coreos/bbolt@v1.3.3/db.go:1001 +0x108
```

Might worth to ask in the LND Dev Slack.

Also try updating to the latest release with:
https://github.com/openoms/lightning-node-management/tree/master/lnd.updates#update-lnd-to-v090-beta

thephizzz, 2020-01-24:
Thanks. will try the TLS script.

I thought those lines looked a little strange. I will upgrade to 0.9.

thephizzz, 2020-01-25:
After updating to 0.90 I get this on reboot:

 

Installed lnd version 0.9.0-beta commit=v0.9.0-beta

Input wallet password:

[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"

admin@raspivoice:~ $

 

Then after re-boot, bitcoin doesn’t start:

 

*** Check Basic Config ***

*** Check bitcoin Running ***

error: Could not connect to the server 127.0.0.1:8332

 

Make sure the bitcoind server is running and that you are connecting to the correct RPC port.

*** Check Basic Config ***

*** Check bitcoin Running ***

*** Check bitcoin Responding *** (can take a while)

(1/33) still waiting ..

(2/33) still waiting ..

(3/33) still waiting ..

 

I have port 10009 forwarded and this node had been running fine for weeks before all this happened.

 

Thanks for any advice.

 

Jim

thephizzz, 2020-01-26:
UPDATE:
After re-imaging the SD and re-installing Raspiblitz/lnd/Tor I was up and running again but with no funds.
I had used a channel-backup file and a lnd-rescue file, but I received errors every time i re-booted and no funds were restored.
Yesterday - on advice from openoms (above) I updated to lnd 0.9.0.
Then today after re-booting, the channel-backup was accepted and all my funds re-appeared.
So my backup must have been done with lnd 0.9.0, but the re-install defaulted to lnd 0.7.1 and therefore couldn't recognize the files.
Thanks for the help and I hope this can help someone else.

closed 2020-01-26 by thephizzz