raspiblitz/raspiblitz #998: Tor [WARN] fixes

openoms, 2020-01-24:
Ths message in Nyx (`sudo -u bitcoin nyx`) is a returning discussion point:
```
 [WARN] Before Tor can create a control socket in "/run/tor/control", the directory "/run/tor" needs to exist, and to be accessible only by the user and group account that is running Tor. (OnOn
some Unix systems, anybody who can list a socket can connect to it, so Tor is being careful.) [406 duplicates hidden]n]
 [WARN] /run/tor is not owned by this user (bitcoin, 1002) but by debian-tor (108). Perhaps you are running Tor as the wrong user? [405 duplicates hidden]n]
```

It can be fixed by modifying the torrc (`sudo nano /etc/tor/torrc`)
to use the user debian-tor 
```
User debian-tor
```
and making the /mnt/hdd/tor dir to be owned by the user  `debian-tor`
`sudo chown -R debian-tor:debian:tor /mnt/hdd/tor`

```
 [WARN] Your log may contain sensitive information - you disabled SafeLogging, and you're logging more than "notice". Don't log unless it serves an important reason.
Overwrite the log afterwards.
```
This is fixed in torrc:
change SafeLogging 0 to:
```
SafeLogging 0 to 1
```
[WARN] Skipping obsolete configuration option "PortForwarding".
In torrc delete 
```
PortForwarding 1
```

All about torrc in: https://helpmanual.io/man5/torrc/

openoms, 2020-01-24:
Also this line: https://github.com/rootzoll/raspiblitz/blob/v1.4/home.admin/config.scripts/internet.tor.sh#L315  does nothing a  the directory changes back to be owned by debian-tor on Tor restart.

See the test:
```
admin@raspberrypi:~ $ ls -la /var/run/tor/
total 0
drwxr-sr-x  2 debian-tor debian-tor  60 Jan 24 21:40 .
drwxr-xr-x 28 root       root       880 Jan 22 13:35 ..
srw-rw----  1 debian-tor debian-tor   0 Jan 24 21:40 control

admin@raspberrypi:~ $ sudo chown -R bitcoin:bitcoin /var/run/tor/

admin@raspberrypi:~ $ ls -la /var/run/tor/
total 0
drwxr-sr-x  2 bitcoin bitcoin  60 Jan 24 21:40 .
drwxr-xr-x 28 root    root    880 Jan 22 13:35 ..
srw-rw----  1 bitcoin bitcoin   0 Jan 24 21:40 control

admin@raspberrypi:~ $ sudo systemctl restart tor

admin@raspberrypi:~ $ ls -la /var/run/tor/
total 0
drwxr-sr-x  2 debian-tor debian-tor  60 Jan 24 21:41 .
drwxr-xr-x 28 root       root       880 Jan 22 13:35 ..
srw-rw----  1 debian-tor debian-tor   0 Jan 24 21:41 control
```

openoms, 2020-01-24:
Two more things needed for LND to be able to access the auth cookie:
`sudo chmod -R 740 /mnt/hdd/tor`
 and:
`sudo usermod -a -G debian-tor bitcoin`
as in: https://stadicus.github.io/RaspiBolt/raspibolt_22_privacy.html#installing-tor

openoms, 2020-01-24:
Stuck with this because LND was not able to access the Tor auth cookie.
Permission are going strange as even if `bitcoin` is added to the `debian-tor` group:
```
$ cat /etc/group | grep debian-tor
debian-tor:x:116:bitcoin
admin@raspberrypi:~ $ 
```
```
$ sudo -u bitcoin ls -la /mnt/hdd/

drwxr----- 13 debian-tor debian-tor       4096 Jan 24 21:38 tor

$ sudo -u bitcoin ls -la /mnt/hdd/tor

ls: cannot access '/mnt/hdd/tor/..': Permission denied

d????????? ? ? ? ?            ? sys
```

openoms, 2020-02-26:
The line : 
```
ExecStartPre=/usr/bin/install -Z -m 02755 -o debian-tor -g debian-tor -d /run/tor

```
in `/lib/systemd/system/tor@default.service` changes back the ownership of `/run/tor` to `debian-tor` on every start of Tor.

Changing `debian-tor` to `bitcoin` solved the user related warn:
```
ExecStartPre=/usr/bin/install -Z -m 02755 -o bitcoin -g bitcoin -d /run/tor
```

Running my nodes with this change now, will do a PR if no problems arise.

rootzoll, 2020-03-16:
@openoms can we merge this into v1.5 (before hackday) or better push to v1.6

openoms, 2020-03-17:
An Odroid HC1 node is running with these settings since.
Will run another round of test on the RPi4.

Before the fix:
`sudo -u bitcoin nyx`
[image]

Applying:
```
sudo sed -i "s/debian-tor/bitcoin/g" /lib/systemd/system/tor@default.service
sudo sed -i "s/SafeLogging 0/SafeLogging 1/g" /etc/tor/torrc
sudo sed -i "s/PortForwarding 1//g" /etc/tor/torrc
sudo systemctl restart tor
```

openoms, 2020-05-10:
LND now can use the `HASHED PASSWORD` auth method to Tor: https://github.com/lightningnetwork/lnd/blob/master/docs/configuring_tor.md#authentication  
so does not need access to the Tor cookie file.   
Tor and and lnd can be all run with a separate user.

openoms, 2020-05-13:
Refactored the Tor settings utilising the pattern in lndcreds. Testing on 1.5 so far without errors (and warns). Tor is run with the deafult `debian-tor`  user:
```
# create torcontrol group
sudo /usr/sbin/groupadd --force --gid 9800 torcontrol

# FUNCTIONS
function tor_set_perms() {
  local file_name=${1}  # the file name (e.g. admin.macaroon)
  local group_name=${2} # the unix group name (e.g. lndadmin)

  sudo /bin/chown --silent admin:"${group_name}" /mnt/hdd/tor/sys/"${file_name}"
  sudo /bin/chmod --silent 640 /mnt/hdd/tor/sys/"${file_name}"
}

# make sure bitcoin is member of tor control
sudo /usr/sbin/usermod --append --groups torcontrol bitcoin

#revert config to default
sudo chown -R debian-tor:debian-tor /mnt/hdd/tor
sudo sed -i "s/User bitcoin/User debian-tor/g" /etc/tor/torrc
sudo sed -i "s/PortForwarding 1//g" /etc/tor/torrc
sudo chmod -R 700 /mnt/hdd/tor
sudo chown -R debian-tor:debian-tor /mnt/hdd/tor 

echo "- ensure unix ownerships and permissions for the Tor control cookie"
tor_set_perms control_auth_cookie torcontrol
echo "OK DONE"

sudo systemctl restart tor
```

rootzoll, 2020-05-27:
One aim for the v1.7 release should be to go TOR by default - I think thats a good time to take a look at those warning and optimizations.

openoms, 2020-05-28:
Agree with Tor as default and it will be also beneficial to ship the Tor binary with the SDcard image (the insatllation could part of the prepare step of the script).

Apparently the torproject.org website is blocked in some jurisdictions (so cannot download), but the Tor network is still usable.

frennkie, 2020-08-02:
> sudo sed -i "s/debian-tor/bitcoin/g" /lib/systemd/system/tor@default.service

@openoms do you expect this to be override e.g. if somebody runs `apt upgrade` (and a newer package of tor is installed)?

Also: This fix for the logging message is currently NOT in v1.6, right?

openoms, 2020-08-02:
> > sudo sed -i "s/debian-tor/bitcoin/g" /lib/systemd/system/tor@default.service
> 
> @openoms do you expect this to be override e.g. if somebody runs `apt upgrade` (and a newer package of tor is installed)?
> 

the fact that we are rynning Tor with the `bitcoin` user is problematic. It breaks `apt upgrade` as well currently. Should use `debian-tor` instead, but permissions need to be sorted.

> Also: This fix for the logging message is currently NOT in v1.6, right?
> 
> Related issue: #402
> 
> The fix in https://github.com/rootzoll/raspiblitz/issues/998#issuecomment-599960455 works for me.

Great, we could include the log fixes. That is not controversial.

frennkie, 2020-08-02:
As there are several things in this issue let's try to "sort" this a bit

 1. Minor improvements for the settings (SafeLogging, PortForwarding)
 2. Cookie Auth
 3. The general question whether to run Tor as `bitcoin` or `debian-tor`

For 1) I agree that this is non-controversial - we should create a PR and fix it (either for 1.6 or 1.7).

For 2) - not really urgent right now, agreed?

For 3) This currently leads to the "[WARN]" messages. 

A workaround is the `sudo sed -i "s/debian-tor/bitcoin/g" /lib/systemd/system/tor@default.service` which has the downside of being overwritten on apt upgrade.

What permission issue are there if we run Tor with `debian-tor`? Does this only affect `bitcoind` ? Or are others apps affected?

frennkie, 2020-08-08:
Idea: Include check for `debian-tor` in `/lib/systemd/system/tor@default.service` in `internet.tor.sh` which is then called every hour(?!) by  `_background.sh`.

nolith, 2020-09-05:
> 
>  2. HASHED PASSWORD auth
> 
>  3. The general question whether to run Tor as `bitcoin` or `debian-tor`
> 
> 
> For 2) - not really urgent right now, agreed?
> 
> 

Do we use tor control socket only from bitcoind and lnd, right?

Hashed password will allow the resolution for both 2 and 3 as we can run tor with the default user not messing with apt-get update. 

I'm quite sure I have this configuration on my Debian based testnet setup (not RaspiBlitz)

rootzoll, 2020-09-08:
Note to self: If not merged earlier ... for v1.7 take a look at: https://github.com/rootzoll/raspiblitz/pull/1506

frennkie, 2020-09-09:
> Do we use tor control socket only from bitcoind and lnd, right?

@openoms Can you confirm that those two are the only reason we are running `Tor` as user `bitcoin`...?

With the changes in PR and things in the comments there we no longer have a problem with `apt-get update`.

openoms, 2020-09-09:
yes, only bitcoind and lnd are running with the bitcoin user and need access to the control port

rootzoll, 2020-09-28:
Most of this should be fixed by #1571 merged into v1.6.1 ... but check again when working on v1.7

openoms, 2020-10-14:
Can confirm this works.

This got solved too:
> One more things needed for LND to be able to access the auth cookie:
> 
> `sudo usermod -a -G debian-tor bitcoin`
> 
> as in: https://stadicus.github.io/RaspiBolt/raspibolt_22_privacy.html#installing-tor

> Stuck with this because LND was not able to access the Tor auth cookie.
> Permission are going strange as even if `bitcoin` is added to the `debian-tor` group:
>
> ```
> $ cat /etc/group | grep debian-tor [...]
TIL that +x is needed as well to be allowed to move into a directory: https://github.com/JoinMarket-Org/joinmarket-clientserver/issues/703#issuecomment-708057776

Getting rid of this line from `torrc` solved the issue:
```
CookieAuthFile /mnt/hdd/tor/sys/control_auth_cookie
```

The auth cookie now is created in:  
`/var/run/tor/control.authcookie`

Usage of the auth cookie is described here: 
https://github.com/JoinMarket-Org/joinmarket-clientserver/blob/master/docs/PAYJOIN.md#configuring-tor-to-setup-a-hidden-service

Fixes included in this PR: https://github.com/rootzoll/raspiblitz/pull/1666/files#diff-ead50f54f76f0367e510770c5b88ea64c56fcd48350a7035ae1d5f44710d22ceR133

openoms, 2021-03-18:
This is no longer relevant since moved to running Tor with the `debian-tor` user.

openoms, 2021-03-23:
We can reprecate the Hashed Tor Control password as it is already tested with JoinMarket. Now works with lnd and bitcoind too.
https://github.com/lightningnetwork/lnd/blob/master/docs/configuring_tor.md#authentication

rootzoll, 2021-03-23:
OK merged the new authentification scheme ... now on final testing.

closed 2021-04-18 by rootzoll