raspiblitz/raspiblitz #1000: Keysend  -- spontanous payments

openoms, 2020-01-25:
See: https://twitter.com/joostjgr/status/1220294554307497986

Tested successfully on a node with lnd v0.9.0-beta by editing the lnd.service file:  

`sudo nano /etc/systemd/system/lnd.service`

add `--accept-keysend```:
```
ExecStart=/usr/local/bin/lnd --accept-keysend --tor.active --tor.streamisolation --tor.v3 --listen=127.0.0.1:9735
```

restart:  
` sudo systemctl restart lnd`  
`lncli unlock`  

Send payments to it with the syntax:
`lncli sendpayment --keysend -d 028a2cb8d51e44d7d7e108c2e80a98cc069145e05a6d2025cf554bd8866fe32993 -a 1 --data 7629168=$(echo -n "sent from a RaspiBlitz" | xxd -pu -c 10000)`

Output with a direct channel:
```
$ lncli sendpayment --keysend -d 028a2cb8d51e44d7d7e108c2e80a98cc069145e05a6d2025cf554bd8866fe32993 -a 1 --data 7629168=$(echo -n "test" | xxd -pu -c 10000)
{
    "payment_error": "",
    "payment_preimage": "8228248114408ad4ad05147ce66b337efa38e54d0eea543e048c06d24f017841",
    "payment_route": {
        "total_time_lock": 614511,
        "total_fees": "0",
        "total_amt": "1",
        "hops": [
            {
                "chan_id": "675528949131837440",
                "chan_capacity": "1999846",
                "amt_to_forward": "1",
                "fee": "0",
                "expiry": 614511,
                "amt_to_forward_msat": "1000",
                "fee_msat": "0",
                "pub_key": "028a2cb8d51e44d7d7e108c2e80a98cc069145e05a6d2025cf554bd8866fe32993",
                "tlv_payload": true,
                "mpp_record": null,
                "custom_records": {
                    "7629168": "74657374",
                    "5482373484": "8228248114408ad4ad05147ce66b337efa38e54d0eea543e048c06d24f017841"
                }
            }
        ],
        "total_fees_msat": "0",
        "total_amt_msat": "1000"
    },
    "payment_hash": "9ec90832d05af2d1488ff06067297509316bf22e6714cf8eff92e35b2bb0865b"
}
```

Shall we enable this by default? Not sure of any drawbacks.

ascisco, 2020-01-25:
I can receive direct keysend payments but im getting the following error when trying to send them:

`[lncli] IncorrectOrUnknownPaymentDetails(amt=10000 mSAT, height=614538)@1`

openoms, 2020-01-26:
> I can receive direct keysend payments but im getting the following error when trying to send them:

Everything needs to be fulfilled what is needed for a successful LN payment + the receiver needs to be paid through a public channel as keysend has no routehints available (which are provided by nodes behind private channels in the invoice, for example mobile wallets) 

* Try paying an invoice to that node. Does that succeed? 
* Try keysend through a direct channel.

ascisco, 2020-01-26:
> Everything needs to be fulfilled what is needed for a successful LN payment + the receiver needs to be paid through a public channel as keysend has no routehints available (which are provided by [...]

I have a direct channel opened already with the person im trying to send to. I was able to do a keysend with others who I do not have a channel open with though.

fusion44, 2020-01-30:
I think this error occurs usually when the receiving peer doesn't run LND with --accept-keysend.

rootzoll, 2020-02-09:
We should add on option to activate keysend in the next 1.5 release.

rootzoll, 2020-03-21:
OK added for v1.5 RC1 release. Needs still testing.

rootzoll, 2020-03-22:
I actived keysend on both my testing RaspiBlitzes (both lnd 0.9.2) but when I try to send with

`lncli sendpayment -d 0244f87803308d569992458ce9d755a10f0609a6febbd81c01490722f4409ee6fa -a 10 --keysend`

I just get an error "destination hop doesn't understand new TLV payloads" - @openoms how can I check that keysend is really activated on the other node?

openoms, 2020-03-23:
@rootzoll the only way I know to query about the supported features is:
`lncli getnodeinfo 0244f87803308d569992458ce9d755a10f0609a6febbd81c01490722f4409ee6fa`

Unfortunately that does not return much for that node, neither can I see it's IP address.

I don't see the keysend specific feature for my nodes, but they advertise the same to the network what is given on `lncli getinfo`.

openoms, 2020-03-23:
Joost here suggest just to try and see if it works for now: https://github.com/lightningnetwork/lnd/pull/3795#discussion_r356420971

rootzoll, 2020-03-28:
- Test sendkey with RaspiBlitz
- Write a sendkey script, that tests all other nodes if supporting sendkey and donates all of them 10 sats 
- Connect RaspiBlitz with "SenMany" App - see
https://github.com/fusion44/sendmany

@fusion44 can you mentor this sprint issue over the Lightning Hackday (first April Weekend)

fusion44, 2020-03-29:
@rootzoll Sure, I'd gladly help out on that. Is there a Slack channel or similar for it?

One more reason to fix my RaspiBlitz now :)

rootzoll, 2020-03-29:
We have a MatterMost (OpenSource Slack) we will use. Will post more details the next days.

rootzoll, 2020-04-14:
Outcome of Hacksprint:
* Made a keysend chat over a RaspiBlitz with keysend activated (3 hops on LN)
* Add SendMany App to the RaspiBlitz for install

Looks good - closing issue for v1.5 release :)

closed 2020-04-14 by rootzoll