raspiblitz/raspiblitz #1001: ZAP over TOR Support

rootzoll, 2020-01-25:
With the upgrade to LND 0.9.0 - somhow the connection over TOR is not working anymore.
https://github.com/LN-Zap/zap-iOS/issues/333

For a manual try follow: https://github.com/openoms/bitcoin-tutorials/blob/master/Zap_to_RaspiBlitz_through_Tor.md

theotherside, 2020-01-25:
I just followed the commands suggested in the above link. In the latest Zap from ios app store on 20200125 and I got the same exact error shown in https://github.com/LN-Zap/zap-iOS/issues/333

rootzoll, 2020-01-28:
Moving to v1.5 release.

lordgivemestrength, 2020-02-08:
Would this issue be the reason the QR code does not come up when trying to generate it to scan in Zap?

I have been playing with this for days, to no avail.
Everything else seems to work OK, but I get no joy when trying to connect any wallet - no QR code is displayed on either the PI or laptop

rootzoll, 2020-02-09:
@lordgivemestrength if no QR code shows up on trying to connect any of the wallets this #1001 seems not to be your issue. Can you open a new issue and try to give more detail on your problem?

rootzoll, 2020-03-20:
Doing some tests and writing down the results. So with LND 0.9.2 connecting with:
 
- iOS (appstore) IP based --> works

iOS (appstore) over TOR --> gives this error:
[image]

iOS (testflight mainnet) over TOR --> gives this error:
[image]

rootzoll, 2020-03-20:
pusching the error to v1.6

Astro92, 2020-04-02:
Can confirm I receive the same error (could not connect to server. (unknown error 1)) on 1.5

mandreko, 2020-05-12:
I see this error as well :(

CoderBoogs, 2020-05-24:
I also see this error. Has anyone tried any of the other mobile apps? Does fully noded work well?

rootzoll, 2020-05-27:
There is no movement on the ZAP repo on this issue, we remove the TOR option again for now with v1.6 release. As an alternative way v1.6 will have IP2TOR bridge subscription services ... so you can run your node behind tor but rent a public IP+Port that can ce used to connect apps like Zap. See #1194

closed 2020-05-27 by rootzoll

rootzoll, 2020-07-04:
Switched ZAP over TOR for Android back on based on request:
https://github.com/rootzoll/raspiblitz/issues/1308

openoms, 2020-07-06:
The issue looks to be fixed soon on the Zap side as well: https://github.com/LN-Zap/zap-iOS/issues/333
Will need testing once appears in the App store

mandreko, 2020-07-06:
Will we need to update Raspiblitz to test once the new version of Zap appears in the Apple store?

openoms, 2020-07-06:
Appears to work, try for yourself:
[image]

rootzoll, 2020-07-16:
I readded the option to connect Zap iOS over Tor to the Raspiblitz fpr v1.6: https://github.com/rootzoll/raspiblitz/issues/1308

woeisme, 2020-08-27:
Someone should update [this page](https://github.com/openoms/bitcoin-tutorials/blob/master/Zap_to_RaspiBlitz_through_Tor.md) to show it working again

openoms, 2020-08-27:
> Someone should update [this page](https://github.com/openoms/bitcoin-tutorials/blob/master/Zap_to_RaspiBlitz_through_Tor.md) to show it working again

done