raspiblitz/raspiblitz #1008: -bash: lndmanage: command not found

grnqrtr, 2020-02-01:
I'm on 1.4RC3, and went to test lndmanage but get "command not found":

```
admin@raspberrypi:~ $ manage
NOTICE: Needs at least one active channel to run without error.
to exit (venv) enter ---> deactivate
-bash: lndmanage: command not found
(venv) admin@raspberrypi:~/lndmanage $ lndmanage status
-bash: lndmanage: command not found
```

grnqrtr, 2020-02-01:
I just manually entered these lines from `/home/admin/config.scripts/bonus.lndmanage.sh`: 

```
cd /home/admin/lndmanage
  # activate virtual environment
  python3 -m venv venv
  source /home/admin/lndmanage/venv/bin/activate
  # get dependencies
  sudo apt install -y python3-dev libatlas-base-dev
  python3 -m pip install wheel
  python3 -m pip install lndmanage==0.8.0.1
```
It doesn't seem like lndmanage was installed. Now it seems to work. I'm not sure why it wasn't installing by using the script though?

frennkie, 2020-02-01:
Why did you copy the lines manually?

You can run the scripts from the commandline ` /home/admin/config.scripts/bonus.lndmanage.sh: on`

rootzoll, 2020-02-01:
@grnqrtr just to confirm: So you activated the "lndmanage" under SERVICES and after that the command was not available? Do you see in the MAIN MENU the option LNDMANAGE?

grnqrtr, 2020-02-01:
Yes, I activated the "lndmanage" under SERVICES and after that the command was not available. I can see the option LNDMANAGE in the MAIN MENU, but the command was still not available. I even deactivated LNDMANAGE, and reactivated it from the MAIN MENU and it was command was still unavailable. @frennkie I only entered them one at a time so that I could see what was going wrong. I didn't realize I could do it like you said.

frennkie, 2020-02-01:
I just installed 1.4RC3 and can not reproduce this. `manage` runs as expected.

grnqrtr, 2020-02-02:
I just re-flashed 1.4RC3 and was able to reproduce, got the same problem again. This time I have the script output though.

This seems to be the pertinent part:

```
THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    grpcio==1.19.0 from https://www.piwheels.org/simple/grpcio/grpcio-1.19.0-cp37-cp37m-linux_armv7l.whl#sha256=00518c0daaeec358c9fb7a6984eaab7a4be59487e3a9248880d4eba74a00cd2b (from lndmanage==0.8.0.1):
        Expected sha256 00518c0daaeec358c9fb7a6984eaab7a4be59487e3a9248880d4eba74a00cd2b
             Got        da78cb2b6026b8439557ac958dede170886fe3447f6446d7d9381209df2ede5f
```
Full output below:

```
dialogcancel(0)
Testnet Setting unchanged.
Autopilot Setting unchanged.
Loop Setting unchanged.
Dynamic Domain unchanged.
BTC UPnP Setting unchanged.
LND AUTONAT Setting unchanged.
TOR Setting unchanged.
RTL Webinterface Setting unchanged.
BTC-RPC-Explorer Setting unchanged.
LND Autounlock Setting unchanged.
Touchscreen Setting unchanged.
LCD Rotate Setting unchanged.
ElectRS Setting unchanged.
BTCPayServer setting not changed.
lndmanage Setting changed ..
*** INSTALL LNDMANAGE ***
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-dev is already the newest version (3.7.3-1).
python3-dev set to manually installed.
Suggested packages:
  libatlas-doc liblapack-doc
The following NEW packages will be installed:
  libatlas-base-dev libatlas3-base
0 upgraded, 2 newly installed, 0 to remove and 39 not upgraded.
Need to get 5,365 kB of archives.
After this operation, 32.1 MB of additional disk space will be used.
Get:1 http://ftp.jaist.ac.jp/pub/Linux/raspbian-archive/raspbian buster/main armhf libatlas3-base armhf 3.10.3-8+rpi1 [2,399 kB]
Get:2 http://ftp.jaist.ac.jp/pub/Linux/raspbian-archive/raspbian buster/main armhf libatlas-base-dev armhf 3.10.3-8+rpi1 [2,966 kB]
Fetched 5,365 kB in 4s (1,217 kB/s)            
Selecting previously unselected package libatlas3-base:armhf.
(Reading database ... 97081 files and directories currently installed.)
Preparing to unpack .../libatlas3-base_3.10.3-8+rpi1_armhf.deb ...
Unpacking libatlas3-base:armhf (3.10.3-8+rpi1) ...
Selecting previously unselected package libatlas-base-dev:armhf.
Preparing to unpack .../libatlas-base-dev_3.10.3-8+rpi1_armhf.deb ...
Unpacking libatlas-base-dev:armhf (3.10.3-8+rpi1) ...
Setting up libatlas3-base:armhf (3.10.3-8+rpi1) ...
update-alternatives: using /usr/lib/arm-linux-gnueabihf/atlas/libblas.so.3 to provide /usr/lib/arm-linux-gnueabihf/libblas.so.3 (libblas.so.3-arm-linux-gnueabihf) in auto mode
update-alternatives: using /usr/lib/arm-linux-gnueabihf/atlas/liblapack.so.3 to provide /usr/lib/arm-linux-gnueabihf/liblapack.so.3 (liblapack.so.3-arm-linux-gnueabihf) in auto mode
Setting up libatlas-base-dev:armhf (3.10.3-8+rpi1) ...
update-alternatives: using /usr/lib/arm-linux-gnueabihf/atlas/libblas.so to provide /usr/lib/arm-linux-gnueabihf/libblas.so (libblas.so-arm-linux-gnueabihf) in auto mode
update-alternatives: using /usr/lib/arm-linux-gnueabihf/atlas/liblapack.so to provide /usr/lib/arm-linux-gnueabihf/liblapack.so (liblapack.so-arm-linux-gnueabihf) in auto mode
Processing triggers for libc-bin (2.28-10+rpi1) ...
Error: Timeout was reached
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting wheel
  Downloading https://files.pythonhosted.org/packages/8c/23/848298cccf8e40f5bbb59009b32848a4c38f4e7f3364297ab3c3e2e2cd14/wheel-0.34.2-py2.py3-none-any.whl
Installing collected packages: wheel
Successfully installed wheel-0.34.2
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting lndmanage==0.8.0.1
  Downloading https://files.pythonhosted.org/packages/26/58/2caa4e02d53be1591fb3e70049ac93f3ad0199057e401ed5cfcbcafc2fe0/lndmanage-0.8.0.1-py3-none-any.whl (113kB)
    100% |████████████████████████████████| 122kB 1.2MB/s 
Collecting networkx==2.2 (from lndmanage==0.8.0.1)
[... 19 lines cut ...]
  Downloading https://www.piwheels.org/simple/googleapis-common-protos/googleapis_common_protos-1.5.9-py3-none-any.whl (80kB)
    100% |████████████████████████████████| 81kB 149kB/s 
Collecting kiwisolver==1.0.1 (from lndmanage==0.8.0.1)
  Downloading https://www.piwheels.org/simple/kiwisolver/kiwisolver-1.0.1-cp37-cp37m-linux_armv7l.whl (1.4MB)
    100% |████████████████████████████████| 1.4MB 261kB/s 
Collecting protobuf==3.7.1 (from lndmanage==0.8.0.1)
  Downloading https://files.pythonhosted.org/packages/f5/e4/7c86c9c9b4bad7632e107064fb07d0812b9858a49830d0b2477081bf1267/protobuf-3.7.1-py2.py3-none-any.whl (404kB)
    100% |████████████████████████████████| 409kB 771kB/s 
Collecting python-dateutil==2.8.0 (from lndmanage==0.8.0.1)
  Downloading https://files.pythonhosted.org/packages/41/17/c62faccbfbd163c7f57f3844689e3a78bae1f403648a6afb1d0866d87fbb/python_dateutil-2.8.0-py2.py3-none-any.whl (226kB)
    100% |████████████████████████████████| 235kB 1.2MB/s 
Collecting grpcio==1.19.0 (from lndmanage==0.8.0.1)
  Downloading https://www.piwheels.org/simple/grpcio/grpcio-1.19.0-cp37-cp37m-linux_armv7l.whl (11.3MB)
    49% |███████████████▉                | 5.6MB 12.4MB/s eta 0:00:01
Requirement already satisfied: setuptools in ./venv/lib/python3.7/site-packages (from kiwisolver==1.0.1->lndmanage==0.8.0.1) (40.8.0)
[... cut ...]
    grpcio==1.19.0 from https://www.piwheels.org/simple/grpcio/grpcio-1.19.0-cp37-cp37m-linux_armv7l.whl#sha256=00518c0daaeec358c9fb7a6984eaab7a4be59487e3a9248880d4eba74a00cd2b (from lndmanage==0.8.0.1):
        Expected sha256 00518c0daaeec358c9fb7a6984eaab7a4be59487e3a9248880d4eba74a00cd2b
             Got        da78cb2b6026b8439557ac958dede170886fe3447f6446d7d9381209df2ede5f

usage: https://github.com/bitromortac/lndmanage/blob/master/README.md
To start type: 'manage' in the command line.
Needs at least one channel to start without error.
To exit the venv - type 'deactivate' and press ENTER
```

grnqrtr, 2020-02-02:
I just deactivated lndmanage, and reactivated it, and the second time around it worked. It seems like the first time `grpcio-1.19.0-cp37-cp37m-linux_armv7l.whl ` only downloaded partially causing the error, but on the second time around it downloaded fully and no more problem.

```
Downloading https://www.piwheels.org/simple/grpcio/grpcio-1.19.0-cp37-cp37m-linux_armv7l.whl (11.3MB)
    49% |███████████████▉                | 5.6MB 12.4MB/s eta 0:00:01
```

rootzoll, 2020-02-09:
For the final v1.4 release it should add a test after install if the command is available and runnig.

closed 2020-02-17 by rootzoll