raspiblitz/raspiblitz #1017: error: Could not connect to the server 127.0.0.1:8332

DoktorWhat, 2020-02-07:
Installed v1.4rc3 on a complete new setup and followed the instructions to restore my funds and channels. At first the node was running fine and I successfully installed some services but after installing BTC-RPC-Explorer I got the message "error found in logs" on startup.

error: Could not connect to the server 127.0.0.1:8332

I already tried to flash a new SD but get the same error.

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.4
chainnetwork: bitcoin / main
 11:32:31 up 26 min,  2 users,  load average: 0.95, 0.70, 0.60

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset:                                                                                                                                enabled)
   Active: activating (auto-restart) since Fri 2020-02-07 11:32:22 GMT; 9s ago
  Process: 31392 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log bl                                                                                                                               ockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 31397 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.                                                                                                                               bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, stat                                                                                                                               us=0/SUCCESS)
 Main PID: 31399 (code=exited, status=0/SUCCESS)

Feb 07 11:32:22 raspberrypi systemd[1]: bitcoind.service: Succeeded.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Fri 2020-02-07 11:05:51 GMT, end at Fri 2020-02-07 11:32:31 GMT                                                                                                                               . --
Feb 07 11:31:09 raspberrypi systemd[1]: Started Bitcoin daemon.
Feb 07 11:31:31 raspberrypi systemd[1]: bitcoind.service: Succeeded.
Feb 07 11:32:01 raspberrypi systemd[1]: bitcoind.service: Service RestartSec=30s                                                                                                                                expired, scheduling restart.
Feb 07 11:32:01 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job,                                                                                                                                restart counter is at 27.
Feb 07 11:32:01 raspberrypi systemd[1]: Stopped Bitcoin daemon.
Feb 07 11:32:01 raspberrypi systemd[1]: Starting Bitcoin daemon...
Feb 07 11:32:01 raspberrypi systemd[1]: Started Bitcoin daemon.
Feb 07 11:32:22 raspberrypi systemd[1]: bitcoind.service: Succeeded.
1581075119 ERROR: Error found in Logs
1581075144 ERROR: Error found in Logs

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2020-02-07T11:32:21Z Error: Error: A fatal internal error occurred, see debug.lo                                                                                                                               g for details
2020-02-07T11:32:21Z txindex thread exit
2020-02-07T11:32:21Z Loaded 36140 addresses from peers.dat  301ms
2020-02-07T11:32:21Z init message: Starting network threads...
2020-02-07T11:32:21Z net thread start
2020-02-07T11:32:21Z dnsseed thread start
2020-02-07T11:32:21Z addcon thread start
2020-02-07T11:32:21Z opencon thread start
2020-02-07T11:32:21Z msghand thread start
2020-02-07T11:32:21Z init message: Done loading
2020-02-07T11:32:21Z tor: Thread interrupt
2020-02-07T11:32:21Z Shutdown: In progress...
2020-02-07T11:32:21Z torcontrol thread exit
2020-02-07T11:32:21Z dnsseed thread exit
2020-02-07T11:32:21Z addcon thread exit
2020-02-07T11:32:21Z opencon thread exit
2020-02-07T11:32:21Z net thread exit
2020-02-07T11:32:21Z msghand thread exit
2020-02-07T11:32:21Z scheduler thread interrupt
2020-02-07T11:32:21Z Shutdown: done

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enab                                                                                                                               led)
   Active: active (running) since Fri 2020-02-07 11:08:43 GMT; 23min ago
  Process: 6022 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lig                                                                                                                               htning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 6027 (lnd)
    Tasks: 13 (limit: 4915)
   Memory: 128.9M
   CGroup: /system.slice/lnd.service
           └─6027 /usr/local/bin/lnd --externalip=81.10.240.79:9736
[... 59 lines cut ...]
powerMIN='856300 microVolt'
tempFAIL=0
tempWARN=0
tempMAX='4700 centiGrad'

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.1.198'
tempCelsius='44.3'
uptime=1591
upsStatus='OFF'
startcountBlockchain=28
bitcoinActive=0
bitcoinErrorShort='Error found in Logs'
bitcoinErrorFull='2'
# OK: log 'ERROR: Error found in Logs' written to /home/admin/systemd.blockchain                                                                                                                               .log
startcountLightning=1
lndActive=1
lndRPCReady=0
walletOpened=0
walletReady=0
walletLocked=1
lndRPCReady=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=1

DoktorWhat, 2020-02-07:
sudo tail -f /mnt/hdd/bitcoin/debug.log

2020-02-07T12:21:54Z Bitcoin Core version v0.19.0.1 (release build)
2020-02-07T12:21:54Z Assuming ancestors of block 00000000000000000005f8920febd3925f8272a6a71237563d78c2edfdd09ddf have valid signatures.
2020-02-07T12:21:54Z Setting nMinimumChainWork=000000000000000000000000000000000000000008ea3cf107ae0dec57f03fe8
2020-02-07T12:21:54Z Using the 'standard' SHA256 implementation
2020-02-07T12:21:54Z Default data directory /home/bitcoin/.bitcoin
2020-02-07T12:21:54Z Using data directory /mnt/hdd/bitcoin
2020-02-07T12:21:54Z Config file: /home/bitcoin/.bitcoin/bitcoin.conf
2020-02-07T12:21:54Z Using at most 40 automatic connections (1024 file descriptors available)
2020-02-07T12:21:54Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2020-02-07T12:21:54Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2020-02-07T12:21:54Z Using 4 threads for script verification
2020-02-07T12:21:54Z Wallet disabled!
2020-02-07T12:21:54Z scheduler thread start
2020-02-07T12:21:54Z HTTP: creating work queue of depth 16
2020-02-07T12:21:54Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2020-02-07T12:21:54Z HTTP: starting 4 worker threads
2020-02-07T12:21:54Z init message: Loading banlist...
2020-02-07T12:21:54Z Cache configuration:
2020-02-07T12:21:54Z * Using 2.0 MiB for block index database
2020-02-07T12:21:54Z * Using 127.8 MiB for transaction index database
2020-02-07T12:21:54Z * Using 8.0 MiB for chain state database
2020-02-07T12:21:54Z * Using 886.2 MiB for in-memory UTXO set (plus up to 244.1 MiB of unused mempool space)
2020-02-07T12:21:54Z init message: Loading block index...
2020-02-07T12:21:54Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2020-02-07T12:21:54Z Opened LevelDB successfully
2020-02-07T12:21:54Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000
2020-02-07T12:22:06Z LoadBlockIndexDB: last block file = 1957
2020-02-07T12:22:06Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=32, size=38854811, heights=616296...616362, time=2020-02-07...2020-02-07)
2020-02-07T12:22:06Z Checking all blk files are present...
2020-02-07T12:22:06Z Opening LevelDB in /mnt/hdd/bitcoin/chainstate
2020-02-07T12:22:06Z Opened LevelDB successfully
2020-02-07T12:22:06Z Using obfuscation key for /mnt/hdd/bitcoin/chainstate: eadc35e843c5834c
2020-02-07T12:22:07Z Loaded best chain: hashBestChain=0000000000000000000b3c58abd75d7a59526cd86b07f8336557c9873bad3d23 height=616362 date=2020-02-07T09:51:11Z progress=0.999932
2020-02-07T12:22:07Z init message: Rewinding blocks...
2020-02-07T12:22:07Z init message: Verifying blocks...
2020-02-07T12:22:07Z Verifying last 6 blocks at level 3
2020-02-07T12:22:07Z [0%]...[16%]...[33%]...[50%]...[66%]...[83%]...[99%]...[DONE].
2020-02-07T12:22:13Z No coin database inconsistencies in last 6 blocks (10422 transactions)
2020-02-07T12:22:14Z  block index           19640ms
2020-02-07T12:22:14Z Opening LevelDB in /mnt/hdd/bitcoin/indexes/txindex
2020-02-07T12:22:14Z Opened LevelDB successfully
2020-02-07T12:22:14Z Using obfuscation key for /mnt/hdd/bitcoin/indexes/txindex: 0000000000000000
2020-02-07T12:22:14Z txindex thread start
2020-02-07T12:22:14Z Syncing txindex with block chain from height 97
2020-02-07T12:22:14Z block tree size = 616395
2020-02-07T12:22:14Z nBestHeight = 616362
2020-02-07T12:22:14Z ERROR: ReadBlockFromDisk(CBlock&, CBlockIndex*): GetHash() doesn't match index for CBlockIndex(pprev=0x13d2810, nHeight=97, merkle=cc4d0dd009ccf08805d9dca9232a47c60681375312a520f3609045649c2a1690, hashBlock=000000000ecab80b1bee41b0bff9514502d7d8c62bb842655da7b30008c891cc) at FlatFilePos(nFile=0, nPos=21723)
2020-02-07T12:22:14Z *** ThreadSync: Failed to read block 000000000ecab80b1bee41b0bff9514502d7d8c62bb842655da7b30008c891cc from disk
2020-02-07T12:22:14Z Error: Error: A fatal internal error occurred, see debug.log for details
2020-02-07T12:22:14Z txindex thread exit
2020-02-07T12:22:14Z Bound to [::]:8333
2020-02-07T12:22:14Z torcontrol thread start
2020-02-07T12:22:14Z Bound to 0.0.0.0:8333
2020-02-07T12:22:14Z init message: Loading P2P addresses...
2020-02-07T12:22:14Z Leaving InitialBlockDownload (latching to false)
2020-02-07T12:22:14Z Loaded 36140 addresses from peers.dat  298ms
2020-02-07T12:22:14Z init message: Starting network threads...
2020-02-07T12:22:14Z net thread start
2020-02-07T12:22:14Z dnsseed thread start
2020-02-07T12:22:14Z opencon thread start
2020-02-07T12:22:14Z msghand thread start
2020-02-07T12:22:14Z init message: Done loading
2020-02-07T12:22:14Z addcon thread start
2020-02-07T12:22:14Z tor: Thread interrupt
2020-02-07T12:22:14Z Shutdown: In progress...
2020-02-07T12:22:14Z addcon thread exit
2020-02-07T12:22:14Z torcontrol thread exit
2020-02-07T12:22:14Z dnsseed thread exit
2020-02-07T12:22:14Z opencon thread exit
2020-02-07T12:22:14Z net thread exit
2020-02-07T12:22:14Z msghand thread exit
2020-02-07T12:22:14Z scheduler thread interrupt
2020-02-07T12:22:14Z Shutdown: done

rootzoll, 2020-02-09:
Normally after swiching on the BTC-RPC-Expolrer the bitcoind needs to start building the txindex. It looks like this is not happening on your machine and the bitcoind is shutting down because of this error:

```
2020-02-07T12:22:14Z txindex thread start
2020-02-07T12:22:14Z Syncing txindex with block chain from height 97
2020-02-07T12:22:14Z block tree size = 616395
2020-02-07T12:22:14Z nBestHeight = 616362
2020-02-07T12:22:14Z ERROR: ReadBlockFromDisk(CBlock&, CBlockIndex*): GetHash() doesn't match index for CBlockIndex(pprev=0x13d2810, nHeight=97, merkle=cc4d0dd009ccf08805d9dca9232a47c60681375312a520f3609045649c2a1690, hashBlock=000000000ecab80b1bee41b0bff9514502d7d8c62bb842655da7b30008c891cc) at FlatFilePos(nFile=0, nPos=21723)
2020-02-07T12:22:14Z *** ThreadSync: Failed to read block 000000000ecab80b1bee41b0bff9514502d7d8c62bb842655da7b30008c891cc from disk
2020-02-07T12:22:14Z Error: Error: A fatal internal error occurred, see debug.log for details
2020-02-07T12:22:14Z txindex thread exit
```

It looks like you need to redownload the blockchain. On the terminal run the `repair` command and delete the blockchain.

nyxnor, 2021-04-18:
Pls close this.

closed 2021-04-19 by rootzoll