raspiblitz/raspiblitz #1027: HTTPS Proxy & LetsEncrypt

rootzoll, 2020-02-12:
Extra App like BTCPayServer and LNBits (for a working LNURL extension) need a HTTPS with an letsencrypt cert that either works with SSH-Reverse Tunnel or dynDNS&PortForwarding. For this it makes sense to have a HTTPS proxy server (nginx) with letsencrypt that other services can share in the future.

rootzoll, 2020-03-16:
Because this is a bigger task .. moving to v1.6 so that v1.5 is not delayed

frennkie, 2020-05-23:
As #255 already suggests using `nginx` is a very good solution.

Let's focus this issue on the "getting a certificate" part. 

Let's encrypt has (at least) two different ways to get cert https://letsencrypt.org/docs/challenge-types/: 

* HTTP-01 (requires inbound Port 80 connectivity)
* DNS-01 (requires control over a DNS (sub) domain with possibility to set TXT records)

Especially `DNS-01` looks very promising! :-D

rootzoll, 2020-05-27:
@frennkie do you want to push the Lets Encrypt Part to v1.7

frennkie, 2020-05-27:
I would like to include the script to install (and restore) the client in v1.6 if possible. This would include the entry in 'Services'.

But everything else still needs a lot of documentation and deciding on use cases that make sense.

frennkie, 2020-05-30:
I merged the `acme.sh` installation which also includes an update for the FAQ. While this gives (advanced) users the option to get LE certificates there is no integration with e.g. Nginx yet.

Regarding the Apps I have already merged the Nginx integration for `LNbits` and for `RTL`. But the other Apps that host web services are still open. I think having dedicated issues for each would be the best way to track it.

Therefore I marked this as `final testing` with the intend to close it after `v1.6` is both tested and released.

rootzoll, 2020-06-04:
Is there anything that needs to be recovered on a update?

frennkie, 2020-07-12:
@rootzoll I noticed that the `Let's Encrypt` client is now in the `Node Settings & Options` menu. I first looked in the `Additional Services`. 

Selecting the entry triggers the installation - but not more. What is our plan forward? 
Suggestions: Installing the client adds a new menu entry in the main menu where actions (list, get, configure) can selected.

In that case I would still find it more intuitive to place the client installation in `Additional Services`.

rootzoll, 2020-07-12:
Actions like (list, get, configure) are basically what we introduced with Subscriptions. We could even remove "lets encrypt" from the "Settings" and install it on the first subsbription and delete it after the canceling the last "lets encrypt subscription".

So when a user goes into SUBSCRIBE there will be a an additional "NEW2" option called something like "Lets Encypt HTTPS" and then (like choosing a IP2TOR Bridge Host) we can offer a fixed list (no need to conact a outside API here) of DNS-Services that we offer to use like "DuckDNS" or "FreeDNS".

It makes sense to have this as a subscription (even it costs nothing) there is a renewable process of the LetsEncrypt cert in the background.

rootzoll, 2020-07-25:
Looks good on testing. Will be part of v1.6. Closing issue.

closed 2020-07-25 by rootzoll