raspiblitz/raspiblitz #1030: cloning a raspiblitz hdd

woeisme, 2020-02-13:
Hello, 
My node has been down since Sept 2019 as my RPI4 broke. 
I had some funds in open channels and some funds in local channels. 
I recently got a new RPI4 and am attempting to clone the old setup into the new one, I had taken a rescue backup before the node went down, but it was also running out of space. 
Recently I cloned the old hdd to a larger one of the same partition and fs type (FAT32 ext4) and synced all the blockchain data over, labelled the disk the same and copied over the same disk UUID. 
It boots now but I'm getting messages in menu about bitcond not started, when I manually start it from the bitcoin user account it runs however. 
Any help would be appreciated.

openoms, 2020-02-13:
did you enlarge the partition on the new disk?
also post a link with your logs:
`./XXdebugLogs.sh | nc termbin.com 9999`

woeisme, 2020-02-13:
logs are up [here](https://termbin.com/in7h)
Yes I create a partition on the new disk larger and with the duplicate qualities of the original.

openoms, 2020-02-13:
LND has a problem connecting to bitcoind.

Make sure your password B matches in the bitcoin.conf and lnd.conf or just change the password from the menu.

openoms, 2020-02-13:
you can reinstall and or update bitcoin and lnd individually with the scripts here: https://github.com/openoms/raspiblitz-extras/blob/master/README.md#lnd-update

Starting with a. clean SDcard could also solve the problem. Use the same or find the latest image in the v1.4 branch.

woeisme, 2020-02-13:
new [output](https://termbin.com/fonf) 
I noticed it don't have a bitcoin.conf either.

woeisme, 2020-02-14:
seems to be working now. Node is back online and it looks like my funds from last Sept when it went down are still there even.  
Thanks for your guidance as always @openoms you're a great part of the dev community here.

closed 2020-02-14 by woeisme