raspiblitz/raspiblitz #1032: RaPi4 with official 7" touch screen incompatible

SteakDestroyer, 2020-02-14:
It seems when using RaPi4 with a 7" touch screen, on boot the screen will hang at "Load/Save RF Kill Switch Status."
The Pi continues to function properly but the LCD will not update.

openoms, 2020-02-15:
Did you try this? https://github.com/rootzoll/raspiblitz/issues/767
Available in the 1.3RC3 image: https://t.co/mWjwDW9GiA?amp=1

orweinberger, 2020-02-16:
@SteakDestroyer Although I was seeing a different message when my setup hanged, I suggest you try SSHing and executing `sudo ./LCD-Show/LCD7B-show`. That worked for my 7" official setup.

SteakDestroyer, 2020-02-20:
@orweinberger #  Thankyou for the info. I just finished syncing the blockchain and am ready to try this out,  however I'm very inexperienced with Linux and terminal. 
I have tried your suggested command, with a response of
"command not found"

SteakDestroyer, 2020-02-20:
@orweinberger Never mind, Just figured it out. :)
Thankyou again for the hot tip!

closed 2020-02-20 by rootzoll