raspiblitz/raspiblitz #1034: Blockchain Progress 85% - Heat at 90 degree Celsius

future001-bt, 2020-02-15:
Hello,

I am new in this space. Just did all like the tutorial said.

I am in the step where "Node is Syncing (202)"
Blockchain progress 85,46%
Lighting progress 100%

Is it normal that the pi4 (4GB & 480SSD) heats up to 91,5°C (passive cooling sinks) ? Also I feel it's not normal that the sync is starting from 85% or ? All I read is that it starts after ~90% ... Downloaded it from the torrent initial.

Thanks for the help

Best regards.
Rob

future001-bt, 2020-02-15:
no the pi just switched off - I think. the display is just lighting without text ect.

rootzoll, 2020-02-17:
I would not recommend selfvalitading the blockchain on a RaspberryPi4 with just passive cooling - it will get very hot and might stop. Thats why the RaspiBlitz parts shopping list includes active fan cooling to do that. If you want to work with passive cooling I would recommend sync/validating the blockchain on your laptop and then copy it over to the RaspberryPi during setup with the COPY option. See README & FAQ for more details on that.

closed 2020-02-17 by rootzoll