raspiblitz/raspiblitz #1039: Touchscreen displays incorrect version.

hermanlim, 2020-02-19:
Running updated 1.3->1.4

Touchscreen displays Raspiblitz v1.3 - bitcoin - mainnet 
instead of
Raspiblitz v1.4 - bitcoin - mainnet

rootzoll, 2020-02-19:
Thanks .. should now be fixed for final v.14-2020-02-19 release.

hermanlim, 2020-02-20:
Just out of curiosity, which file needed changing? I looked through the source but couldn’t find the relevant file to fix it myself. 

Herman

Please excuse the brevity and any typos as this message has been composed on a mobile device.

rootzoll, 2020-02-20:
Take a look at your `/mnt/hdd/raspiblitz.conf` there you can update version to v1.4 .. the final v1.4 release will now automatically update that version string. My testing of updating a v1.3 to v1.4 with touchscreen showing 1.4 was sucessfull. Closing issue.

closed 2020-02-20 by rootzoll