raspiblitz/raspiblitz #1040: lndmanage install problem

grnqrtr, 2020-02-19:
This is an update to the problem I had on 1.4RC3 (#1008)

On 1.4 final, I enabled lndmanage from the services menu and in the log I noticed I had the same problem (grpcio-1.19.0-cp37-cp37m-linux_armv7l.whl downloaded to 49% and stopped) and I noticed the message at the end:

```
!! FAIL --> Was not able to install LNDMANAGE
!! Maybe because of internet network issues - try again later.
```

I tried to enable again from the services menu, stuff blipped on the screen so fast I couldn't see what happened, but LNDMANAGE didn't appear to be installed. I did `manage` to see if it was installed and it said it wasn't, but when I did `sudo /home/admin/config.scripts/bonus.lndmanage.sh on` it says it is already installed:

```
admin@raspberrypi:~ $ manage
lndmanage not installed - to install run:
sudo /home/admin/config.scripts/bonus.lndmanage.sh on
admin@raspberrypi:~ $ sudo /home/admin/config.scripts/bonus.lndmanage.sh on
LNDMANAGE already installed
```
I restarted the device to see if that would help, and I checked `/mnt/hdd/raspiblitz.conf` which says `lndmanage=off` but even after restart I get:

```
admin@raspberrypi:~ $ manage
lndmanage not installed - to install run:
sudo /home/admin/config.scripts/bonus.lndmanage.sh on
admin@raspberrypi:~ $ sudo /home/admin/config.scripts/bonus.lndmanage.sh on
LNDMANAGE already installed
```

rootzoll, 2020-02-19:
Thanks for testing. Because I cannot reproduce that problem with the failing lndmange pip install (it works on my machine) this was a blind try to fix. I will take another look.

rootzoll, 2020-02-19:
@grnqrtr Can you run "./XXsyncScripts.sh" on your RaspiBlitz Terminal and then test again please.

grnqrtr, 2020-02-20:
I synced scripts and restarted, and that seemed to fix the issue of being in an in-between-state of installed/not installed for lndmanage. For whatever reason I still had trouble with the downloads failing. I had to run  `sudo /home/admin/config.scripts/bonus.lndmanage.sh on` 3 or 4 times before everything downloaded fine and the install succeeded, but that seems to only be an issue I am encountering for what ever reason.

```
admin@raspberrypi:~ $ manage
lndmanage not installed - to install run:
sudo /home/admin/config.scripts/bonus.lndmanage.sh on
admin@raspberrypi:~ $ sudo /home/admin/config.scripts/bonus.lndmanage.sh on
*** INSTALL LNDMANAGE ***
Reading package lists... Done  
Building dependency tree       
Reading state information... Done
libatlas-base-dev is already the newest version (3.10.3-8+rpi1).
python3-dev is already the newest version (3.7.3-1).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: wheel in ./venv/lib/python3.7/site-packages (0.34.2)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting lndmanage==0.8.0.1
  Downloading https://files.pythonhosted.org/packages/26/58/2caa4e02d53be1591fb3e70049ac93f3ad0199057e401ed5cfcbcafc2fe0/lndmanage-0.8.0.1-py3-none-any.whl (113kB)
    100% |████████████████████████████████| 122kB 1.1MB/s 
Collecting googleapis-common-protos==1.5.9 (from lndmanage==0.8.0.1)
  Downloading https://www.piwheels.org/simple/googleapis-common-protos/googleapis_common_protos-1.5.9-py3-none-any.whl (80kB)
    100% |████████████████████████████████| 81kB 146kB/s 
Collecting decorator==4.4.0 (from lndmanage==0.8.0.1)
  Downloading https://files.pythonhosted.org/packages/5f/88/0075e461560a1e750a0dcbf77f1d9de775028c37a19a346a6c565a257399/decorator-4.4.0-py2.py3-none-any.whl
Collecting networkx==2.2 (from lndmanage==0.8.0.1)
  Downloading https://www.piwheels.org/simple/networkx/networkx-2.2-py2.py3-none-any.whl (1.5MB)
    100% |████████████████████████████████| 1.5MB 245kB/s 
Collecting protobuf==3.7.1 (from lndmanage==0.8.0.1)
  Downloading https://files.pythonhosted.org/packages/f5/e4/7c86c9c9b4bad7632e107064fb07d0812b9858a49830d0b2477081bf1267/protobuf-3.7.1-py2.py3-none-any.whl (404kB)
    100% |████████████████████████████████| 409kB 595kB/s 
Collecting grpcio==1.19.0 (from lndmanage==0.8.0.1)
  Downloading https://www.piwheels.org/simple/grpcio/grpcio-1.19.0-cp37-cp37m-linux_armv7l.whl (11.3MB)
    100% |████████████████████████████████| 11.3MB 39kB/s 
Collecting six==1.12.0 (from lndmanage==0.8.0.1)
  Downloading https://files.pythonhosted.org/packages/73/fb/00a976f728d0d1fecfe898238ce23f502a721c0ac0ecfedb80e0d88c64e9/six-1.12.0-py2.py3-none-any.whl
Collecting grpcio-tools==1.13.0 (from lndmanage==0.8.0.1)
  Downloading https://www.piwheels.org/simple/grpcio-tools/grpcio_tools-1.13.0-cp37-cp37m-linux_armv7l.whl (26.2MB)
    27% |█████████                       | 7.3MB 15.0MB/s eta 0:00:02
Collecting cycler==0.10.0 (from lndmanage==0.8.0.1)
  Downloading https://files.pythonhosted.org/packages/f7/d2/e07d3ebb2bd7af696440ce7e754c59dd546ffe1bbe732c8ab68b9c834e61/cycler-0.10.0-py2.py3-none-any.whl
Collecting pyparsing==2.4.0 (from lndmanage==0.8.0.1)
  Downloading https://files.pythonhosted.org/packages/dd/d9/3ec19e966301a6e25769976999bd7bbe552016f0d32b577dc9d63d2e0c49/pyparsing-2.4.0-py2.py3-none-any.whl (62kB)
    100% |████████████████████████████████| 71kB 1.2MB/s 
Collecting python-dateutil==2.8.0 (from lndmanage==0.8.0.1)
  Downloading https://files.pythonhosted.org/packages/41/17/c62faccbfbd163c7f57f3844689e3a78bae1f403648a6afb1d0866d87fbb/python_dateutil-2.8.0-py2.py3-none-any.whl (226kB)
    100% |████████████████████████████████| 235kB 695kB/s 
Collecting numpy==1.16.2 (from lndmanage==0.8.0.1)
  Downloading https://www.piwheels.org/simple/numpy/numpy-1.16.2-cp37-cp37m-linux_armv7l.whl (7.4MB)
    61% |███████████████████▌            | 4.5MB 15.3MB/s eta 0:00:01
Collecting kiwisolver==1.0.1 (from lndmanage==0.8.0.1)
  Downloading https://www.piwheels.org/simple/kiwisolver/kiwisolver-1.0.1-cp37-cp37m-linux_armv7l.whl (1.4MB)
    100% |████████████████████████████████| 1.4MB 241kB/s 
Requirement already satisfied: wheel in ./venv/lib/python3.7/site-packages (from lndmanage==0.8.0.1) (0.34.2)
Requirement already satisfied: setuptools in ./venv/lib/python3.7/site-packages (from protobuf==3.7.1->lndmanage==0.8.0.1) (40.8.0)
THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    grpcio-tools==1.13.0 from https://www.piwheels.org/simple/grpcio-tools/grpcio_tools-1.13.0-cp37-cp37m-linux_armv7l.whl#sha256=f8c58b198f3a6c557c9cc65ff09e472b9022524917d538610fd7658f3493fa79 (from lndmanage==0.8.0.1):
        Expected sha256 f8c58b198f3a6c557c9cc65ff09e472b9022524917d538610fd7658f3493fa79
             Got        e262a436e2127ea8be88e684d5fbf97da974a73fd9f21a85c8be10a8bc581b2c

    numpy==1.16.2 from https://www.piwheels.org/simple/numpy/numpy-1.16.2-cp37-cp37m-linux_armv7l.whl#sha256=77035a883468dc8ad00101cd798e8df128ee90cc93165eac37943533bbb7216a (from lndmanage==0.8.0.1):
        Expected sha256 77035a883468dc8ad00101cd798e8df128ee90cc93165eac37943533bbb7216a
             Got        d2bd1de3994cb916df6a76e517ebc9fbcefd4ed4b4a74ef52adf531d307f3e80

#!! FAIL --> Was not able to install LNDMANAGE
#!! Maybe because of internet network issues - try again later.
```

rootzoll, 2020-02-20:
Good to hear that it worked in the end and we have the script now download fail safe- this is now part of the final v1.4 release 2020-02-20. If you ever find the reason why the download on your machine is that flacky, let us know.

closed 2020-02-20 by rootzoll