raspiblitz/raspiblitz #1041: Why not using a configuration management tool?

cedricwalter, 2020-02-19:
I went through the code, there is lots of code and i was just asking myself why not using Puppet, Chef, Ansible or any other configuration management tool. The code is great no question.
I just see a lot of code that could be replaced while still being more readable...

@rootzoll  What do you think?

rootzoll, 2020-02-19:
There are multiple factors to cosider.

1) reduce tools and external dependencies needed to setup your miniumum node: bitcoind & lnd ... thats also the reason why we dont use docker. even if they are open-source they add a lot of extra complexity that can widen the attack surface - we running a hat wallet here, so less is more. Sure if you install additional services like RTL you also install nodeJS .. but thats then an optional choice later. The minimum should stay just to the really required.

2) I checked quickly puppet and chef and they run and it seems you need to have a client sevice of them running in the systemd services. Also I quick look at Ansible talks abour cient and master servers. That makes sense if you have to control a server farm or a scalable production setup for a startup, but RaspiBlitz is a single Homenode that should not depend on a master server. Maybe you can use those tools without a master, but this shows that those tools are comming with a different mindset and problem to solve.

3) Such tools add another layer of complexity in understanding the code. Everybody running a linux based server knows at least some bash script because its basically unix commands with some basic syntax sugar. That made it easy in the past for a lot of people to contribute and take a look at the code. Tools like chef, puppet and asible are frameworks that are not familar to everybody. For some it will make the code more clean - most others would have to learn a new framework first. There are some ideas to maybe use more python code for RaspiBlitz in the future .. but thats at least a very basic and common programing lang.

4) Yes there is a lot of configuration ... but there is also a lot of processes and interaction wit the user. Also this interaction creates a lot of different configurations ... this sounds also very complicated to setup is such frameworks or at least needs some bash/python on the side.

You got me thinking if Ansible might be something to look deeper into for the future, because it would fit well with python. But can it be used without installing this master server stuff?

cedricwalter, 2020-02-19:
yes it is an added dependency but

1. All these tools (puppet, chef, ansible, ...) abstract the underlining operating system -> with the same code you could run Raspiblitz on any kernel or Linux LSB. (not required for now)
2. You don't need an agent nor a puppet master you can just apply a script on the host
3. yes and no, these tools are really common and solve all the complexity to install service descriptor, define services dependencies, restart service automatically if needed. 
4. All interactions could just configure the puppet YAML (i don't know how to do it in other tool but for sure it is also there)

I'm just surprised to see some much bash code for stuff like changing permissions, modifying configuration files, getting dependencies and more...
-> you could create templates in puppet
-> puppet could monitor files changed and revert these permissions according to the descriptor automatically reducing misconfiguration and undetected filesystem changes
-> you don't need to open a new port in the firewall, puppet could just every 15min checking that all files, permissions, and service are there according to the plan. 
-> if one service (LND, Bitcoin, RTL, ...) fails and crash puppet could restart them in the correct order, if any
-> you get repeatable runs and proper error messages without the need to code them yourself
And the list goes on

Just my 2 cents, I'm also a big fan of bash!

lessless, 2021-03-02:
@rootzoll Ansible is agentless and would be a perfect fit. I'm 100% sure that it would be easier to maintain.
I think converting the existing sh file is a bearable task but will it be accepted or not is another topic.