raspiblitz/raspiblitz #1042: Cannot SSH back into device after blockchain starts to download via torrent.

fifthedition, 2020-02-19:
I tried to install clean copy raspiblitz-v1.4-2020-02-18.img.gz with a fresh clean HDD but I cannot SSH back into the device once the torrent with the blockchain starts to download, the device even says that it is alright to close the terminal and I can ssh back into the device later. I can see the torrent to download but once I close the terminal I cannot ssh back. I get a response that the device refused the connection.

rootzoll, 2020-02-20:
Do you have the ShimFan running?

fifthedition, 2020-02-20:
Yes, I do.

fifthedition, 2020-02-21:
I was able to SSH back in the device after removing FanShim and using normal/other fan. Will the FanShim ever be useful again? I feel like there is more and more issues with the FanShim as the time goes.

rootzoll, 2020-03-02:
I am testing an alternative to the ShimFan at the moment: A new LCD-Hat with fan included on the back ... it seems to be promissing and ShimFan seems to be phased out soon.

nyxnor, 2021-04-18:
Pls close this.

closed 2021-04-19 by rootzoll