raspiblitz/raspiblitz #1052: Restoring data after update from 1.3 to 1.4

fstetter, 2020-02-26:
I followed the update guideline for my blitz and downloaded the restore file. Unortunately this is a **lnd-rescue-...tar.gz** instead  of a **raspiblitz-...tar.gz**, so when i try to upload to migrate i get an error, as it demands an **raspiblitz-...tar.gz** file. 

I tried to alter the shell command and also renaming the file and the **lnd.conf** to **raspiblitz.conf** but i always run into some error. How can i handle this situation properly?

(unfortunately, now my old sd card does not boot like normal anymore)

openoms, 2020-02-26:
It is not migration you need, but to follow setting up a Bitcoin and Ligthning node. Once you get to creating the lightning wallet you will have the prompts to upload the lnd-rescue file.

fstetter, 2020-02-27:
Oh, i see. I think i was to afraid messing up my hard drive with this choice. Thanks for the help!

closed 2020-02-27 by fstetter