raspiblitz/raspiblitz #1053: Cannot SSH after electrs install

ostermayer, 2020-02-26:
Updated 1.4 then install electrs from menu --> after restart ssh doesn't work due to operation timeout.  let sit for a few hours but no ability to ssh 

Also could not ssh after upgrade from 1.3 (with electrs installed )-> 1.4.  needed to do a full fresh install and restore in order to get 1.4 to work.

rootzoll, 2020-03-02:
Do you found any other cooling options for your raspiblitz? I had ssh login problems on a Raspi that was overheating and got instable. Is it now (after 1.4 fresh install) running without problems?

ostermayer, 2020-03-02:
it runs on a 40c temp or less sitting in a always cool spot

ostermayer, 2020-03-09:
[image]

ostermayer, 2020-03-09:
this was the cause and is only fixed by attaching a keyboard and then ssh worked afterwards

rootzoll, 2020-03-18:
For the v1.5 RC1 release trying to add a forced fsck for such cases - like described here:
https://github.com/krishraghuram/mess/issues/31#issuecomment-370811935

Hopefully this will fix this "Give root password for maintenance" issue.

closed 2020-04-18 by rootzoll

joernroeder, 2020-09-25:
quick note: I just ran into this with a after updating to v1.6

ostermayer, 2020-09-25:
yes as did I

rootzoll, 2020-09-28:
Yepp .. this was sadly not fixing this. If you run into it you need to make a fresh sd card and let the RaspiBlitz recover.

Our next hope to fix this is to run OS from SSD and not SD card in the a future release - that option may come with v1.7 or v1.8 https://github.com/rootzoll/raspiblitz/issues/1235