raspiblitz/raspiblitz #1054: "The bitcoind service is not running." happened 3 times in last week. I did fresh reinstall every time.

fifthedition, 2020-02-26:
I got this issue 3 times since raspiblitz 1.4 was released. I do a fresh install of 1.4 rapsiblitz with formated hdd. As soon as I start to install services or mobile wallet support, I run to this message after reboot:
```
               Blockchain Info                                                
│  The bitcoind service is not running.   │                                               
│  Login for more details:                │                                               
│  Use CTRL+c to EXIT to Terminal         │
```
So far I received the above message after I installed RTL, Block Explorer and ZAP mobile wallet client. It seemed that if dont enable anything the raspiblitz will run perfectly fine.

here is a whole log file obtained with this code
`sudo cat /mnt/hdd/bitcoin/debug.log | nc termbin.com 9999`
https://termbin.com/fkxx

This full log is after fresh installation and an attempt to enable ZAP for iOS in the menu. After reboot I got the above message (The bitcoind service is not running.)

Any help on this would be appreciated.

PS: I am not having FanShim plugged in as that one is causing a lot of problems as it seems

openoms, 2020-02-26:
The blockchain is corrupted on the disk:
```
*** ThreadSync: Failed to read block 000000008ec9525cb2092c6bd7d1c2580e595bcfa54bac51cf6f0278b242f25d from disk
```
Type:
`repair`
and Reset the Blockchain 
Sync or copy it over again.

To avoid issues in the future use tested parts and do not pull the power while writing the disk.

fifthedition, 2020-02-26:
Thank you for the lightning fast reply (pun intended).
I will try that.
I am using 7-8 year's old HDD maybe it started to fail.

fifthedition, 2020-03-01:
So, I upgraded my old HDD to new Samsung 860 QVO 1TB SSD. I downloaded new copy of the blockchain via torrent trough raspiblitz and I got the same error.

I suspect that there is something wrong with block 73 in the blockchain downloaded via torrent. 

```
2020-03-01T00:46:26Z ERROR: ReadBlockFromDisk(CBlock&, CBlockIndex*): GetHash() doesn't match index for CBlockIndex(pprev=0x567e230, nHeight=73, merkle=41b48c64cba68c21e0b7b37f589408823f112bb7cbccef4aece29df25347ffb4, hashBlock=000000008ec9525cb2092c6bd7d1c2580e595bcfa54bac51cf6f0278b242f25d) at FlatFilePos(nFile=0, nPos=16371)
2020-03-01T00:46:26Z *** ThreadSync: Failed to read block 000000008ec9525cb2092c6bd7d1c2580e595bcfa54bac51cf6f0278b242f25d from disk
2020-03-01T00:46:26Z Error: Error: A fatal internal error occurred, see debug.log for details
```

here is the full log:
https://termbin.com/qr3b

I also tried clean instal of raspiblitz and torrent blockchain download but once I enable blockchain indexing I run into the same error and block 73 is causing problems.

Anybody got the same issue?

openoms, 2020-03-01:
Interesting, curious if anybody experiences the same.

With the SSD:
* make sure you use a tested USB adapter (for example the one form the shopping list or others listed in the issues)
* the RPi4 can sync the blockchain from the bitcoin network on 2-4 days so would recommend not using the Torrent option.

rootzoll, 2020-03-02:
You have 6 temp warnings on your RaspiBlitz running at 72 Grad. This may impact the stability of your RaspiBlitz. Do you have a fan for cooling like adviced in the shopping list?

rootzoll, 2020-03-02:
Just saw that you removed the shim fan. Which country are you living in ... I would recommend chacking your amazon if you can order one of those new LCD-Hats with a fan on the back. They seem to work OK ... you could help testing them and let us know your local shopping link.

nielsoo, 2020-06-23:
Hi All 

I have the same issue, with a freshly bought raspiblitz. 

the INTERESTING part is that the error I have happens to be on the EXACT same block 

2020-06-23T10:55:57Z ERROR: ReadBlockFromDisk(CBlock&, CBlockIndex*): GetHash() doesn't match index for CBlockIndex(pprev=0x5a6def8, nHeight=73, merkle=41b48c64cba68c21e0b7b37f589408823f112bb7cbccef4aece29df25347ffb4, hashBlock=000000008ec9525cb2092c6bd7d1c2580e595bcfa54bac51cf6f0278b242f25d) at FlatFilePos(nFile=0, nPos=16371)
2020-06-23T10:55:57Z *** ThreadSync: Failed to read block 000000008ec9525cb2092c6bd7d1c2580e595bcfa54bac51cf6f0278b242f25d from disk
2020-06-23T10:55:57Z Error: Error: A fatal internal error occurred, see debug.log for details
2020-06-23T10:55:57Z txindex thread exit

so this definitely isn't any "hardware" issue like some people suggest here

rootzoll, 2020-06-23:
@nielsoo did you used the prepared blockchain that came with the RaspiBlitz?

nielsoo, 2020-06-23:
Yes I did, but I have deleted it now and am now syncing from scratch!

animalmother1988, 2020-06-23:
Hi Nielsoo, 

when did you bought your RaspiBlitz?

nielsoo, 2020-06-23:
ordered may 21th arrived june 4th

nielsoo, 2020-06-29:
Blockchain synced and up and running for a few days now without issues.. so it just seemed to have been a bad block in the shipped presynced blockchain :-)

notverygoodatcoding, 2020-11-07:
Snap. Same error, same block (73) after torrenting. Just lettting you know.

rootzoll, 2020-11-07:
@notverygoodatcoding using the torrent is not recommended anymore. If you haev RP4 then let sync by itself of you have an old RP3 then sync the Blockchain with a Bitcoin Core on your laptop and then use COPY option over LAN.

notverygoodatcoding, 2020-11-08:
Thanks for the response, and great project by the way. I've had loads of fun messing around with my node for the past three months and feel like I've learnt a lot. Currently syncing the chain whilst growing my facial hair.......Should be downloaded and fully bearded any day now......

techric, 2021-03-13:
I'm having the same issue. It appears there was no definitive solution to this problem. 

Should I just wait four days like nielsoo did?

rootzoll, 2021-03-14:
@techric doing a clean resync could be a good chance to fix this

techric, 2021-03-29:
For continuity's sake ... 

My problem appears to have been remedied by changing the SSD and case to the one listed by Raspiblitz.

nyxnor, 2021-04-18:
Pls close this.

closed 2021-04-19 by rootzoll