raspiblitz/raspiblitz #1055: BTRFS tests (experimental feature)

openoms, 2020-02-26:
Tested the BTRFS disk layout  a couple of times with empty disks and fresh 1.4 SD and could not get to make it work yet.
The first 30GB partition gets mounted to /mnt/hdd and runs out of space while copying the blockchain over.
Will need to look into the fstab configuration.

rootzoll, 2020-03-28:
Marked as Hackathon Sprint:

- To test BRFS & Raid and optimize solution
- Add OZFS as further option

openoms, 2020-05-19:
Testing this to recover from a BTRFS disk on Armbian buster with Linux 5.4.28-odroidxu4

It does not mount the BTRFS partititons for some reason and is very slow scanning the BTRFS filesystem.
```
$ sudo /home/admin/config.scripts/blitz.datadrive.sh status
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=0
isBTRFS=0
isSSD=0

# SETUP INFO
hddCandidate=''

# RAID
isRaid=0
raidCandidate[0]='zram1 0 GB zram1'
raidCandidates=1

# SWAP
isSwapExternal=0
```

```
admin@odroidxu4:~$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 931.5G  0 disk 
├─sda1        8:1    0    30G  0 part 
├─sda2        8:2    0 867.5G  0 part 
└─sda3        8:3    0    34G  0 part 
mmcblk1     179:0    0  29.8G  0 disk 
└─mmcblk1p1 179:1    0  29.5G  0 part /
zram0       253:0    0    50M  0 disk /var/log
zram1       253:1    0 996.7M  0 disk [SWAP]
```

Will continue to investigate this.

hermanlim, 2020-05-21:
Given most users that are running 500GB SSDs will probably upgrade to 1000GB (or greater) SSDs eventually, is it reasonable to have a script to facilitate upgrading/migration that would:
1. Format the new SSD into BTRFS
2. Copy the old SSD data to be the new SSD
3. Format the old SSD as the RAID

rootzoll, 2020-05-21:
@hermanlim I know having a migration that would just copy data from one HDD to another bigger one seems to be the first guess. But that would mean you need to connect two HDDs at the same time to the RaspiBerryPi and power them ... this not a good idea and has a high risk of data corruption due to power loss. Yes if you put a powerd USB-Hub inbetween this might work, but this needs a additional hardware and also the setup still be bad (weak USB Hub power outlets, etc) ... so the easiest way was to go with a migration script that will:

a) Copy critical data over LAN from RaspiBlitz to Laptop
b) Exchange HDD and format on RaspiBlutz
c) Copy critical data over LAN from Laptop to RaspiBlitz
d) reload/revalidate non-critical data like the blockchain

Please see details on the migration script here: https://github.com/rootzoll/raspiblitz/blob/master/README.md#import-a-migration-file

rootzoll, 2021-01-04:
see: https://github.com/rootzoll/raspiblitz/issues/1910#issuecomment-751818293

rootzoll, 2024-11-14:
Closing this issue, because it seems we will remove BTRFS - see: https://github.com/raspiblitz/raspiblitz/issues/2045#issuecomment-2471845122

closed 2024-11-14 by rootzoll