raspiblitz/raspiblitz #1056: cannot access '/mnt/hdd/raspiblitz.conf': Input/output error

brianhrad, 2020-02-27:
While trying to upgrade from 1.3 to 1.4 I ran across an error message that I am struggling with.

****
bitcoin still not running
ls: cannot access '/mnt/hdd/raspiblitz.conf': Input/output error

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
FAIL: /mnt/hdd/raspiblitz.conf should exists at this point, but not found!
Please report to: https://github.com/rootzoll/raspiblitz/issues/293
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
*****

Not sure how this came about.  Any advise?

openoms, 2020-02-27:
It is most likely to be a hardware error.
Was your node fully working on 1.3?
HW details?

Start again with a fresh 1.4 SDcard.
If funds involved see the restore options in the FAQ.

brianhrad, 2020-03-03:
Yep, looks like it was a hardware problem with the HDD.  Got a new HDD and got it back up and running.

closed 2020-03-03 by brianhrad