raspiblitz/raspiblitz #1058: i2c /dev entries driver

Lockejoe, 2020-02-29:
Hello all,

Hope someone can help me.
I tried to update my Raspiblitz from 1.3 to 1.4.

No I got following Issue with the new image, while booting.
_i2c dev entries driver_

System stuck at this point.
I tried different sd cards, tried different images, downloads, nothing helped.

Any ideas how to solve it?
Also tried to google, but I couldn't find a solution, which fits for me. (I'm also not that familiar with linux, especially with kernel problems.)

Thanks in advance

rootzoll, 2020-02-29:
The "i2c dev entries driver" is your last message on the LCD, correct?
Can you still ssh in? There are some reports that the LCD is just freezing on boot, but the rest works. Also do you have the ShimFan? Ist ist also not bootingif you remove the shimfan?

Lockejoe, 2020-03-01:
Hello,
Thanks for your answer.
Yes, this is my last message, but I do not have a LCD, I also do not have a ShimFan.
I only use with ssh.
After the update it wasn‘t possible to connect via ssh, neither with wifi nor with ethernet.
So I cecked with a „normal“ monitor to find out the root cause.
And it always stuck i2c /dev entries driver, as last message.

openoms, 2020-03-01:
@Lockejoe simply need to unplug the monitor from the HDMI and boot without it. Login via ssh and follow the guide.
During the initial setup using the HDMI freezes the boot process. Later it will be possible to use. Look for HDMI in the issues to get the script helping you to change to that from the default GPIO LCD output.

Lockejoe, 2020-03-02:
Hello @openoms 
Thanks for your answer. Ok maybe my first explanation was not that good.
What I did:
1. Updated my sd card (tried with different sd cards)
2. started my rasperry pi (plugged in) - there was no display connected, only my hdd
3. waited for ca. 10-15 min (and also trying with known ips to ssh...)
4. checked my network to identify correct ip (if the old ip is still correct) - wondering, that I couldn‘t find an active one for the raspberry pi
5. connected a monitor to identify the root cause why couldn‘t ssh the pi

Hope this helps

ZukoTFL, 2020-03-06:
Hey @Lockejoe I just had this issue after powering off for a little while and turning back on. Got the same "i2c /dev entries driver" message at the end. I have my hdd in an external enclosure which I forgot to power off. Once I powered both off and back on, I immediately got to the main screen. Hope this helps

Lockejoe, 2020-03-13:
Hello @ZukoTFL, Thanks for this hint, but unfortunately I do not have an external enclosure. Energy by plugging in USB (standard WD elements 1TB HDD) .

csc0711, 2020-03-14:
@Lockejoe I just had this issue while trying to set up a new node.  Was starting the node with the HDMI plugged and using a fan, screen would hang up at "i2c /dev entries driver".  I tried a few things similar to you, but hadn't tried SSH to view the node status.  I figured what I was getting via HDMI was accurate, but it wasn't.  I was able SSH into my node and continue setting up the node. I haven't tried to reboot w/o the HDMI to see it that resolves the display issue but currently my Pi seems to be working correctly.  
Doesn't help much since you can't SSH in through, sorry.  I would see if you can SSH or display the node in a different way (other than HDMI), to see if the Pi is actually getting hung up or if its just a display issue.  You might also try downgrading the image back to v1.3 if that was working properly before.
Good Luck!

Lockejoe, 2020-03-15:
Hello @csc0711 , Hello @rootzoll,

As a hint:
Today I did a downgrade and what shall I say, it's working like a sharme.
No issues with v1.3 for me.

Regards
Lockejoe

rootzoll, 2020-03-16:
@Lockejoe interesting that a downgrade fixed the LCD problem. I will check again what changed from the last version on the LCD driver side.

rootzoll, 2020-03-18:
For the v1.5 RC1 I will set back the LCD driver to the commit of the v1.3 version. The only thing that changed since then is this - the version of the xserver-xorg-input-evdev

``
-- sudo dpkg -i -B ./xserver-xorg-input-evdev_1%3a2.10.5-1_armhf.deb 2> error_output.txt
++ sudo dpkg -i -B ./xserver-xorg-input-evdev_1%3a2.10.6-1+b1_armhf.deb 2> error_output.txt
``

@Lockejoe once the v1.5 RC1 is out - it would be great if you can test if this fixes the bug for you - thx

Lockejoe, 2020-03-18:
@rootzoll

rootzoll, 2020-03-20:
Tested: the old LCD driver works also with the new LCD-with-integrated Fan.

raspito, 2020-06-15:
hi guys, same thing is happening for me.

my setup:
- no screen, just network access
- downloaded image around march 30

was working until a restart recently, wasn't showing up on network. everything looked fine (lights etc) so I plugged in a screen. boot hangs at i2c /dev entries driver

I've tried booting without the HDMI then plugging it in 10 min later (same result). Have also removed case fan and external HDD to isolate.

Any advice for how to resolve this?

Cheers,

Tom

rootzoll, 2020-06-15:
This error is a nasty one ... still not sure why this is happening to some people. Can you try to build an raspiblitz sd card yourself from a fresh raspnian image and see at which point you run in that error? See how to build: https://github.com/rootzoll/raspiblitz/blob/master/README.md#build-the-sd-card-image

raspito, 2020-06-17:
Hi guys, will get back to you on this. Now having some hard disk errors, so will resolve that, re-flash to 1.5.1 then test.

Teegardin, 2020-08-29:
Installation on my RPi 3 has gotten stuck at dev entries driver trying to update from ver 1.5 to ver 1.6. Using RPi 3 b+, 32GB Samsung Evo+, 1TB WD Passport.

chinedu0507, 2020-11-11:
Hello guys, I am not sure if this bug is still being looked at, but I am also encountering the same issue with version 1.6.1; unplugging the SSD and/or HDMI cable makes no difference. Please advice if there is a workaround this. Thanks.

rootzoll, 2020-11-11:
@chinedu0507 whats your setup? We dont have a quick solution here - but this seems related to the LCD drivers. For v1.7 we will try to update to 64bit RasbianOS with updated LCD driver and maybe even see if we remove LCD drivers completly of no LCD connected. So I reopened this issue as a reminder to check on - but it will take some weeks before we will get to this.

chinedu0507, 2020-11-12:
So my setup is as follows:

Rpi 4 (4GB RAM) with a 
500GB NAND SSD

I do not have an LCD, but I figured that I could ssh into the Pi after getting my IP address. I do have a Pi case that has dual fans (don't think this is important). But that's all. 

> @chinedu0507 whats your setup? We dont have a quick solution here - but this seems related to the LCD drivers. For v1.7 we will try to update to 64bit RasbianOS with updated LCD driver and maybe [...]

strowheim, 2020-11-12:
Same issue with my brand new Raspberry PI. Flashed with raspiblitz-v1.6.1-2020-10-20.img.gz

Using SABRENT SSD case and Samsung 1TB

Had MULTIPLE issues with the hard drive, and mutliple hangs at i2c /dev entries driver

Reflashing to same version didn't work
Troubleshoot the drive, which was not initializing -- it looks fine from other machines and I did  badblock testing, fsck testing, smartctl etc all normal.

sven3550, 2020-11-25:
Same issue. Raspberry Pi v4. Tried with v1.6.1, v1.6 and v1.4

takiseso, 2020-11-29:
Same issue. Raspberry Piv4 also. Is there a way to login to RPi4?

openoms, 2020-11-29:
@strowheim @sven3550 @takiseso please make sure you are using a compatible and tested USB-disk adapter with the RPi4. The best is to use model recommended in the shopping list or an X825 adapter board.

sven3550, 2020-11-30:
> @strowheim @sven3550 @takiseso please make sure you are using a compatible and tested USB-disk adapter with the RPi4. The best is to use model recommended in the shopping list or an X825 adapter board.

My setup was working with the same HDD (WD, WDBU6Y0020BBK-0B) since v1.1.
I run into the same problem even with no HDD connected

ak1008, 2020-12-02:
Same issue here on a new RPi4. I dont have a LCD. 
Writing the raspiblitz image (raspiblitz-v1.6.1-2020-10-20) to SD card (on MacOS) resulted in "unable to mount fs on unknown-block(179 2)". Used two different and new SD cards to avoid SD card issue.
Writing the raspiblitz image with PI image results in "i2c dev entries driver"
I dont plug in the HDMI until a few minutes after starting the raspberry. 
Hope this will be fixed in the next version :-/.

openoms, 2020-12-02:
v1.6.2 is out

ak1008, 2020-12-02:
same "i2c dev entries driver" issue with 1.6.2

sven3550, 2020-12-02:
> same "i2c dev entries driver" issue with 1.6.2

Same for me still

openoms, 2020-12-03:
Can you try to boot without the disk attached and try to plug the disk in the USB2 (black) port?
This makes it slower, but tells if the problem is with the USB disk adapter.

ak1008, 2020-12-03:
Booting via USB results in loop of the following "Timeout waiting for hardware cmd interrupt". "i2c dev entries driver" seems resolved, but not geeting an IP address and hence cannot login via SSH. Any idea?
[image]

ak1008, 2020-12-03:
ok, got this to work by just inserting the SC hard. Now getting to the login screen, but the admin/raspiblitz credentials dont work, neither does pi/raspberry. any ideas? :( Also ssh doesnt work and I dont get an IP address from the router.

openoms, 2020-12-03:
First of all always try with a freshly flashed image in a new environment.

Also disconnect HDMI for boot and then ssh in, can activate the HDMI later with the shortcut:
`hdmi`

openoms, 2020-12-03:
what you mean buy 

>booting via USB

?

You are booting from the SDcard with disk attached to a  USB3 (blue -default) port, USB2 to test or no disk attached at all.

ak1008, 2020-12-03:
sorry, I totally have messed this up. I thought you wanted me to flash USB key and boot from USB drive. 
So please forget about this one.
What I have done now is without any disk attached to USB drive:

1) With HDMI attached: Booted raspiblitz image from SD card slot without any disk attached and it results in "i2c dev entries driver" issue as before. 
2) Without HDMI: as I cannot see anything, I'm relying that my router shows new device so I can SSH into device. Doesnt work, no new IP is assigned (and it's not he LAN cabel as this work fine with other device).

openoms, 2020-12-03:
When you boot the same device it usually gets the same IP even with a new SDcard.

You might be able to ssh in from another device like Termux on a phone or MobaxTerm on windows.

The hostname which should appear on your router is `raspberrypi`.

ak1008, 2020-12-03:
No IP is given a I cannot ping raspberrpy nor raspberrypi.local. Stuck on "i2c dev entries driver" i as said before.

openoms, 2020-12-05:
You can try anotyer SDcard or if that fails test a clean RaspberryOS image to make sure it is not a problem with the board itself.

Sknej, 2020-12-06:
I'm having the same issue as **ak1008** and tested like you suggested. I mounted a raspberryOS image into my SDcard and it booted up correctly. When I tested again a 1.6.2 image, it got stucked on "i2c dev entries driver" as before.

openoms, 2020-12-06:
After reading this: https://openest.io/en/2020/01/18/activate-raspberry-pi-4-i2c-bus/
 I presume  the issue can be solved by disabling the i2c  device driver altogether.

The display driver activates it (unnecessarily imo): https://github.com/goodtft/LCD-show/blob/d9b3378f9360174157ee0217df5104f0d0514f54/LCD35-show#L20
(this is run when building the sdcard image)

Test solution: 
Put the fresh v1.6.2 SDcard image in the desktop computer and edit:
the config.txt on the boot partition (the only one windows can read iirc) and comment out the line: `dtparam=i2c_arm=on` so it shall look like this with a `#` in front:

`#dtparam=i2c_arm=on`

Save and boot the RPi with the edited image. Hope this solves the problem.

Achillian, 2020-12-07:
Same issue on 1.6.1, started happening after a power loss.
Don't have LCD screen, don't have HDMI connected. Only UGREEN usb 3.0 with 1TB evo 860 SSD inside. Had this setup working for a while since 1.3

openoms, 2020-12-07:
@Achillian how do you know it is this issue without any display attached?

Changing the SDcard would be the first step, at least try to reflash and use the current v1.6.2 image. You have got all necessary for an emergency (force close) restore, but the disk has all the channel states. It can be read by any linux including your Rpi and Tails/Ubuntu booted from USB on your desktop.

Achillian, 2020-12-07:
@openoms after a few days of my node not responding to SSH and reboots, I ordered micro HDMI cable and connected it to a monitor to see what's going on.

Thing is I don't want to force close all the channels, I spent months building liquidity to be able to accept payments for my online store, and now if I use channel.backup file with the new SDcard image I will lose all the channels.
I am going to connect the disk to a ubuntu, what file am I looking for? Thanks.

openoms, 2020-12-07:
@Achillian the lnd-rescue file only contains the lnd directory from.the disk. You can just copy that in place of a working setup to restore your old wallet.

There are:
wallet.db and channel.db containing the channel states, but just copy the whole directory.

You must make sure that you don't run an outdated copy of the channel.db.

Achillian, 2020-12-07:
@openoms so I just set up a fresh 1.6.2 install with a new LND wallet/seed phrase and then replace lnd folder with my old lnd folder?

openoms, 2020-12-07:
@Achillian yes, exactly. Same happens when you use the lnd-rescue...tar.gz file.

Sknej, 2020-12-07:
> After reading this: https://openest.io/en/2020/01/18/activate-raspberry-pi-4-i2c-bus/
> I presume the issue can be solved by disabling the i2c device driver altogether.
>
> The display driver activates [...]

Even doing that, the screen stills freezes at the i2c /dev entries driver.

openoms, 2020-12-07:
@Sknej that is strange.
You could try building the raspiblitz sd-card manually or could test this work-in-progress image : https://github.com/rootzoll/raspiblitz/pull/1833#issuecomment-737295779
built on a different base (the 64bit RaspberryOS) and cleaner config.txt settings.

Achillian, 2020-12-08:
I was able to fix the issue and recover all the channels intact.

What I've done:
1. Made a new SDcard image with 1.6.2
2. Before inserting it into Rpi, opened `config.txt` and commented out this line "`#dtparam=i2c_arm=on`" as suggested by @openoms to disable i2c_arm
3. Connected my SSD which was untouched (with all the data on it)
4. Started the setup process normally, selected "Setup Bitcoin and Lightning (DEFAULT)"
5. It asked me to make new Password A

After that it compiled everything on SD card and just started syncing from the last block.
All the channels and all the funds are there as if nothing even happened.
Hopefully this helps someone else.

Zestforlife808, 2020-12-08:
I am having the same issue as @Sknej.  My Pi refuses to boot even when using a freshly made SDcard with 1.6.2 with #dtparam=i2c_arm=on" commented out in config.txt prior to first boot.

I was able to get the PI to boot again by using the RaspberryOS 64 bit build linked by openoms but now I can't get joinmarket to work.  Everything was working fine (Electrum personal server, joinmarket, etc) until I had a power outage and then the pi would refuse to boot past the i2c error.  

If I attempt to run the joinimarket scripts directly I get the following
`
(jmvenv) joinmarket@raspberrypi:~/joinmarket-clientserver/scripts $ python wallet-tool.py generate
Traceback (most recent call last):
  File "wallet-tool.py", line 2, in <module>
    from jmbase import jmprint
ModuleNotFoundError: No module named 'jmbase'
`

When doing an upgrade from the joininbox GUI it eventually errors out with : 
`
Exception: Version mismatch: this is the 'cffi' package version 1.14.4, located in '/tmp/pip-build-env-xjx22cwu/overlay/lib/python3.7/site-packages/cffi/api.py'.  When we import the top-level '_cffi_backend' extension module, we get version 1.12.2, located in '/usr/lib/python3/dist-packages/_cffi_backend.cpython-37m-aarch64-linux-gnu.so'.  The two versions should be equal; check your installation.
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/joinmarket/joinmarket-clientserver/jmvenv/bin/python3 /home/joinmarket/joinmarket-clientserver/jmvenv/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmp8srhwdxn Check the logs for full command output.
Joinmarket was not installed. Exiting.
`

openoms, 2020-12-08:
@Zestforlife808 Something went wrong during the joinmarket installation, try to reinstall it from the SERVICES  menu.
If it continues to be an problem open a new issue as this is not anymore related to the I2C error.

rootzoll, 2020-12-15:
We will give it a with a clean uped `config.txt` with the move to 64-bit for the  v1.7 as @openoms described - lets keep fingers crossed.

Cranky4u, 2020-12-23:
i2c hang issue here too...set up is
Raspberry Pi 3B+
Samsung 1TB SSD
RaspiBlitz 1.6.2
SD = 32GB
Development systems = Win 7

Pi hangs at ~5 sec mark with i2c issue described in this thread. Tried several setups including;
Multiple SD cards (15, 32, 64 & 128)
SSD unplugged / plugged
HDMI unplugged / plugged

Of interest, the Pi does acquire an IP address as seen by my router and can be pinged. SSH is refused - have tried establishing / turning on SSH and confirmed ports open on my internal home LAN (6 PCs, Xbox, smart TV, NAS, phones, etc...)

After 4 days fault finding, think I will hang around for raspiblitz 1.7

Johan4j, 2021-01-29:
#1987

Johan4j, 2021-01-30:
I got the same issue. I get the i2c issue!

Marcioshub, 2021-02-18:
Same issue, tried it on 2 different RPI4 and 3 different SSDs (samsung, Crucial, Pny) and nothing worked.

openoms, 2021-02-18:
@Marcioshub did you use the same USB- disk adapter? Is that from the shopping list?
Make sure to not connect anything to the HDMI when booting.

Marcioshub, 2021-02-18:
> @Marcioshub did you use the same USB- disk adapter? Is that from the shopping list?
> Make sure to not connect anything to the HDMI when booting.

Yup still nothing... A few times when restarted and i ssh into it, it says "WARNING: NO HDD FOUND -> Shutdown, connect HDD and restart" over and over. Anyone know any better alternatives for the rpi4? This software is too buggy.

openoms, 2021-02-18:
Clearly you have a hardware issue even not releated to this one. Use a compatible USB adapter and disk and free to try any other project.

Mr-Rog, 2021-02-20:
hello I have the same issue
last line 4.511126 i2c /dev entires driver
last messagen on my screen, I trx 5 Diffrent 5 zoll LCD Screen and 3 Monitors, same Issue, try but without HDMI etc, try 3 diffrent Raspi, nothing helps?
Donwload the newest image?
Thx for a short Feeback

rootzoll, 2021-03-25:
In the 64-bit base image we use for v1.7
https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2020-08-24/

these are the dafult values:
```
# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on
```

Please report if you see any further `i2c /dev entries driver` problems in the v1.7RC1. Putting this in final testing.

[1 more reply]