raspiblitz/raspiblitz #1059: RESEARCH: Adding Specter-Desktop to RaspiBlitz

rootzoll, 2020-02-29:
_The goal of this project is to make a convenient and user-friendly GUI around Bitcoin Core with a focus on multisignature setup with airgapped hardware wallets._

See Demo:
https://twitter.com/CryptoAdvance/status/1233833767283941376?s=20

See code:
https://github.com/cryptoadvance/specter-desktop

stepansnigirev, 2020-03-03:
@k9ert is working on it :)

k9ert, 2020-03-09:
I've created a branch "specter" in my fork here:
https://github.com/k9ert/raspiblitz/tree/specter
which is currently this commit:
https://github.com/k9ert/raspiblitz/commit/6ebf1f5fd57e0a139d726a5911404826e24c5d16

The current state enables:
* Installation of specter-desktop via services like the others in a virtualenv + pip install cryptoadvance.specter (in home/admin/.specter/.env)
* On top it creates a self-signed-cert and exposes it via https (on port 25441)
* also it creates a Tor-hidden-service. 
* Authentication is enabled by default (Pin of "Password B")
[image]

There are three things which are missing, still:
* The current deinstallation-procedure is deleting the whole /home/admin/.specter directory. That's not very good for updates. I want that to be optional. The user needs to be asked whether that directory should be deleted.
* The tor-service doesn't allow the use of the camera for security reasons. More research needed but a minor anyway for now.
* HWI doesn't work on the raspiblitz yet. This is really something we should get working as not many people have a specter yet and a multisig-setup would need it anyway.

Any hints on how to get the last point working very much appreciated.

rootzoll, 2020-03-11:
Cool .. I am just resetting my development node and will test :)

rootzoll, 2020-03-16:
Merged and in testing for v1.5 release :)

rootzoll, 2020-03-18:
@k9ert got the following expection after install:

```
Mar 18 20:33:17 raspberrypi python3[31788]: Traceback (most recent call last):
Mar 18 20:33:17 raspberrypi python3[31788]:   File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
Mar 18 20:33:17 raspberrypi python3[31788]:     "__main__", mod_spec)
Mar 18 20:33:17 raspberrypi python3[31788]:   File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
Mar 18 20:33:17 raspberrypi python3[31788]:     exec(code, run_globals)
Mar 18 20:33:17 raspberrypi python3[31788]:   File "/home/bitcoin/.specter/.env/lib/python3.7/site-packages/cryptoadvance/specter/__main__.py", line 11, in <module>
Mar 18 20:33:17 raspberrypi python3[31788]:     from .bitcoind import (BitcoindDockerController,
Mar 18 20:33:17 raspberrypi python3[31788]:   File "/home/bitcoin/.specter/.env/lib/python3.7/site-packages/cryptoadvance/specter/bitcoind.py", line 15, in <module>
Mar 18 20:33:17 raspberrypi python3[31788]:     from .server import DATA_FOLDER
Mar 18 20:33:17 raspberrypi python3[31788]:   File "/home/bitcoin/.specter/.env/lib/python3.7/site-packages/cryptoadvance/specter/server.py", line 13, in <module>
Mar 18 20:33:17 raspberrypi python3[31788]:     from .views.hwi import hwi_views
Mar 18 20:33:17 raspberrypi python3[31788]:   File "/home/bitcoin/.specter/.env/lib/python3.7/site-packages/cryptoadvance/specter/views/hwi.py", line 23, in <module>
Mar 18 20:33:17 raspberrypi python3[31788]:     HWI_EXEC = which("hwi")
Mar 18 20:33:17 raspberrypi python3[31788]:   File "/home/bitcoin/.specter/.env/lib/python3.7/site-packages/cryptoadvance/specter/helpers.py", line 221, in which
Mar 18 20:33:17 raspberrypi python3[31788]:     raise Exception("Couldn't find executable %s" % program)
Mar 18 20:33:17 raspberrypi python3[31788]: Exception: Couldn't find executable hwi
```

Any quick hint whats missing?

stepansnigirev, 2020-03-18:
Looks like [HWI](https://github.com/bitcoin-core/HWI) is missing in the PATH.
There is exactly one place where we need to run hwi as a subprocess, and we should probably get rid of it.
I opened an issue (https://github.com/cryptoadvance/specter-desktop/issues/91) and will try to fix it upcoming days, shouldn't be that hard.

stepansnigirev, 2020-03-18:
Should be fixed with https://github.com/cryptoadvance/specter-desktop/pull/92, @k9ert will test and merge if it works :)

rootzoll, 2020-03-19:
@k9ert can you update also the pip repo so that I can test the merged fix?

rootzoll, 2020-03-26:
@k9ert the PIP is still on version 0.1.2 ... is there an update coming in?
https://pypi.org/project/cryptoadvance.specter/

k9ert, 2020-03-27:
I just made a release v0.2.0. Unfortunately we're not yet very good at creating release-notes. In the new version Ben Kaufmann did a nice job adding a labeling-system to addresses.
Just deinstalling and reinstalling specter-desktop should work for the upgrade.

> @k9ert can you update also the pip repo so that I can test the merged fix?
I'm not sure what is meant by that. "The merged fix" is the inmtegration with raspiblitz. Installing specter-desktop should be possible since the first release on pypi. 
Now we have a newer version and you can test the upgrade. That's what you meant?!

rootzoll, 2020-03-27:
@k9ert now its running :) So we are good for the v1.5 RC1 release

About the PIP: yes Stepan did that one fix in specter desktop that was still needed and the raspiblitz script is installing from your pip repo ... so to get it running on the RaspiBlitz an update of the pip repo was needed. thx.

rootzoll, 2020-03-28:
Marked for Hackathon sprint:

- Test Specter Desktop on RaspiBlitz and document (text or video) how to use with coldcard and/or Specter-DIY-Hardware wallet

@k9ert or @stepansnigirev can you mentor this sprint issue over the Lightning Hackday (first April Weekend)

stepansnigirev, 2020-03-29:
@rootzoll how will it happen? should we be somewhere online in a chat and help people with problems?

rootzoll, 2020-03-29:
@stepansnigirev responding here on Github on this issue on questions would be the basic. We will have a Mattermost online where people/groups can have chats on sprint issues - would be cool if mentors would join there too over the weekend if people choose to be active on that issue. More details will follow in the next days.

stepansnigirev, 2020-03-29:
ok, I will keep myself fully available over that weekend :)

rootzoll, 2020-03-29:
perfect :)

k9ert, 2020-04-02:
I'll be there, too!

rootzoll, 2020-04-02:
Cool to have you on board @k9ert & @stepansnigirev :D see all details on the Hacksprint on the wiki: https://wiki.fulmo.org

closed 2020-04-18 by rootzoll