raspiblitz/raspiblitz #1060: Cannot connect to ElectRS over SSL

ascisco, 2020-03-02:
- Running a clean install of Raspiblitz 1.4
- ElectRS installed properly with no errors and fully indexed with no errors
- I Can connect to Electrs over localhost:50001:t
- I Cannot connect to Electrs over localhost:50002:s or address.onion:50002:s

`sudo nginx -t`
returns
`nginx: [emerg] BIO_new_file("/etc/ssl/certs/localhost.crt") failed (SSL: error:02001002:system library:fopen:No such file or directory:fopen('/etc/ssl/certs/localhost.crt','r') error:2006D080:BIO routines:BIO_new_file:no such file) nginx: configuration file /etc/nginx/nginx.conf test failed`

I have been running Raspiblitz 1.3 for months with ElectRS working fine over SSL. 
It seems that something is wrong with the initial electrs/nginx setup process in 1.4

openoms, 2020-03-02:
Here is the answer describing to solve this:
https://github.com/rootzoll/raspiblitz/issues/859#issuecomment-590280339
Start with running this to create a new self-signed cert:
Create a new cert key for nginx using the available script:
`/home/admin/config.scripts//internet.selfsignedcert.sh`

ascisco, 2020-03-02:
That fixed it, thank you. 

Just wanted to note that I got this error on a clean install of 1.4, I did not upgrade from 1.3.

openoms, 2020-03-02:
The line in the solution is run during the electrs setup script, but must be failing.
Could not demonstrate this issue yet on my test nodes, but will keep an eye on it.
Thank you for reporting!

rootzoll, 2020-03-22:
@openoms can we to anything for the v1.5 to improve on this?

openoms, 2020-03-23:
maybe we could run the `internet.selfsignedcert.sh` later on or last in the script.

pretyflaco, 2020-04-09:
> That fixed it, thank you.
> 
> Just wanted to note that I got this error on a clean install of 1.4, I did not upgrade from 1.3.

Had the same issue and this fixed it for me, too. (upgraded from v1.2 to v1.5rc1)

closed 2020-10-27 by openoms