raspiblitz/raspiblitz #1062: RESEARCH: CoinKite Bunker on the RaspiBlitz

openoms, 2020-03-04:
ckbunker.com is great to have for people having a ColdCard MK3 already.

It is ideal to run on this reasonably secure always-on linux server we already use.

Made it work quite easily, documented here: https://github.com/openoms/bitcoin-tutorials/blob/master/ckbunker_on_blitz.md

Could be a new optional feature in the next release.

openoms, 2020-10-10:
This feature got a big boost with JoinMarket developments to allow external signing: https://github.com/JoinMarket-Org/joinmarket-clientserver/issues/663#issuecomment-704546610
https://bitcoinhackers.org/web/statuses/104993677417416904

Funding with a PSBT is already available in LND (using bitcoin-cli): https://github.com/lightningnetwork/lnd/blob/master/docs/psbt.md or with Balance of Satoshis https://github.com/alexbosworth/balanceofsatoshis:
 `bos open [pubkey]`

Initial metal case design with a ColdCard integrated (for HSM mode)
https://twitter.com/zkdesignstudio/status/1314683278067200001

rootzoll, 2022-06-05:
Waiting for PR #3148 to be merged - then changing to final testing.

rootzoll, 2022-06-17:
OK merged for experimental CLI testing as part ob 1.8.0 release. Anybody please ask your questions and report your improvement ideas here - @openoms will be happy to respond.