raspiblitz/raspiblitz #1064: migrating from raspibolt to raspiblitz?

radar24, 2020-03-05:
Hi 

I'm not sure here is the right place to ask this question but
Is there any way to migrate raspibolt to raspiblitz?
I have LN setup with pi3b and raspibolt and I want to turn this to raspiblitz with adding lcd screen and updated pi 4 4GB. I've already built raspiblitz hardware wise. I have searched that LN can't be backuped? without losing balance on LN channel?.. Some insights will be very appreciated. 
Thanks

rootzoll, 2020-03-06:
What you can try is:
- stop lnd on the Raspibolt: sudo systemctl stop lnd
- make sure its really stopped: sudo systemctl status lnd
- copy the complete lnd folder (/mnt/hdd/lnd) to your laptop - use scp
- shutdown the RaspiBolt

on a fresh setuped RaspiBlitz (without any funds on):
- stop lnd service: sudo systemctl stop lnd
- make sure its really stopped: sudo systemctl status lnd
- delete the /mnt/hdd/lnd folder
- copy your backuped lnd folder from laptop to raspiblitz /mnt/hdd/lnd - use scp
- run: sudo chown -R bitcoin:bitcoin /mnt/hdd/lnd
- reboot the RaspiBlitz

If LND is not working on RaspiBlitz after this - it should be safe to boot up your old RaspiBolt and reset the RaspiBlitz to fresh sd card and formatted HDD.

radar24, 2020-03-06:
Thanks for your help!
before doing your recommendation, i want to make sure one thing.
So I should recover bitcoin wallet(that I was using on raspibolt) on new raspiblitz before doing lnd folder recover? or are your suggestion with newly created bitcoin wallet on raspiblitz? 
I have only assembled RaspiBlitz hardware wise and not yet software wise. 
Thanks

rootzoll, 2020-03-06:
The normal bitcoin on-chain wallet you use to fund your lnd node and open channels with is managed by LND and is transfered the way described above - bitcoind is just an information servcie and its own wallet is off by default.

If you activated that bitcoind wallet you could also backup it and move it over, but sending the funds to back to a hardware wallet and then later on to your RaspiBlitz would prob be easier if you can spare the fee.

radar24, 2020-03-08:
I have backed up lnd folder from raspibolt to raspiblitz as you instructed. and i get this error

lnd[23602]: /home/bitcoin/.lnd/lnd.conf:26: unknown option: bitcoind.rpcuser

I inserted bitcoind.rpcuser=raspibolt and password on lnd.conf but I'm still getting the error.. does this mean the my lnd backup is not working on raspiblitz with pi 4?...shoud I give up?..

openoms, 2020-03-08:
@radar see our sample/default lnd.conf: https://github.com/rootzoll/raspiblitz/blob/master/home.admin/assets/lnd.bitcoin.conf

Edit it at:. 
`/mnt/hdd/lnd/lnd.conf`  
and restart the node.

radar, 2020-03-08:
Wrong radar.

@radar24

radar24, 2020-03-11:
> @radar24 see our sample/default lnd.conf: https://github.com/rootzoll/raspiblitz/blob/master/home.admin/assets/lnd.bitcoin.conf
> 
> Edit it at:.
> `/mnt/hdd/lnd/lnd.conf`
> and restart the node.

Thanks! this worked!

closed 2020-03-11 by rootzoll