raspiblitz/raspiblitz #1065: Raspiblitz 1.4: bitcoind not running after fresh setup /

kaimeinke, 2020-03-07:
After a clean setup of Raspiblitz 1.4 I ran into trouble to finish the setup process. I retried four times now and always get the same results.

- bitcoind not running
- unable to restart or power off via console commands

I am using a Raspberry Pi with a 3TB HDD and ShimFan which was running for weeks on 1.3 without any failure.

Do you have any idea how I can continue or what I am missing here?

sudo tail -f /mnt/hdd/bitcoin/debug.log
returns
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory
tail: no files remaining

bitcoind
returns
Error: Cannot write to data directory '/mnt/hdd/bitcoin'; check permissions.

sudo shutdown -r now
leads to a few seconds of waiting and then is ignored and just brings me back to console

poweroff does not power off as well

I already tried a full reindex with 'sudo bitcoind -reindex' and got
2020-03-07T08:27:38Z UpdateTip: new best=00000000000003036ea3db6008dfb36b77033731540144c14c78fba68071dcbb height=202481 version=0x00000001 log2_work=68.830123 tx=7816952 date='2012-10-09T06:42:07Z' progress=0.015263 cache=315.7MiB(2792773txo)
2020-03-07T08:27:38Z msghand thread exit
2020-03-07T08:27:39Z scheduler thread interrupt
2020-03-07T08:27:39Z Dumped mempool: 8e-06s to copy, 0.01127s to dump
2020-03-07T08:27:48Z *** Disk space is too low!
2020-03-07T08:27:48Z Error: Disk space is too low!
Error: Disk space is too low!
2020-03-07T08:27:48Z ForceFlushStateToDisk: failed to flush state (Disk space is too low! (code 0))
2020-03-07T08:27:48Z *** Disk space is too low!
2020-03-07T08:27:48Z Error: Disk space is too low!
Error: Disk space is too low!
2020-03-07T08:27:48Z ForceFlushStateToDisk: failed to flush state (Disk space is too low! (code 0))
2020-03-07T08:27:50Z [default wallet] Releasing wallet
2020-03-07T08:27:50Z Shutdown: done

Thank you very much for your help and for all the heart and work you put into this amazing project.

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.4
chainnetwork: bitcoin / main
 08:00:49 up 6 min,  2 users,  load average: 0.18, 0.63, 0.37

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; disabled; vendor preset: enabled)
   Active: inactive (dead)

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Sat 2020-03-07 07:17:01 GMT, end at Sat 2020-03-07 08:00:49 GMT. --
-- No entries --

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory

*** LND SYSTEMD STATUS ***
Unit lnd.service could not be found.

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Sat 2020-03-07 07:17:01 GMT, end at Sat 2020-03-07 08:00:49 GMT. --
-- No entries --

*** LAST 30 LND INFO LOGS ***
sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log' for reading: No such file or directory

- TOUCHSCREEN is OFF by config

- Loop is OFF by config

- RTL is OFF by config

- Electrum Rust Server is OFF by config

- BTCPayServer is OFF by config

- LNBits is OFF by config

*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0
raspberryPi=4
powerFAIL=0
powerWARN=0
powerMIN='850000 microVolt'
tempFAIL=0
tempWARN=0
tempMAX='5900 centiGrad'

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.178.24'
tempCelsius='48.6'
uptime=404
upsStatus='OFF'
startcountBlockchain=0
bitcoinActive=0
bitcoinErrorShort='Error found in Logs'
bitcoinErrorFull='0'
# OK: log 'ERROR: Error found in Logs' written to /home/admin/systemd.blockchain.log
startcountLightning=0
lndActive=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=0

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

UNKOWN STATE - there is blockain data config, but blockchain service is not running
It seems that something went wrong during sync/download/copy of the blockchain.
Or something with the config is not correct.
Sometimes a reboot helps --> sudo shutdown -r now

openoms, 2020-03-07:
Try to boot with without the shimfan. See the about the problems with the shimfan interfering.

If still have an issue the problem is a faulty disk.

kaimeinke, 2020-03-07:
Thank you very much. I will try this and come back.

kaimeinke, 2020-03-07:
Okay, everything is running MUCH more smooth now. It's faster, more responsive and errors seems to be gone. I will continue setup now.

closed 2020-03-07 by kaimeinke