raspiblitz/raspiblitz #1067: Collaboration with cryptoanarchy deb repo?

Kixunil, 2020-03-07:
Hi, I'd like to explore the possibility of collaboration with [cryptoanarchy deb repository](https://github.com/Kixunil/cryptoanarchy-deb-repo-builder), which approaches the topic of making a Bitcoin full node with accompanying software using Debian packages.

# The advantages of Debian packaging

* Automatic updates
* Automatic signature verification
* Dependency resolution solved by APT
* Clearer boundaries around individual components (that means not 100% clear, but I have some ideas to make them even clearer than now)
* Easy installation and integration of new components (as opposed to manually connecting things together or ad-hoc scripts). Just write `apt-get install X` and `X` gets installed and configured together with its dependencies automatically.
* Easy, clean uninstallation
* Integrated configuration using `debconf` and other tooling.
* `debconf` makes it possible to avoid irrelevant configuration options (e.g. port numbers) without missing the important questions (e.g. "Your disk space is low, select another location for Bitcoin timechain data")
* Various linting tools making sure the packages are less likely to break
* Detection of conflicting files (unless the package specifies `Replaces`, obviously)
* Time-tested approach to software deployment
* Active development of the system outside of Bitcoin circle

# The disadvantages of Debian packages

* Additional overhead beyond steps necessary to make things function - e.g. mandatory documentation. I created my own tools to help with this.
* May take longer to understand for new contributors unfamiliar with Debian packaging
* Necessary compliance with somewhat strict rules if the packages should end up in official repositories - this is not mandatory and I currently sidestep this problem by having [my own deb repository](https://deb.ln-ask.me). It'd be nice to have the packages as official if possible, but it's not my top priority. (Note: the rules are very sensible, they just shoot for a higher quality than I'm currently willing to deliver.)

# Current status of my project

Cryptoanarchy deb repo builder is currently experimental and so I wouldn't advise to integrate it into raspiblitz **at this moment**. This issue is meant to open the discussion about sharing our resources in order to create a better system, hopefully sooner.

Things that might be relevant/interesting for raspiblitz, in no particular order:

* Electrs fails to cross-compile, didn't try normal compilation on RPi yet. Compiles fine on x64
* The project adheres to Linux Filesystem Hierarchy Standard and seeks to adhere to other relevant standards (if not, it's considered a bug, unless it has to break the standard for some very good technical reason)
* The project already separates all relevant processes under different uids.
* The project uses [`btc-rpc-proxy`](https://github.com/Kixunil/btc-rpc-proxy) to make permissions even more granular. This is a notable exception to the above because there's zero security benefit running it under a different user than `bitcoin-mainnet`
* The project strictly separates mainnet from testnet/signet, although currently by virtue of non-mainnet instances not existing (:D), the plan is to support it natively, under their respective users.
* All packages use debconf as the primary source of configuration - this auto-generates configuration UIs from provided templates, although many questions have sensible defaults.
* The packages connect to each other intelligently (assuming the user doesn't manually bypass debconf). Example: `btc-rpc-proxy` is automatically configured to use correct RPC port of `bitcoind` and it's automatically, correctly reconfigured when it's changed using `dpkg-reconfiure`. Making this work was a bit tedious, so there are plans to make it even cleaner.
* Appropriate services are restarted automatically when upgraded (though there is a not too harmful bug with unrelated services being restarted)
* bitcoind can be reconfigured to use a different directory than `/var/lib/bitcoin-mainnet`
* systemd is used for managing all services
* No usage of Docker - it'd go against the goals of making *integrated* system (Docker seems to strive to make *disintegrated* systems and doesn't play with deb packaging well)
* btcpayserver uses `dbconfig` (not to be confused with `debconf`) to automatically, correctly configure the database
* because of lack of Docker, btcpayserver uses what they call "manual deployment" which isn't supported much. I plan to open a discussion about it with btcpay devs to figure out how to improve the situation.
* btcpayserver isn't currently automatically integrated with nginx - this is the task being worked on right now
* There is a special configuration for cases where a package depends on other package being configured with some specific settings. The most notable example is `electrs` requiring the absence of pruning. In this case, there's a special package `bitcoin-fullchain-mainnet` that's only responsible for configuration - has no code, nor services. The installation of this package turns off pruning. `electrs` then depends on this package. (Actually, during writing of this I realized, I forgot to add this dependency correctly. :D Will be easy fix soon.)
* All packages intentionally leave user data (bitcoin timechain, LN wallet, btcpay data...) and system users after uninstallation. I wonder how to make this cleaner without risking accidental data loss.

Interesting planned features

* Cleaner, even more declarative packages - easier auditing, might allow adding new semi-trusted repositories that can't compromise whole system during installation
* Deterministic builds
* More Bitcoin/non-Bitcoin packages: RTL, Wasabi RPC server, Open Bazaar...
* Much cleaner ways of connecting various services
* Tor support
* TLS support
* Multiple lightning wallets/btcpayservers behind different Tor addresses - for privacy
* SSO
* Systematic approach to backups - a well-defined interface for packages to register their backupable data and a way to suspend services during backups, if needed (think database transactions).
* Automated tests

# What I need to know

We have discussed this briefly at LNConf, so as far as I remember, this is interesting for you. I'd like to know some things to understand how should I continue.

* What features are missing in my project that would be necessary for it to be used within raspiblitz?
* What kind of additional support is expected from me? (editing some existing files? more stable apt repository? ...)
* What kind of support can I get? (what I'm interested in is help with packaging new things, testing, bug reports, brainstorming)

## Disclaimer/clarification

Raspberry Pi is currently **not** my target platform. I have some specific requirements about automation and UX. As much as I like to contribute to Open Source projects, I can't afford to do this kind of contribution without getting anything back. I'm seeking balanced cooperation, where I provide a framework for doing the things described above and support for that framework and get back help with testing and packaging new things. I figured it's better to be open and honest about this. Hope we can find some common ground.

rootzoll, 2020-03-11:
I can see that idea and if a trusted community builds around that repo it can be a source to make installs of apps easier and get interesting for a project like RaspiBlitz. @Kixunil keep us updated.

Kixunil, 2020-03-12:
What would be the best approach in your opinion? To wait until every piece of Raspiblitz is packaged or to have mixed version first? Mixed version seems to be easier to me, but maybe there's a very good reason to avoid it?

rootzoll, 2020-03-12:
Is it possible to start with apps like wasabi/btcpayserver/qpay while keeping bitcoin/lnd like it is for the time beeing? Also can you maybe give on the next Lightning Hackday a short overview how to plan to build up a community review over that repo?

Kixunil, 2020-03-13:
Since bitcoin and lnd are dependencies of all others doing it in the opposite direction is not great, but it is technically possible. We could create special packages that contain nothing and provide the necessary features (configuration mainly) and have `Provides` field. Ideally we would have them as a separate component of the resulting deb repo to avoid accidentally messing up normal installations.

I think the overview is short enough to write it here. :)

* First, I want to have something working to show how user-friendly the packages are
* If people want to trustlessly pick up from that, they could review the build scripts and build it themselves
* I already "cheat" with `bitcoind` and `lnd` by downloading already deterministically built binaries (this was originally to cut build time, but it looks like I will keep it due to security benefits)
* According to docs Debian packaging itself is deterministic, it's possible I introduced some quirks, I will check at some point and fix if needed
* I plan to sign the whole thing with [`codechain`](https://github.com/frankbraun/codechain)
* Publish signed statement about the hashes of all packages
* Anyone who runs the builder should get the same hashes, so they can sign the same statement - multiple signatures -> more confidence
* In far future I might be able to provide financial incentive for reviewing
* There are some issues around `debcrafter`, that I'd like to solve before all this (mainly to make debcrafter more modular, cleaner and easier to review; possibly also moving the whole building to it, skipping makefiles, which are annoying to work with)

I don't plan to attend Lightning Hackday in person at this moment (this could change, but unlikely, especially due to Corona, I'm surprised it wasn't canceled yet), but I would be interested in remote participation if possible.

rootzoll, 2020-03-16:
The Lightning Hackday will be online/remote - so there will be a chance.

Maybe it makes sense to start first a new kind of device to use your repo for - is there a open router or a NAS based on debian for which the repo can be used to easily build a node plugin?

Once there is a good trusted codechain governance we could think about changing the lnd & bitcoind installs.

Kixunil, 2020-03-16:
I don't know about existing Debian-based routers/NAS. I did some basic tests on Odroid H2 with freshly installed Debian 10 and it looks good. I will definitely test more. I'd like to finish RTL (working on it right now) before serious testing.

openoms, 2021-03-17:
@Kixunil just a heads up, that we are moving to 64bit arm (aarch64) with the next release. 

We should look into this more as using the repo would make the installations much quicker especially using:
* electrs
* btcpayserver
* btc-rpc-explorer
* RTL
* Thunderhub
would save a lot of time versus compiling from source.
It could also help in: https://github.com/rootzoll/raspiblitz/issues/1126

My question is how much can we keep up with the quick version iteration of the apps like ThunderHub? Building from source leaves a lot of freedom, but as I understand we would need to update in https://github.com/Kixunil/cryptoanarchy-deb-repo-builder first to be able to test the latest releases.

Kixunil, 2021-03-17:
Great to hear that! I already have someone testing 64 bit version of the repository on RPi 4 and it seems to mostly work, though not entirely without problems.

> how much can we keep up with the quick version iteration of the apps like ThunderHub?

Depends on what your requirements are. If you want to review every single release with zero trust in the author, it'd be tough. However, if you assume the author of the app is trusted, then it's very easy most of the time - just add a changelog entry and re-run tests.

As an example, [latest BTCPayServer update](https://github.com/Kixunil/cryptoanarchy-deb-repo-builder/commit/47a6ee8dc7ebe36ca7305ca7c1e11af81087f8ff) was trivial, just needed to update the tests as BTCPay improved some flows. On the other hand some things need other changes - changed APIs, signature validation, new useful features, configuration migrations.

In the specific case of ThunderHub, there was a significant roadblock regarding breaking change of root path. This means my version is lagging behind as I didn't have the time to resolve it properly.

There's multi-stage testing going on - first the automated tests are run (x86_64 only, though). If they succeed, the package is added to `experimental` repository - only that one currently contains arm64 version. It needs special setup which is hidden to discourage random noobs from using less-tested, possibly hazardous software. (Thank God, it had one major incident and this limited the damage to a single instance.) Feel free to contact me privately for usage instructions.

After an app was in `experimental` for some time and no issues were reported, it goes to `beta`. There's no `stable` yet but I have a plan to create it ~soon.

Exceptions when an update can bypass experimental are highly critical bugs. E.g. in the past there was a bug in ThunderHub that could completely knock out LND. So the updated package went straight to `beta` (but still with automated test being run). Thankfully, it didn't affect anyone using `beta`.

rootzoll, 2021-03-23:
For v1.8 we will have some refactoring todo anyway .. I will mark this for a closer look and teamdiscuss again.

Kixunil, 2021-03-23:
Feel free to ask any questions if something needs to be clarified! I should be flexible enough to reserve time for more interactive communication if needed.

nyxnor, 2021-04-18:
Very cool. This would solve the problem of not being able to run on a VM, as he run on Quebes OS with Debian. Also, the packaging style helps upgrade and remove of services, using native debian tool (APT).
Just 1 question, the user need to store the files in a server, in order to be able to update independently?

Kixunil, 2021-04-18:
> the user need to store the files in a server, in order to be able to update independently?

I don't understand, everything is handled for the user.

nyxnor, 2021-04-18:
@Kixunil Sorry if Im misinterpreting something. I was because of the need to add your key and reacg your repository. The plan is each user self hosting it, and using a main repo for updates on releases?

Kixunil, 2021-04-20:
@nyxnor various people have various levels of trust. Those who trust me (and Microsoft) sufficiently can just add those keys/repositories and then everything works with `apt install`. Those who don't can build from source after reviewing it.

Since most packages are deterministic I have hopes for some nice multi party verification scheme one day in the future.

rootzoll, 2021-04-29:
Hi @Kixunil to give it a closer look in preparation for v1.8 I tryied to use it on a clean RaspberryPi - with this base image that we also use for RaspiBlitz v1.7: https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2021-04-09/

I floowed the setup steps: https://github.com/debian-cryptoanarchy/cryptoanarchy-deb-repo-builder#how-to-setup-the-beta-debian-repository

But when I for example want to install `sudo apt install selfhost-dashboard` ... I get:
```
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 selfhost-dashboard : Depends: selfhost-dashboard-bin (>= 0.2.1) but it is not installable
                      Recommends: selfhost (< 0.2.0) but it is not going to be installed
                      Recommends: selfhost (>= 0.1.6-2) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

What did I miss? How to fix this?

Kixunil, 2021-05-01:
Beta doesn't have packages compiled for arm64, experimental has. I like to avoid blatantly publishing the instructions for experimental, so that noobs wouldn't use it. I can send it using Keybase, do you use it?

rootzoll, 2021-05-01:
I dont use keybase for messaging .. I thinks ist safe to post instructions here - and it has the name experimental in it, so it should quite clear that this is just for testing.

Kixunil, 2021-05-01:
OK, if any noob sees this I'm not providing support for it to people unfamiliar with command line.

There's a [different key](https://raw.githubusercontent.com/debian-cryptoanarchy/cryptoanarchy-deb-repo-builder/master/tests/data/experimental_key.gpg), which you can just wget and import to system keyring. Then `/beta` needs to be replaced with `/experimental` in the APT sources. That should work.

Note that BTCPayServer doesn't work because MS doesn't provide deb packages. (I hope to build them on my own, but didn't have time yet.) ThunderHub doesn't work because npm sucks at cross compilation of native code.

Other things should work fine.

rootzoll, 2021-05-02:
@Kixunil ok so this seems to work so far on my test raspberrypi-64bit:
```
wget https://raw.githubusercontent.com/debian-cryptoanarchy/cryptoanarchy-deb-repo-builder/master/tests/data/experimental_key.gpg
gpg --import ./experimental_key.gpg
gpg --export F3BA603CE991ED0E1ED2CDDA77575D3DC7473C86 | sudo apt-key add -
gpg --keyserver hkp://keyserver.ubuntu.com --recv-keys BC528686B50D79E339D3721CEB3E94ADBE1229CF
gpg --export BC528686B50D79E339D3721CEB3E94ADBE1229CF | sudo apt-key add -
echo 'deb [arch=amd64,arm64,armhf] https://packages.microsoft.com/debian/10/prod buster main' | sudo tee /etc/apt/sources.list.d/microsoft.list > /dev/null
echo 'deb [signed-by=F3BA603CE991ED0E1ED2CDDA77575D3DC7473C86] https://deb.ln-ask.me/experimental buster common local desktop' | sudo tee /etc/apt/sources.list.d/cryptoanarchy.list > /dev/null
sudo apt update
```

It also looks good when i install the `selfhost-dashboard` .. I see a port listening on `127.0.0.1:9009` where can I config it so that I can call the dashboard from the outside within local network?

frennkie, 2021-05-02:
@rootzoll  @Kixunil reading the code I would think that `127.0.0.1` is currently hardcoded: https://github.com/debian-cryptoanarchy/selfhost-dashboard/blob/master/selfhost-dashboard/src/main.rs#L144 (as a workaround you can use either SSH tunneling or an Nginx Reverse config... :-D )

Kixunil, 2021-05-03:
Correct, it's hard-coded because it relies on Nginx anyway to properly route requests to different apps.