raspiblitz/raspiblitz #1068: raspiblitz 1.4 install does not complete and no SSH connection after reboot

MrSlumber, 2020-03-08:
Hi there, 

please allow me at first to thank Christian and all the other contributors for this fantastic project!!! Thank you!!! :-)

Ok, here's what happened: did a fresh 1.4 install (formatted SD, SSD and the 32GB USB Stick). I was running 1.4 on the exact same Pi for some days without any problems but found out that the 500GB SSD just was to small longterm. So I ordered a 1TB SSD and a new SSD-Case 
(  [https://www.amazon.com/UGREEN-Enclosure-External-Adapter-Housing/dp/B07D2BHVBD/ref=sr_1_3?keywords=ugreen+ssd+case&qid=1583686324&sr=8-3](url) ). 

Replaced the old SSD+Case with the new one (SSD+Case) and the problems started to show: couldn't complete 4 installs, every time because "bitcoind didn't start". Sometimes it did start but after the first reboot I wasn't able to SSH into the Pi because the "connection was refused". 

Tried some more installs with BTRFS feature and without, still at least one of the errors mentioned above could be noted...

To be brief if someone else has these kind of problems and doesn't want to spend nearly half a day to figure it out, here's why: 

**The problem was the USB C 3.1 case!** I replaced it with the case mentioned in the shopping list and bingo: even the BTRFS install showed no problems! 

So, lesson learned, **please stick to the shopping list** mentioned here and save yourself some time :-) 

Hope that helps anyone with the same problems! 

P.S. I had some **problems before with the shim-fan**. I **replaced it with a geekworm case** which now keeps the Pi passive cool in between 40 - 52° Celsius! 
( [https://www.amazon.com/dp/B07VD568FB/ref=twister_B07VD58LGV?_encoding=UTF8&psc=1](url) )

ostermayer, 2020-03-10:
we're you running with a screen?

it is possible you are experiencing this bug https://github.com/rootzoll/raspiblitz/issues/1070

MrSlumber, 2020-03-10:
Yes, I am running the display from the shopping list. I tried it with and without the display. For me I didn't make a difference speaking of the above mentioned problem. For me it seems that the display wasn't the "troublemaker".

MrSlumber, 2020-03-10:
> it is possible you are experiencing this bug #1070

Just read about this but no, never happened on my machine.

jpduhen, 2020-03-15:
SAme overhere, when I take out the SD and manually copy a SSH-file an a spa-supplicant.conf in the /boot directory, all is working again. Only to get stuck at a loop when LND won't start after creating a wallet.
Also the /etc/resolv.conf is incorrect as it holds name server 192.168.188.1 instead of 8.8.8.8 and 1.1.1.1 or 8.8.4.4.

rootzoll, 2020-03-16:
> Also the /etc/resolv.conf is incorrect as it holds name server 192.168.188.1 instead of 8.8.8.8 and 1.1.1.1 or 8.8.4.4.

Thats a great find! Will check that out!

rootzoll, 2020-03-18:
OK - version v1.5 RC1 will have the feature to set a flag file on the sd card to make the RaspiBlitz reset/repair the SSH certs. See: https://github.com/rootzoll/raspiblitz/blob/v1.5/FAQ.md#how-can-i-repair-my-ssh-login

Also /etc/resolv.conf should not contain any old data.

If v1.5 RC1 is out (next weeks) - further testing is welcome, but closing this issue for now.

closed 2020-03-18 by rootzoll