raspiblitz/raspiblitz #1070: Boot required control d

ostermayer, 2020-03-09:
[image]

ostermayer, 2020-03-09:
could not boot fully into os until i did control d

when running headless without screen this. it made it appear as if ssh was down

ostermayer, 2020-03-09:
this occurred after rtl install on reboot

openoms, 2020-03-09:
Thanks for reporting I have this issue sometimes too, could not pin down what causes it yet.

ostermayer, 2020-03-10:
i believe this happens if reboot occurs before lnd and bitcoind have stopped.  was able to reproduce by doing shutdown -now while lnd and bitcoind were in the process of stopping

MrSlumber, 2020-03-10:
Keep my fingers crossed that this can be solved. Thanks to everyone contributing! :-)

MrSlumber, 2020-03-11:
Oh no... Today I did a reboot knowing that the Pi wasn't fully synced and... same as @ostermayer posted above... :-(  No chance via SSH also because "connection refused"

Appreciate your help on this topic! Thank you :-)

rootzoll, 2020-03-11:
@MrSlumber if you havent tried - power down and make a fesh sd card, see if it catches up

cereal88, 2020-03-13:
> @MrSlumber if you havent tried - power down and make a fresh sd card, see if it catches up

**cant** login via SSH but now i can ping the pi and i see the loginscreen (login to pi via ssh admin@ip.....password is raspiblitz)

MrSlumber, 2020-03-13:
> @MrSlumber if you havent tried - power down and make a fresh sd card, see if it catches up

Thank you but it didn't change. Did a fresh install but after some time again the same error.

rootzoll, 2020-03-16:
It can be that your SSH keys are corrupted on the HDD - in that a fresh sd card will not help.
See if this fixes the SSH login problem: https://github.com/rootzoll/raspiblitz/issues/420#issuecomment-522248515

MrSlumber, 2020-03-17:
> It can be that your SSH keys are corrupted on the HDD - in that a fresh sd card will not help.
> See if this fixes the SSH login problem: [#420 (comment)](https://github.com/rootzoll/raspiblitz/issues/420#issuecomment-522248515)

Thank you, will try :-)

rootzoll, 2020-03-21:
Also added "fsck.mode=force" for the v1.5 RC1 release ... hope this will fix that issue .. see:
see https://github.com/rootzoll/raspiblitz/issues/1053#issuecomment-600878695

Closing issue.

closed 2020-03-21 by rootzoll