raspiblitz/raspiblitz #1071: Since 3 days Raspiblitz stopped working

lucbran, 2020-03-10:
Hi,
in spite of several reboots I get this error message as attached.
[image]

openoms, 2020-03-10:
https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-do-i-generate-a-debug-report
Share the link with the debug output after running this:
`/home/admin/XXdebugLogs.sh | nc termbin.com 9999`

lucbran, 2020-03-10:
Hi,
here is the log  report: https://termbin.com/8qvi

openoms, 2020-03-10:
your node is syncing, it takes some patience on a RPi3:
`progress=0.998828 `

lucbran, 2020-03-10:
ok so when i read:bitcoinlnd is not running, is not correct info?

openoms, 2020-03-10:
Your bitcoin node is not active, because it is syncing. 

You can say the message is inaccurate.

lucbran, 2020-03-10:
but why this new long syncing after same was running well for long time?

lucbran, 2020-03-14:
hi
looks like I am stuck at the same point...?
https://termbin.com/m8au

openoms, 2020-03-15:
It is hard to tell from this, but likely to be a hardware failure / corruption.
Type:
`repair`
make a backup of the lnd data and reset the blockchain to start syncing from scratch. If does not work worth looking into changing the hard drive.

nyxnor, 2021-04-18:
Pls close this.

closed 2021-04-19 by rootzoll