raspiblitz/raspiblitz #1073: Fan Shim & Sunfounder 3.5" LCD issue

radar24, 2020-03-11:
sorry for another fan shim & LCD issue. there seems no open issue regarding this so I created another one. 
ok so I get the "reboot and shutdown command not working" issue with these two combination.
reboot and shutdown work great when I only use one either Fan Shim or LCD. 
The issue happens only when Fan Shim and LCD is both attached to GPIO.. 
I'm using the latest build of raspiblitz. 
Has anyone solved this issue? 
I have found a similar behavior with other HAT
[https://github.com/raspberrypi/linux/issues/3065](url)
Are Fan Shim and LCD causing the same issue?

openoms, 2020-03-11:
Unfortunately the only solution seems to be to abandon the shimfan: https://github.com/rootzoll/raspiblitz/issues/1006#issuecomment-585752925

redmac68, 2020-03-14:
Unless there is a known solution, the shimfan should probably be taken off the official parts list.

miningdave, 2020-05-08:
Has anyone tried this alternative: https://www.argon40.com/argon-fan-hat-for-raspberry-pi-4-raspberry-pi-3b-and-raspberry-pi-3-b.html

openoms, 2020-05-08:
> Has anyone tried this alternative: https://www.argon40.com/argon-fan-hat-for-raspberry-pi-4-raspberry-pi-3b-and-raspberry-pi-3-b.html

First time I see this. Please let us know your experience if you try it.

The fanshim has been removed from the recommendations due to persisting issues so closing this.

closed 2020-05-08 by openoms

Kingsport95, 2023-05-25:
> > Has anyone tried this alternative: https://www.argon40.com/argon-fan-hat-for-raspberry-pi-4-raspberry-pi-3b-and-raspberry-pi-3-b.html
> 
> First time I see this. Please let us know your experience if you try it.

Hi there, seems nobody has tried the Argon40 at the end?