raspiblitz/raspiblitz #1074: Waiting Close Channels

woeisme, 2020-03-14:
Hello all, 
I managed to restore a RaspiBlitz node that I had run last August that crashed. I recovered what I think were much of the funds by restoring a backup and channels. 
However there was some funds in a channel that I decided to close and now it seems to be stuck in 
'Waiting Close Channels' on RTL for over several weeks. 
Is there some steps I can take to diagnose how to get the funds released?

rootzoll, 2020-03-16:
This sounds more like a LND issue - try to get more details on that on the LND community support forums: https://dev.lightning.community/

closed 2020-05-14 by woeisme

openoms, 2020-05-14:
use `lncli  closechannel` https://api.lightning.community/#closechannel

to rebroadcast the transactions.