raspiblitz/raspiblitz #1078: Update LND 0.9.2

rootzoll, 2020-03-16:
https://github.com/lightningnetwork/lnd/releases/tag/v0.9.2-beta

especially because of bugf fix for RaspberryPi
https://github.com/golang/go/commit/8b77410

openoms, 2020-03-17:
Running 0.9.2 already without issues. The only difference from v0.9.1 is the fix.
Manual update has been tested: https://github.com/lightningnetwork/lnd/issues/4052#issuecomment-597274642

21-btcclub, 2020-03-17:
my node is now on 0.9.0, do I need to update first to 0.9.1 then to 0.9.2 or is there a script to follow?

openoms, 2020-03-17:
you can update LND from any version with this script:
https://github.com/openoms/lightning-node-management/tree/master/lnd.updates#update-lnd-to-v092-beta

rootzoll, 2020-04-18:
Extended even to the update to LND v0.10.0 --> https://github.com/rootzoll/raspiblitz/issues/1116

closed 2020-04-18 by rootzoll