raspiblitz/raspiblitz #1083: Snagging Jetson NANO install

jjohare, 2020-03-20:
Linux Raspiblitz 4.9.140-tegra #1 SMP PREEMPT Mon Dec 9 22:47:42 PST 2019 aarch64 aarch64 aarch64 GNU/Linux

To get this working on an Nvidia Jetson Nano I've just had to switch Python properly and run the script. Thanks! I have some kind of permissions issue with the script that installs and starts Electrum Rust though. 

Here is everything running without Electrum
http://termbin.com/ofv1

Here's the error from the script turning the service on:

info: downloading installer
warning: $HOME differs from euid-obtained home directory: you may be using sudo
warning: $HOME directory: "/home/admin"
warning: euid-obtained home directory: /home/electrs
error: could not create bin directory: '/home/admin/.cargo/bin'
error: caused by: Permission denied (os error 13)
Get:1 file:/var/cuda-repo-10-0-local-10.0.326  InRelease
Ign:1 file:/var/cuda-repo-10-0-local-10.0.326  InRelease
Get:2 file:/var/visionworks-repo  InRelease
Ign:2 file:/var/visionworks-repo  InRelease
Get:3 file:/var/visionworks-sfm-repo  InRelease

report after
http://termbin.com/3npx

I feel like I could work through all the permissions manually but something is wrong regardless.

openoms, 2020-03-22:
Hi, not sure where the problem is. As you say likely a permission problem.

Here is the script which installs Electrs: https://github.com/rootzoll/raspiblitz/blob/master/home.admin/config.scripts/bonus.electrs.sh#L197

You can run line-by-line (or block) to see what fails.

Here is the doc with various ways to monitor once running: https://github.com/openoms/bitcoin-tutorials/tree/master/electrs#monitor-electrs

jjohare, 2020-03-30:
Looks like it's in the electrs install script

+ sudo -u electrs curl --proto =https --tlsv1.2 -sSf https://sh.rustup.rs

+ sudo -u electrs sh -s -- --default-toolchain 1.39.0 -y
info: downloading installer
warning: $HOME differs from euid-obtained home directory: you may be using sudo
warning: $HOME directory: "/home/admin"
warning: euid-obtained home directory: /home/electrs
error: could not remove 'rustup-bin' file: '/home/admin/.cargo/bin/rustup'
error: caused by: Permission denied (os error 13)

I'll try a manual install of it.

jjohare, 2020-04-14:
This is the line that fails when run as admin on the Jetson 
 sudo -u electrs curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sudo -u electrs sh -s -- --default-toolchain 1.39.0 -y
If I sudo directly to electrs and run the installer then it all works fine.

nyxnor, 2021-04-18:
@jjohare How is it going? Did you manage to make it work?

jjohare, 2021-04-18:
I went Pi 4 so I could use my nano for a gakken worldeye in the end but thanks for checking back with me!

closed 2021-04-18 by jjohare