raspiblitz/raspiblitz #1088: Electrum Service Not Running

corbimos, 2020-03-22:
Hello,

I'm running 1.4 and just enabled Electrum Rust Server as a service.

When i try to run the electrum option from the main menu i get this,

`/home/admin/config.scripts/bonus.electrs.sh: line 108: /home/admin/XXdebugInfo.sh: No such file or directory`

Any ideas on how to proceed?

Thanks for the help!

rootzoll, 2020-03-22:
When you are at the terminal - can you call `/home/admin/XXdebugInfo.sh` and copy the part about electrum? Or go in main-menu to "REPAIR" and then "SOFTWARE".

corbimos, 2020-03-22:
Before i had a chance to do that, I disabled the electrum service and deleted the items in the prompt. 

I re-enabled/installed it and now it seems to be building the index. 

seems like it didn't get installed properly the first time?

@rootzoll Thanks for the response!

closed 2020-03-22 by corbimos