raspiblitz/raspiblitz #1090: Updating BTCPayServer to v1.0.3.164

rootzoll, 2020-03-22:
Small update. Make one test run again, before release.

rootzoll, 2020-03-23:
@openoms I reverted back to the old version to clear another bug first. Can you second check: 

- When I install TOR & then BTCPayServer the server semmes running
- I get a wget response on 23000
- but when calling the onion address in a TOR browser there is no answere

openoms, 2020-03-25:
Couldn't reproduce this. Updated the scripts and uninstalled Tor. Installed BTCPay v1.0.3.164 together and after Tor too, the Hidden Service became available after a couple of seconds of returning to the main menu.

[image]

openoms, 2020-03-25:
Could not reproduce this problem. I have synced and modified the latest script uninstalled Tor and reinstalled Tor + BTCPay together and then one-by-one too.
Don't see a reason in the changelog which could cause this.
v1.0.3.164 working fine:
[image]

rootzoll, 2020-03-25:
OK it looks like its something with my TOR desktop browser - BTCPAY Server is showing up on my mobile TOR browser. So we are good here.

openoms, 2020-04-16:
New major BTCPay release 1.0.4.0 with Payjoin support (requires an other hot wallet, not tested on the blitz yet):
https://docs.btcpayserver.org/features/payjoin

https://twitter.com/BtcpayServer/status/1250853659618967553

Will test and update here.

Their LND support is still partially broken, unfortunately only works properly with LND 0.8.2 for now.
https://github.com/rootzoll/raspiblitz/issues/1104

openoms, 2020-04-16:
Installation looking good on 1.5RC1 with:
https://github.com/btcpayserver/btcpayserver/releases/tag/v1.0.4.0
and
https://github.com/dgarage/NBXplorer/releases/tag/v2.1.24

openoms, 2020-04-16:
Managed to generate a hot wallet and the payjoin link is available to be paid with another BTCPay wallet as described in: https://docs.btcpayserver.org/features/payjoin

rootzoll, 2020-04-17:
OK is merged - looking for final testing on v1.5 RC2 (coming as soon as lnd 0.10.0 is out).

rootzoll, 2020-04-18:
Test look good .. closing for v1.5 RC2 release.

openoms, 2020-04-20:
https://github.com/btcpayserver/btcpayserver/releases/tag/v1.0.4.1

rootzoll, 2020-04-22:
Merged update. TODO: One final install test.

openoms, 2020-04-28:
https://github.com/rootzoll/raspiblitz/commit/05d6e644fcb9bfa54a3d82f5656054f022ac72b0

closed 2020-04-30 by rootzoll