raspiblitz/raspiblitz #1091: rpi reboot on blockchain syncing

grunch, 2020-03-23:
Hi, I'm syncing bitcoin blockchain on a rpi 4, I'm using raspiblitz 1.4, I'm kinda stuck on 90% for two days, the rpi reboots after a period of time and I can see this message.

[image]

After that, I just restarted it and it keep syncing like nothing happened, until it happens again.

Here is the log, I don't see anything related with the error we see in the picture maybe you can help me with this :smile: 

https://termbin.com/fz49

rootzoll, 2020-03-23:
I saw this once on a BTRFS setup ... is this ext4?
Can you SSH in? if yes -> run on terminal `sudo /home/admin/config.scripts/blitz.datadrive.sh swap off` and reboot

grunch, 2020-03-24:
I'm sorry, the OS stop working and couldn't ssh in, I just started the process again, I'll let you know if I have this one again or a new issue

closed 2020-03-24 by grunch