raspiblitz/raspiblitz #1094: bitcoind not working

cool-boop, 2020-03-26:
Hello.
After downloading bitcoind and downloading some blocks from the blockchain, the Raspberry Pi tells me that "The bitcoind service is not running." When I look into the debug logs, I see that it says bitcoind is active and running. What is the error in this case and how can I fix it? (The full debug log is here: https://termbin.com/8bc6) (Also attached is the error message the computer gives me, which contradicts the debug log)
[image]

Thanks in advance for any help!

rootzoll, 2020-03-27:
From the logs it looks like its rolling thru all. the blocks ... this may take a while. Please check the logs regularlay and see if yoz still can see progress or if it got stuck on some error.

cool-boop, 2020-03-28:
It seems that I can't download any more blocks; when the Raspberry Pi were to boot up, the system would automatically start checking for bitcoind (which gives the same error as I first described).

rootzoll, 2020-03-28:
If data on your hard drive got corrupted I would maye start fresh. Format HDD to FAT format on your lapop and make a fresh SD card.

cool-boop, 2020-03-30:
Thank you for the advice! I started fresh and formatted the HDD, and it is not giving me the initial error.

closed 2020-03-30 by cool-boop